Z-Bolt/OctoScreen #99: Octoscreen not loading after reboot

jansenchua, 2020-03-06:
Hi I have done a fresh install of octopi and also my 3.5" GPIO LCD driver followed by Octoscreen installation.

after i reboot my pi it just stop at "pi@octopi:"

I did a 
sudo systemctl status octoscreen.service

and below is the result
`octoscreen.service - OctoScreen
   Loaded: loaded (/lib/systemd/system/octoscreen.service; enabled; vendor preset: enabled)
   Active: activating (start-post) since Fri 2020-03-06 11:32:21 GMT; 6s ago
 Main PID: 1323 (xinit); Control PID: 1352 (bash)
    Tasks: 29 (limit: 4497)
   Memory: 24.2M
   CGroup: /system.slice/octoscreen.service
           ├─1323 /usr/bin/xinit /usr/bin/OctoScreen -- :0 -nolisten tcp -nocursor
           ├─1324 /usr/lib/xorg/Xorg :0 -nolisten tcp -nocursor
           ├─1336 /usr/bin/OctoScreen
           ├─1352 /bin/bash /etc/octoscreen/disablescreenblank.sh 0
           └─1353 sleep 10

Mar 06 11:32:21 octopi xinit[1323]:         (++) from command line, (!!) notice, (II) informational,
Mar 06 11:32:21 octopi xinit[1323]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Mar 06 11:32:21 octopi xinit[1323]: (==) Log file: "/var/log/Xorg.0.log", Time: Fri Mar  6 11:32:21 2020
Mar 06 11:32:21 octopi xinit[1323]: (==) Using config directory: "/etc/X11/xorg.conf.d"
Mar 06 11:32:21 octopi xinit[1323]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Mar 06 11:32:22 octopi xinit[1323]: time="2020-03-06T11:32:22Z" level=info msg="OctoPrint's config file found: \"/home/pi/.octoprint/config.yaml\"" file=main.go
Mar 06 11:32:22 octopi xinit[1323]: time="2020-03-06T11:32:22Z" level=info msg="Using \"http://localhost:5000\" as server address" file=main.go func=main.init.0
Mar 06 11:32:22 octopi xinit[1323]: time="2020-03-06T11:32:22Z" level=info msg="Found API key at \"/home/pi/.octoprint/config.yaml\" file" file=main.go func=mai
Mar 06 11:32:22 octopi xinit[1323]: time="2020-03-06T11:32:22Z" level=debug msg="New background task started" file=common.go func="ui.(*BackgroundTask).Start" l
Mar 06 11:32:22 octopi xinit[1323]: time="2020-03-06T11:32:22Z" level=error msg="unexpected status code: 404" file=ui.go func="ui.(*UI).loadSettings" line=194`

I am using this LCD
http://www.lcdwiki.com/3.5inch_RPi_Display

can anyone help me please.

bolsoncerrado, 2020-03-06:
I came here after manually installing Octoscreen (via .deb package) and rebooting.

After reboot I also had to MANUALLY start Octoscreen for it to activate so...where's the magic to have it enabled after system is turned on / reboots?

Thanks! ;)

ssombra, 2020-03-06:
I have the same problem, did you find a solution?

daimon82, 2020-03-06:
edit file:

sudo nano /etc/rc.local

and insert this above the last line so it looks like:

systemctl start octoscreen
exit 0

jansenchua, 2020-03-07:
> edit file:
> 
> sudo nano /etc/rc.local
> 
> and insert this command above the last line so the last two lines looks like:
> 
> systemctl start octoscreen
> exit 0

Yup I already done this but doesn't work at all..

ssombra, 2020-03-07:
These instructions worked for me, follow them one by one;
https://github.com/Z-Bolt/OctoScreen/issues/43#issuecomment-546152208

synman, 2020-03-08:
the problem is the service [Install] block.

Octoscreen defaults to only starting up automatically in a graphical environment.  If you add multi-user as a target it will start up fine for you use case.

```
[Install]
WantedBy=graphical.target multi-user.target
```

bolsoncerrado, 2020-03-08:
Where does that last line suits onto?

bolsoncerrado, 2020-03-10:
> edit file:
> 
> sudo nano /etc/rc.local
> 
> and insert this command above the last line so the last two lines looks like:
> 
> systemctl start octoscreen
> exit 0

This did NOT work for me but instead:

service octoscreen start
exit 0

did the trick.

Is it a wrong way of doing it?

synman, 2020-03-10:
> Where does that last line suits onto?

It's your octoscreen.service file that is used to configure octoscreen as a startup service 

You'll want to want to ensure the service gets reconfigured properly after making the change.

synman, 2020-03-10:
> > edit file:
> > sudo nano /etc/rc.local
> > and insert this command above the last line so the last two lines looks like:
> > systemctl start octoscreen
> > exit 0
> 
> This did NOT work for me but instead:
> 
> service octoscreen start
> exit 0
> 
> did the trick.
> 
> Is it a wrong way of doing it?

In the end do what works for you, but yes, in a purest sense you did it wrong :)

bolsoncerrado, 2020-03-10:
So the correct way would be to edit as you said but first editing octoscreen.service ? 

where can I find that one?

Thanks!!

synman, 2020-03-10:
https://www.digitalocean.com/community/tutorials/how-to-use-systemctl-to-manage-systemd-services-and-units

bolsoncerrado, 2020-03-10:
Thanks @synman

bolsoncerrado, 2020-03-10:
BTW does anyone know where is the TEMPERATURE pre-defined in octoscreen? why is it set to 100?

imlabor, 2020-03-19:
I build it up on this way [modelllabor](https://github.com/imlabor/3d-printing/wiki/Ender-3-Pro---Octoprint-with-LCD-Display)

ArcAIN6, 2021-01-14:
OMG thank you <3

JeffB42, 2021-02-04:
@jansenchua is this still an issue?

closed 2021-03-27 by JeffB42