Z-Bolt/OctoScreen #101: Installation failing...

SoBaldrick, 2020-03-08:
failing with

Get:91 http://raspbian.mirror.uk.sargasso.net/raspbian stretch/main armhf libtxc-dxtn-s2tc armhf 1.0+git20151227-2 [44.7 kB]
Fetched 53.4 MB in 14s (3,613 kB/s)
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/c/cups/libcups2_2.2.1-8+deb9u2_armhf.deb  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

looking at the source the file name has changed from 
libcups2_2.2.1-8+deb9u2_armhf.deb 
to
libcups2_2.2.1-8+deb9u5_armhf.deb

evilbobbins, 2020-03-08:
I fixed the same error this morning on a fresh install with running a sudo apt-get update and then running a sudo apt install --fix-missing

marshalab, 2020-03-16:
Hello.
my rpi3 with LCD-35A on the latest buster OS

pi@RPiLCD35:~/octoscreen $  sudo dpkg -i octoscreen_2.5-1_armhf.deb
(Reading database ... 98366 files and directories currently installed.)
Preparing to unpack octoscreen_2.5-1_armhf.deb ...
Unpacking octoscreen (2.5-1) over (2.5-1) ...
Setting up octoscreen (2.5-1) ...
Installing service /lib/systemd/system/lightdm.service
Failed to preset unit: File /etc/systemd/system/display-manager.service already exists and is a symlink to /lib/systemd/system/lightdm.service.
/usr/bin/deb-systemd-helper: error: systemctl preset failed on octoscreen.service: No such file or directory
Job for octoscreen.service failed because the control process exited with error code.
See "systemctl status octoscreen.service" and "journalctl -xe" for details.

That i can a do?

rcsez, 2020-04-30:
I'm getting the same issue as Marshalab. When I install Octoscreen I get a popup asking if I want lightdm or octoscreen to control x server. I chose octoscreen and it led to this error. 

Should we be choosing lightdm instead?

JeffB42, 2021-02-04:
@SoBaldrick  is this still an issue?

marshalab, 2021-02-04:
no, I haven't found a solution.

JeffB42, 2021-02-05:
Install steps are listed at https://github.com/Z-Bolt/OctoScreen/wiki/Setting-Up-OctoScreen-and-Your-Display and https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen. Look these pages over and see if any of the information there helps.  If not, try installing fresh (meaning, install on a new SD card, with a new/latest image of OctoPi, follow the steps in the previous guide I listed, and also install 2.6.1).  If you still run into issues, turn on logging, set the log level to debug, then look at the log file and look for the first error listed.

JeffB42, 2021-02-05:
Wait, I just read this:

>...on the latest buster OS

Only OctoPi is supported.  Please install from an official OctoPi image, https://octoprint.org/download/

marshalab, 2021-02-05:
Here's how!  .... I'll be setting up another printer soon, I'll try it through octopi.  So much time has passed, it may start.

closed 2021-03-27 by JeffB42