Z-Bolt/OctoScreen #102: Problem with 3,5'' TFT v3.0 SPI 480x320

Meshchankin, 2020-03-10:
Hi

Thank you for this program but unfortunate I can't install it to my PI3 with  3,5'' TFT v3.0 SPI with 480x320 resolution. I see in the description that I should use resolution 800x533 in /boot/config.txt and /etc/octoscreen/config but it doesent help :( 
Resolution 480x320 also work incorrect. 

If I use:
sudo nano /boot/config.txt

==> Add this line to the end of the file
framebuffer_width=800
framebuffer_height=533
hdmi_cvt 800 533 60 6 0 0 0

Then:
sudo nano /etc/octoscreen/config

==>Edit the following line to show this
OCTOSCREEN_RESOLUTION=800x533

Now I see this 
[image]

florin03, 2020-03-12:
Hello,

What driver are you using for that LCD? I am currently using LCD-show-170703 and i get some issues when installing  OctoScreen.

Thx

Stady234, 2020-03-16:
@Meshchankin In my config txt i only change the 
`hdmi_cvt 800 533 60 6 0 0 0
`
and the only thing i added to the octoscreen config was my API key...
--
Can you try those mods and let me know...
--
My issue on mine is that the service does not auto start... i am trying to figure out what i did!  lol

ChaosBlades, 2020-03-18:
I have the same issue on Adafruit PiTFT Plus 480x320 3.5" touchscreen. I have been all through the issues on here and can't crack this one.

fruvous, 2020-03-19:
Instructions are https://github.com/Z-Bolt/OctoScreen/issues/43

Does anyone have the touch coordinate settings?  I got it displaying right but my touch doesn't match.

ChaosBlades, 2020-03-19:
I had that issue as well. If you have an Adafruit like I do you need to follow their instructions.

https://learn.adafruit.com/adafruit-pitft-3-dot-5-touch-screen-for-raspberry-pi/touchscreen-install-and-calibrate

Also found this.

"
My PiTFT's rotation/calibration isn't working in X11

X11 (the graphical system) has changed how it gets touchscreen input, so if you rotate the display and the calibration isn't being picked up:

Check /usr/share/X11/xorg.conf.d for a file called 10-evdev.conf 

If you don't see that file

    You need to sudo apt-get install xserver-xorg-input-evdev, and then...
    If you do have a 40-libinput.conf in that same directory, you must remove it even if/once evdev is installed, since it will override the 10-evdev.conf otherwise.
"

fruvous, 2020-03-19:
I have the Elecrow one and it needs calibration.  Thats the issue is that Octoscreen doesn't run the calibration and I don't really want to bother installing X to run it.  Was hoping to steal the coordinates that other people had in their config file so I can just manully enter it.

Meshchankin, 2020-03-19:
Here is topic and my solve 
https://3dua.info/topic/134-ustanovka-octoscreenoctoprint/?page=2&tab=comments#comment-1304

ChaosBlades, 2020-03-20:
@Meshchankin You might have a second "hdmi_cvt 800 533 60 6 0 0 0" at the end of your /boot/config.txt file. 

@fruvous Somehow I managed to fix the touch coordinates without doing what I am about to explain and just doing the X11 thing in my last comment. But for whatever reason I can't get it to work anymore, so I went on an adventure.

You need to run xtcal, you can't run xtcal because it won't run on OctoPi or OctoScreen. I had to get a second SD Card and install full Raspbian and follow these steps.

Install Raspbian
Install Touchscreen Drivers

You should now see the Raspbian Desktop on your touchscreen.

sudo apt-get install libxaw7-dev libxxf86vm-dev libxaw7-dev libxft-dev
git clone https://github.com/KurtJacobson/xtcal
cd xtcal
make

Clear current touch coordinates.
DISPLAY=:0.0 xinput set-prop "stmpe-ts" 'Coordinate Transformation Matrix' 1 0 0 0 1 0 0 0 1

Run xtcal and touch the + on the screen
DISPLAY=:0.0 ~/xtcal/xtcal -geometry 720x480

On the Raspbian install it will autosave so you can reboot the Raspbian install and test the coordinates out before you switch over and apply them to octoscreen.

Take the coordinates it gives you and put them into this file on your OctoScreen install.
sudo nano /usr/share/X11/xorg.conf.d/20-calibration.conf

```
Section "InputClass"
        Identifier "STMPE Touchscreen Calibration"
        MatchProduct "stmpe"
        MatchDevicePath "/dev/input/event*"
        Driver "libinput"
        Option "TransformationMatrix" "-0.007882 -1.094303 1.033443 1.132679 0.078381 -0.170031 0 0 1"
EndSection
```
reboot

If you are using a 2.4"/2.8"/3.2" 320x240 display with landscape orientation, use 640x480. If you're in portrait, use 480x640.

If you are using a 3.5" display with landscape, use 720x480, portrait is 480x720

**Save the coordinates in case you need to reinstall OctoPi!**

jaburges, 2020-03-26:
Hi, having this same issue (well the original resolution issue not the touchscreen calibration off shoot)

None of these work :(
800x600
800x533
480x240
320x240

any thoughts on how to scale the screen appropriately for a 3.5" GPIO touch screen?

jshessen, 2020-03-26:
I spent quite a few hours attempting to get this issue resolved last night as well.  I am using a "Kuman 3.5 TFT" [https://www.amazon.com/gp/product/B01CNJVG8K](url) which seems to be a pretty limited resolution (480x320).

Using the hdmi_cvt settings I can force the device to honor the resolution settings, but when OctoScreen kicks in the splash octopus and dashboard seem to push beyond the requested limits.

I eventually swapped back to a version of OctoPrint-TFT [https://github.com/darksid3r/OctoPrint-TFT](url). To validate that this was indeed showing the correct setting when configured for 480x320.  but when using OctoScreen it still does not display correctly.

`OCTOSCREEN_RESOLUTION=480x320
...
#dtoverlay=vc4-fkms-v3d
hdmi_force_hotplug=1
dtparam=i2c_arm=on
dtparam=spi=on
enable_uart=1
dtoverlay=tft35a:rotate=90`

An additional note, I used the adafruit-pitft.sh script to ensure that I was getting things setup and that FBCP support was added.
I also used the driver setup from [https://github.com/goodtft/LCD-show.git](url).  It appears that the FBCP is removed by the second script since I am using the GPIO script.
[image]
[image]

jaburges, 2020-03-30:
Still couldn’t get it to run in the correct resolution. Using a 4 inch screen works perfectly.

floridaservices, 2020-03-30:
I have the same problem with a kedei 480x320 3.5" screen.  Changing resolution settings to recommended only makes it bigger.

wesselah, 2020-04-05:
Had the same problems as everybody here. This solved my problem.
https://community.octoprint.org/t/octoscreen-a-new-software-to-use-octoprint-with-lcd/10629/20
This driver emulates the correct screen resolution in software. Works like a dream

erotikbaer, 2020-09-13:
hi, do you have some news for users with an 3.5inch LCD Touchscreen(480x320). 
the behavior is the same with my 3.5inch display. all graphical tools are ok but octoscreen seems to have a problem with this resolution.
my /boot/config.txt:
[all]
hdmi_force_hotplug=1
dtparam=i2c_arm=on
enable_uart=1
dtoverlay=tft35a:rotate=270
hdmi_group=2
hdmi_mode=87
disable_overscan=1
framebuffer_width=800
framebuffer_height=533
dtparam=spi=on
hdmi_cvt 800 533 60 6 0 0 0

and my octoscreen config:
OCTOSCREEN_RESOLUTION=800x533

but octoscreen appears is larger than the display

regards christian

JeffB42, 2020-12-04:
@Meshchankin Could you follow the steps listed in the wiki (located at https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen) and then post a reply to this issue whether the steps listed work and the issue is resolved?  I'd like to start closing out all the display-related issues.

If it doesn't work,
1) make sure you use a Raspberry Pi 3B or Raspberry Pi 4
2) start with a fresh install of OctoPi
3) please take a picture of the screen

Thanks, 

Jeff

JeffB42, 2021-01-02:
@Meshchankin is this still an issue?

karl0ss, 2021-01-10:
I followed the wiki this morning, the only thing I did different was install the latest octoscreen but I have this..
[image]

JeffB42, 2021-01-31:
@karl0ss any luck?  I'm running out of ideas to suggest.  It should run on a RPi3a, but I haven't tried it.  The only things that come to mind are:
* Buy a Raspberry Pi 4
* Start with a fresh install of OctoPi, and re-flash you SD card
* Buy a Hyperpixel screen (the 4" rectangular one, with touch screen)

I wish i could think of something else but I can't think of anything else to try.

Jeff

draycon88, 2021-02-07:
This Video helps me !

https://www.youtube.com/watch?v=brtrtj1rq7g

I had no other solving and tried it...and it works now!

closed 2021-03-27 by JeffB42