Z-Bolt/OctoScreen #103: Swap axis direction, temp settings and f/r

samd12, 2020-03-10:
Great plugin!  I have a few quick questions.
Is there a way to reverse the direction the bed moves in regard to the up and down arrows?  I have it inverted in ocrtopi but from Octoscreen it is reversed.
Also where do you adjust the temps for preset heat settings?  
Is there info on adjusting settings like this that I am missing?

And last a feature request:
A panel that can be tied to raspberry pi gpio pins so you can control devices with a high/low signal.  So you can turn led's, fans and lights on and off if you have them setup especially with an enclosure
Thanks!!!

Rshep67, 2020-03-14:
I have the same issue on my CR-10S Y is reversed on the touch screen.....

Awesome Job guys cant say it enough

MaikelNL, 2020-05-17:
i'm having the same issues in octoscreen. for octoprint onscreen control axis and invert ar working like a charm. but in octoscreen de functions are reversed.

JeffB42, 2021-02-04:
@samd12 @Rshep67 and @MaikelNL if you install the plug in, yu will be able to invert the X, Y, or Z axis.

closed 2021-03-27 by JeffB42