Z-Bolt/OctoScreen #104: Screen shifted

samd12, 2020-03-11:
I have a 7" pi screen but the image is shifted.  Any ideas?  See image
[image]

fidel3d, 2020-03-14:
[image]
I have the same issue, but it is only on the printing screen. The rest of the screens are fine as far as I can see. 
Here is my setup:
- Raspberry PI 3 A
- Latest on Octoprint and Raspian
- HyperPixel 4 rectangular

Thanks,

fidel3d, 2020-03-14:
Hi,

I guess this fixed the issue with the printing screen shifting to the right.

https://github.com/mkutscher/OctoScreen/releases

Thanks,

HaMu56, 2020-03-20:
Hello from Germany
and sorry if my englisch is not the best
I have a PI4 with the newest octoprint image. all works fine .
now i want to install octoscreen to use a 7inch touchdisplay with a resolution of 1024 x 600. input HDMI and USB for power and touchsignals for the pi4. that all works.
octoscreen is 4.6.1
I tried many resolutions, but on the right the screen is cropped, sometimes more, sometimes less. for this I changed the /boot/config.txt and / etc/octoscreen/config. the PI 4 is set to 720x480 (16: 9) with mode 3.

can you please help me or is the matter still under construction?

THX
Hardy 

[octoscreen.txt](https://github.com/Z-Bolt/OctoScreen/files/4359254/octoscreen.txt)
[image]
[image]
[image]

fidel3d, 2020-03-22:
Hi Hardy,

Perhaps, I was luck, but one think that I did was to make an image of sd so if I break it I can always go back to it which I did because of the setting, plugins, updates and upgrade in case  I broke something after I tried all the "helps" from difference sources.

Then went back and I strictly follow the instruction from the manufacture (HyperPixel 4 rectangular) and then I tried only the installation from the git, but I used subjection from the url  https://github.com/mkutscher/OctoScreen/releases

so it worked for me so may you should try again.

Good Luck,

Thanks,

rwaspe, 2020-05-24:
It may be the length of your filenames. I posted a fix here: https://github.com/Z-Bolt/OctoScreen/issues/126

3Dietrich, 2020-09-26:
Hi all, especially  @HaMu56  
I have **exatly** the same problem. I tried several hours to get the right resolution - no success, although everything was clear: [the display](https://smile.amazon.de/gp/product/B07QKT6L58) has 1024x600, even the pi config setting was given 
hdmi_group = 2
hdmi_mode = 87
hdmi_cvt 1024 600 60 6 0 0 0
Setting the OCTOSCREEN_RESOLUTION=1024x600 in /etc/octoscreen/config
resulted in totaly wrong resolution (see att.pictures)
I couldnt find a fitting resolution, although I tried a lot. And _NO, I did not try to load any (longed named) files_ (I didn't came to the point I want to use it, the files also seem 2b empty, but thats a different topic). No. It's just the display size.
Here are some results of OCTOSCREEN_RESOLUTION tests:
[image]
[image]

[image]

[image]
[image]
[image]
[image]
[image]
[image]

[image]
[image]

JeffB42, 2020-09-27:
Hi @3Dietrich , thanks for posting the details and the pictures, and thanks for trying different resolutions.

The screen shift issue that's been reported most of the time is when a person is running OctoSceen on a small display that's only 480x320.  What's happening, is that OctoScreen uses a Linux GUI library called, GTK.  When OctoScreen runs, it creates a main/central window object, and that window object takes up the entire screen.

When a control or text is displayed and there isn't enough to display it, GTK automatically resizes the window.  If one was running on a desktop, this wouldn't be an issue, but since this "window" takes up the entiure screen, it's expanded past the size of the screen.  That's why the screen appear to be "shifted" and "offscreen".

With your case though, you have a larger display, and everything should work.  I don't have a 1024x600 display myself to test this on, but I can run OctoScreen locally on my desktop computer, and I verified that when OCTOSCREEN_RESOLUTION is set to 1024x600, it runs fine and window doesn't expand.

I don't know if this will work, but try this:
First, comment this out
```
hdmi_cvt 1024 600 60 6 0 0 0
```

I suspect (though I'm not positive) that hdmi_cvt is used for HDMI displays that don't have enough pixels, and a soft scale is applied.  (and this only works with screens which use HDMI)

Thy that, and that does't work, also try commenting out 
```
hdmi_group = 2
hdmi_mode = 87
```
...and see if that works.  In my config, I don't have any of those set.  (I'm also running 800x480)

Next, watch Chris Riley's video on setting up OctoScreen with a HyperPixel display - https://www.youtube.com/watch?v=OJ59hXSyBoI&t=72s.  It's specifically addresses the HyperPixel, but there is quite a lot of good tips & suggestions in it, and I think they might be applicable to your screen.

Please try these out and report back if it works (and what you did)

3Dietrich, 2020-09-27:
hm.. 
as I [read in raspi docs](https://www.raspberrypi.org/documentation/configuration/config-txt/video.md), the lines
```
hdmi_group = 2
hdmi_mode = 87
hdmi_cvt = 1024 600 60 6 0 0 0
```
belong together, the `hdmi_mode = 87` (in short) doesn't exist, but signs that `hdmi_cvt` entry/numbers are expected.
btw: in last post there is a "=" missing after `_cvt`, also in [amazons product description](https://smile.amazon.de/gp/product/B07QKT6L58) (of course I used the correction).

However, i tried your suggestion:

1. removed only `hdmi_cvt`, no change.
2. removed all 3 `hdmi...`, (actually I started with this setting yesterday): no change

Then I watched [the video](https://www.youtube.com/watch?v=OJ59hXSyBoI). But I couldn't find any help for me. I did not need to install "hyperpixel4" or rotate my screen, in fact the screen worked directly out of the box watching Debian boot-list and terminal (..it's made for raspi). I started to edit all this stuff only to get Octoscreen working.
The rest of the video he explains [the same is written here](https://github.com/Z-Bolt/OctoScreen/tree/master)..

At the end, on my original Raspi Display with crazy 1024x600 pixel, Octoscreen still works "_best_" with 400x400 :(
At least I can use almost all screens, only the "Network information" has no exit:
[image]

I will let you know when I have found a solution

JeffB42, 2020-09-27:
@3Dietrich thanks for investigating.  I don't have a solution/fix yet, but here are some comments:

First, your screen resolution: It's strange... you really shouldn't need to set the resolution down to 400x400, and I'm not sure why you are seeing the window resizing when you run at 1024x600.  I suspect there is a config not set correctly in your system.  When I run (in my development environment) and set OCTOSCREEN_RESOLUTION to 1024x600 it's running fine, and it (the window) never expands.  I suspect that even though your screen's pixels are 1024x600, there's a configuration setting somewhere that's configuring the HDMI to scale down to something smaller (so, even though yo have a 1024x600, the screen might only be set to something like 480x320, and then the 480x320 window is scaled up to 1024x600 on the hardware... not sure though, it's just a hunch)

Second, I was able to reproduce your issue with the Network screen, and the issue is that 400x400 pixels just isn't enough to display that screen.  I switched over the the 2.5.1 codebase (just to make sure and compare "apples-to-apples"), set the resolution to 400x400 and then ran.  Here's what I observed:

It starts up in 400x400:
[image]

...but that's not enough pixels to display the Idle screen, and the window is immediately expanded to 584x400:
[image]

I then navigated to the Network screen, and the window was expanded to 714x400:
[image]

I did some image editing, and judging from the screenshot you last posted, it looks like your display is running at (roughly) 594x297:
[image]

594x297 is almost 600x300, and my selection was just an approximate... I suspect that somewhere, something is setting the screen to 600x300, or is scaling the screen by 60% x 50%.

JeffB42, 2020-12-04:
@samd12 Could you follow the steps listed in the wiki (located at https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen) and then post a reply to this issue whether the steps listed work and the issue is resolved?  I'd like to start closing out all the display-related issues.

If it doesn't work,
1) make sure you use a Raspberry Pi 3B or Raspberry Pi 4
2) start with a fresh install of OctoPi
3) please take a picture of the screen

Thanks, 

Jeff

JeffB42, 2021-01-02:
@samd12 is this still an issue?

JeffB42, 2021-02-04:
Closed due to inactivity.

closed 2021-02-04 by JeffB42