Z-Bolt/OctoScreen #105: Incorrect print percentage displayed

SupremeCookie11, 2020-03-13:
during a print, example my 16hr print, it shows 1% completed after an hour has passed. octoprint itself, shows the correct percentage completed. where does this random 1% come from? it does show the time remaining correctly

RH-Dreambox, 2020-03-14:
Yes, same problem here. MK3 with Bondtech firmware.
[image]

JeffB42, 2021-02-04:
@SupremeCookie11 When you print, open a browser window and log into your OctoPrint server (typically http://octopi.local).   Compare what OctoScreen displays with what the OctoPrint web page display.  I've seen this as well, but it's b/c the percentage in OctoPrint doesn't match what the actual printer displays.

When you print, if the percentage in OctoScreen matches OctoPrint, then there's nothing that can be done.  On the other hand, if the percentage in OctoPrint matches the display on your printer, but OctoScreen is displaying something else, it's a bug.

Please let me know what you find.

Jeff

closed 2021-03-27 by JeffB42