Z-Bolt/OctoScreen #108: Octoscreen freezes on LCD ...

red-line-rider, 2020-03-16:
Raspi 3B+
TFT LCD 3.5"

Installed Octoscreen according to instructions, wanted to avoid the hassle of login issues on TouchUI. First of all it takes a hell of a long time for the LCD screen to startup and then it freezes on this screen:
[image]

Tried mannually sarting the service 
`pi@octopi:~ $ sudo service octoscreen start`

&
` sudo systemctl start octoscreen`

to no avail. And i did not see this issue in the known issues tab.

Thanks.

1f7, 2020-04-02:
Hi @red-line-rider,
3.5inch RPi LCD (A) work with octoscreen without  problem.
read this first post https://community.z-bolt.ru/t/octoscreen-na-3-5inch-rpi-lcd-a/18

JeffB42, 2021-02-04:
@red-line-rider is this still an issue?

closed 2021-03-27 by JeffB42