Z-Bolt/OctoScreen #111: Restarting all 5 seconds (Waveshare 5" HDMI display)

McClean0815, 2020-03-19:
Hi,
i like the octoscreen interface, nice look and good to use. I had some problems installing driver and get everything to run. It worked the first time until reboot.
Now octoscreen restarts (the hole x system?) every 5 seconds. In this time i see the ui and im possible to use the touch interface.
Im not really familar with linux, what can i do (post error logs)?

I hope its possible to fix this, octoscreen is nice ;=)

Steffen

eggerc, 2020-03-19:
Same here since the latest octoprint update

McPrince96, 2020-03-22:
It's a problem that has been reported and solved in many issues here. Just go to issues and type "restart" in the search bar

Lexorius, 2020-03-28:
is /usr/bin/xset  available ? 
if not please try to install it   apt-get install x11-xserver-utils

McClean0815, 2020-03-28:
Hi Lexorius,

yes you are right! Thank you very much. Now octoscreen keeps running

So this bug is fixed - just a mistake from me...

closed 2020-03-28 by McClean0815