Z-Bolt/OctoScreen #113: Custom controls with multiline commands not showing

ashleysommer, 2020-03-30:
In octoprint config you can specify [custom controls](https://docs.octoprint.org/en/master/features/custom_controls.html#controls) with multiple commands executed sequentially. Like this:
```
- name: MyControl
  commands:
  - G91
  - "G1 Z1.00"
  - G90
```
These show in the Octoprint UI, but do not show in Octoscreen.

Custom controls with a single command _do_ appear, eg:
```
- name: MyControl
  command: "G1 Z1.00"
```
This does show in Octoscreen.

nicedevil007, 2020-05-07:
Hey @ashleysommer 

I know it doesn't helps you with your found bug but I'm looking forward to adjust the UI to my needs and I can't find out how to get those things into the screen.

I'm able to add things to the octoprint controls menu. What I have to do next?
just add those lines with "MyControl" name in it?

I can't find out which of those name / panel is for the name I choose in the config.yml.

```
{
  "name": "Home",
  "icon": "home",
  "panel": "home"
}
```

JeffB42, 2021-03-27:
This was fixed in the 2.7.0 release.

closed 2021-03-27 by JeffB42