Z-Bolt/OctoScreen #114: OctoScreen crashes OctoPrint when the printer is turned off

justinmbrock, 2020-03-30:
Re: issues #10, #56, and #58. OctoScreen attempts to reconnect to the printer every 5 seconds. This causes OctoPrint to crash after several hours, requiring a reboot to function again. This may be due to OctoPrint not closing connections, but it is definitely OctoSCREEN causing it.

jchaltinner, 2020-04-03:
Yes, thank you. I have 3 Pi's running OctorPrint, and only one with OctoScreen, and only the install with OctoScreen has this issue. I do love OctoScreen though!

wesselah, 2020-04-05:
Really love ocroscreen but i have excactly the same problem octoprint stops working when running octoscreen and printer is shutdown

wesselah, 2020-04-05:
Maybe disable autoconnect which causes memory drain and make it a manual setting on the main screen when not connected that way the memory drain could go away. Or make it a config setting so people can choose if they want autoconnect

justinmbrock, 2020-04-05:
...and it happened again. The printer was off for maybe half an hour and OctoPrint is not responding because OctoScreen caused it to crash. On a RPi 4.

justinmbrock, 2020-04-06:
I disabled this crap. Between this, the UI resetting every 5 minutes, the UI scaling itself so buttons are off the screen, and a developer that doesn't think these are problems. I'm trying OctoDash instead.

floridaservices, 2020-04-06:
> I disabled this crap. Between this, the UI resetting every 5 minutes, the UI scaling itself so buttons are off the screen, and a developer that doesn't think these are problems. I'm trying OctoDash instead.

It's really a shame I had high hopes for OctoScreen it looks beautiful but despite what is claimed it does not work so well with <4" screens it seems.  Add in the rest of the nonsensical errors and it's not worth the trouble.  Thanks for the tip on OctoDash.

justinmbrock, 2020-04-06:
Welcome! I'm trying it out now. It's really new, but the dev seems responsive. I suggest using a stylus if you have a resistive touchscreen.

floridaservices, 2020-04-08:
> Welcome! I'm trying it out now. It looks new, and the dev seems responsive. I suggest using a stylus if you have a resistive touchscreen.

Thanks again, I have been looking for a stable, intuitive interface for my pi screen for a while now to replace TouchUI and OctoDash is looking to be amazing so far.

ddt3, 2020-04-16:
Came here to report exactly this.. Liked octoscreen because it allows me to control the printer without an additional computer. This bug kills that idea as I need to reboot the raspberry pi..

roessi, 2020-04-22:
same Problem here, so this i go with ddt3 This bug kills that idea as I need to reboot the raspberry pi..

NiicOw974, 2020-05-06:
### The problem isn't corrected with the new version of octoscreen

**I have the same issue with octoscreen 2.5.1, when printer is powered off :** 

_2020-05-06 07:17:20,429 - tornado.access - WARNING - 404 POST /api/plugin/zbolt_octoscreen (127.0.0.1) 15.73ms
2020-05-06 07:17:20,452 - tornado.access - WARNING - 404 POST /api/plugin/zbolt_octoscreen (127.0.0.1) 17.28ms
2020-05-06 07:17:20,714 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
2020-05-06 07:17:20,752 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial port" to "Error: Failed to autodetect serial port, please set it manually."
2020-05-06 07:17:22,713 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
2020-05-06 07:17:22,764 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial port" to "Error: Failed to autodetect serial port, please set it manually."
2020-05-06 07:17:24,671 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
2020-05-06 07:17:24,723 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial port" to "Error: Failed to autodetect serial port, please set it manually."
2020-05-06 07:17:26,632 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
2020-05-06 07:17:26,669 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial port" to "Error: Failed to autodetect serial port, please set it manually."
2020-05-06 07:17:28,632 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
2020-05-06 07:17:28,669 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial port" to "Error: Failed to autodetect serial port, please set it manually."
2020-05-06 07:17:30,700 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
2020-05-06 07:17:30,737 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial port" to "Error: Failed to autodetect serial port, please set it manually."
2020-05-06 07:17:32,709 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
2020-05-06 07:17:32,751 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial port" to "Error: Failed to autodetect serial port, please set it manually."
2020-05-06 07:17:34,676 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
2020-05-06 07:17:34,715 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial port" to "Error: Failed to autodetect serial port, please set it manually."
2020-05-06 07:17:36,705 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
2020-05-06 07:17:36,753 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial port" to "Error: Failed to autodetect serial port, please set it manually."
2020-05-06 07:17:38,647 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
2020-05-06 07:17:38,690 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial port" to "Error: Failed to autodetect serial port, please set it manually."
2020-05-06 07:17:50,418 - tornado.access - WARNING - 404 POST /api/plugin/zbolt_octoscreen (127.0.0.1) 15.91ms
2020-05-06 07:17:50,442 - tornado.access - WARNING - 404 POST /api/plugin/zbolt_octoscreen (127.0.0.1) 17.80ms_

**And after a few minutes :**

 _- tornado.general - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 784, in _server_request_loop
    ret = yield conn.read_response(request_delegate)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/gen.py", line 1055, in run
    value = future.result()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/concurrent.py", line 238, in result
    raise_exc_info(self._exc_info)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/gen.py", line 307, in wrapper
    yielded = next(result)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/http1connection.py", line 159, in _read_message
    max_bytes=self.params.max_header_size)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/iostream.py", line 256, in read_until_regex
    self._try_inline_read()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/iostream.py", line 711, in _try_inline_read
    pos = self._read_to_buffer_loop()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/iostream.py", line 625, in _read_to_buffer_loop
    if self._read_to_buffer() == 0:
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/iostream.py", line 738, in _read_to_buffer
    chunk = self.read_from_fd()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/iostream.py", line 1052, in read_from_fd
    chunk = self.socket.recv(self.read_chunk_size)
MemoryError_

**Octoscreen is the best interface for me but this problem makes octoprint lose all its usefulness**

OevreFlataeker, 2020-09-04:
See here for another discussion: https://community.octoprint.org/t/octoscreen-causing-octoprint-to-build-up-excessive-thread-count-when-printer-is-offline/23732

The problem is no longer occuring in my version after the fix. However the only thing really needed seemed to be the additional call of sdNotify() to satisfy the systemd watchdog timer? Apart from that all the code seems to be present in the master branch and also in the dev branch. Could a dev please clarify if that is a correct observation? At least in fact I did not have to change anything really (tested with Go 1.15.1)

ArielTulian, 2020-12-26:
Good afternoon, Merry Christmas! query, they were able to solve the issue of having the printer disconnected with the server running, the OctoScreen screen is continuously asking to initialize, after a while it turns off the OctoPrint server. Is there a way to eliminate or, failing that, make the connection manual? Install version 2.5.1 but it does not correct the problem, the LCD is generic 3.5 "

Sorry for my english from google translator

JeffB42, 2020-12-26:
HELLO @ArielTulian I'll try to help if I can.  I tried to reproduce the issue, but I wasn't able to reproduce it.  This was using a Prusa i3 printer, a Raspberry Pi 4 (w/2GB memory), OctoPrint 1.5.2, my Hyperpixel screen, and I tried with both the last released version of OctoScreen (2.5.1), as well as the version I'm currently working on (2.6) but wasn't able to reproduce the issue.  The steps I took were:

1. Started my system.  I have everything on a single power strip, and turning the power strip on turns both my printer as well my Raspberry Pi on at the same time.
2. Waited for OctoPrint and OctoScreen to finish starting up.
3. After everything started up and connected, I then turned off my printer by disconnecting the printer's power supply. 

At this point, OctoScreen changed (the buttons were no longer visible) and went to its connecting screen, stating that the printer was disconnected.

4. I waited a few minutes
5. turned the printer back on
6. and then reconnected.

...and was able to successfully reconnect.

A comment to something you wrote:
>OctoScreen screen is continuously asking to initialize, 
This is expected, kinda.  When disconnected, OctoScreen pings the OctoPrint server for the printer's status, and it's continuously trying to reconnect.  It shouldn't be trying to initialize anything though.

> after a while it turns off the OctoPrint server
This is not expected.  I tried to reproduce this but wasn't able to.

I have a few questions for you:
1. could you list what hardware you are using, especially which Raspberry Pi
2. could you provide step-by-step instructions on the precise steps you are performing?  (please detail exactly what you are doing)  I'll try performing the same steps on my end and see if I can reproduce the issue.  Perhaps the steps I listed above (steps 1-6) are different from what you are doing.
4. GitHub now is able to upload video (mp4 and mov).  Could you record a video and upload it?

ArielTulian, 2020-12-26:
Hi @JeffB42, thanks for responding so quickly. I calculate that it is because of having to have a generic screen and it is different than the Hyperpixel.
1- My hardware is a Rasberry Pi 3b + (with 2GB of RAM), a generic 3.5-inch touchscreen by GPIO. Firmware is Octopi 0.17, Octoscreen 2.5.1 (Previous versions and 2.6.1 of another user were tested)
2- Steps I take:
        * Start raspberry without connecting the printer.
        * After 10 minutes, sometimes 30 minutes trying to connect to Octoprint, it shuts down when the memory overflows.
If I turn on the printer it works perfectly, it works very well. But when I disconnect the USB from the rasp connected to the printer or I turn it off, the screen says initializing and as it cannot, it resets and does the same thing again until I calculate the memory and turn off the Octoprint server.
The same thing that is commented above in this post.
I tried other versions around, 2.6.1 but it does exactly the same.
Now I am reinstalling everything again and when finished I record the video. As I need to turn it off and on remotely the printer, it is difficult for me to have OctoScreen installed when I download the server.

A valid solution would be to be able to connect octoscreen if I am near the printer and I already know that it is turned on so that it does not saturate with questions to the rasp.

Thank you very much for answering.

JeffB42, 2020-12-27:
@ArielTulian I was able to reproduce the issue.  Sort of.  I used a Raspberry Pi 3B, and performed the following:
1. With the printer off, I turned on the Raspberry Pi
2. After OctoPrint and OctroScreen started, OctoScreen appeared on the display, and then displayed that it wasn't able to connect to the printer
3. After a minute or so, OcrtoScreen appeared to restart.  It only appeared that OctoScreen restarted, and not the entire machine. It seemed to me that only OctoScreen restarted (and not the entire machine) because I witnessed the display change from OctoScreen, to the command line, and then back to OcrtoScreen, and it was pretty quick.  On my RPi 3, when it starts up it takes a while (and many commands are executed) and I didn't see that after the initial boot, so I suspect it's only OctoScreen that crashes and restarts, and not the entire system.
4. I let OctoScreen run and then reboot, over and over, for 30 minutes
5. After around 30 minutes, I then turned on my printer
6. After turning on the printer, OctoScreen was then able to successfully connect, and everything worked.

@ArielTulian if I understood you correctly, you are seeing the same as me for steps 1-5, but on step 6, OctoScreen does not connect to the server.  (a video would help out here...)

OK, so this is not great, and I'll look into this in the future, but I have to mention...

1) There's a work around for you, and that is, leave your printer on.  This isn't ideal, I know, but it is a work around.  From your previous post, you said that if the printer is on and OctoPi, OctoPrint, and OctoSceen start, it works fine, as expected.  For the moment, I suggest doing this.

The printer doesn't have to have its nozzle or bed turned on, so it should probably be OK to just turn the printer on and have everything be idle.  Or did I miss something here, and that's not possible?  Sure, the printer will consume some electricity being on and idle, but so will the RPi and the display, and if concerned about wasting electricity, why not have everything off?  Why have the RPi and OctoPrint running, but not the printer?

2) If you are seeing what I'm seeing (steps 1-6 above), and after 30 minutes can turn on your printer and everything connects, I suppose this is another temporary work around.  Rebooting over and over until the printer is turned on isn't great, but the machine (the RPi/OctoPrint server) should only be used for OctoPrint (and nothing else) so having OctoScreen reboot over and over until the print is turned on shouldn't be hurting anything.  Not great, I know, but shouldn't be hurting anything (unless I'm missing something here)

3) What is the use case for having OctoPrint and OctoScreen running, but having the printer turned off?  Personally, for me, when I have one running, I have all of them running, and I was wondering, why are you running OctoPrint and OctoScreen, but don't have the printer turned on?  This isn't to say that there isn't an issue here, or that it shouldn't be fixed, I just would like to better understand why you are running w/o the printer turned on.

Thanks,

Jeff

DisasteR, 2021-01-16:
I'm also concerned by the bug where octoprint crash after some time when printer is off.
As i understood it IS caused by octoscreen repeatedly calling octoprint api. (Trying to reconnect serial)
Maybe it IS possible to stop calling and add a sleeping screen with a retry button ?
Or maybe you could  add some udev magic with [go-udev](https://github.com/jochenvg/go-udev/blob/master/monitor.go) to check if printer is available before calling octoprint ?

JeffB42, 2021-03-27:
This was fixed in the 2.7.0 release.

closed 2021-03-27 by JeffB42