Z-Bolt/OctoScreen #116: Flickering back and forth to the terminal

Mearman, 2020-04-06:
OctoScreen keeps appearing to crash back to the terminal and restart every few seconds.

radimsejk, 2020-04-15:
I have the same problem. Here is video of it: [https://youtu.be/NJz76bidxuQ](url)

Loro77, 2020-04-15:
I have that problem too. 
I'm using a Raspberry Pi 4 and the 7'' touch display. After the installation of OctoPrint and the configuration, I installed the dependencies and System updates with "sudo apt-get update / upgrade" and "sudo apt-get install libgtk-3-0 xserver-xorg xinit x11-xserver-utils" and the .dep package. Everything without errors or warnings. 
The UI comes up until "Cannot connect initialize the printer. Tap "Retry" to try again" - then you briefly see the terminal and the UI boots again until "Cannot connect ...." ... Loop for ever ...

mbfab, 2020-04-17:
I have the same problem, anyone have a solution?

radimsejk, 2020-04-17:
> I have the same problem, anyone have a solution?

Sadly not :( I hope someone will resolve this, because right now it is unusable.

mbfab, 2020-04-17:
Octoscreen works for a few seconds and reboots again, Im having connection with other sites about this issue, if i find a solution, i post it here.

floridaservices, 2020-04-17:
> I have the same problem, anyone have a solution?

try out OctoDash, developer is a bit more involved in the discussions

mbfab, 2020-04-17:
I solved my problem...

I did like this:

when logged in in the Pi with terminal or Putty i wrote

service octoscreen status

Then i saw that Active: sayed Activating, but it should say Active (Running)

I then went in to:

sudo nano /etc/rc.local

scrolled all the way down in this list and before the last exit 0
i wrote:

systemctl start octoscreen

pressed control X, then Y then Enter to save

This might be enough to fix the problem, but in my case it was not, but after doing this i reinstalled
by pasting in following:

sudo apt-get install libgtk-3-0 xserver-xorg xinit x11-xserver-utils

Then after running that i pasted in:

wget https://github.com/Z-Bolt/OctoScreen/releases/download/v2.5.1/octoscreen_2.5-1_armhf.deb
sudo dpkg -i octoscreen_2.5-1_armhf.deb

Now OctoScreen works, hope this is help for anyone else here.

radimsejk, 2020-04-18:
> Now OctoScreen works, hope this is help for anyone else here.

Thank you very much, that solved my problem too!

mbfab, 2020-04-18:
No problem... Youtube rules ;)

Uchira, 2020-04-26:
Sorry to say that it did not solve the problem for me

dirkvranckaert, 2020-05-13:
For me I had the same problem because it could not connect to OctoPrint. As soon I as I resolved the connection to OctoPrint it stopped flickering. However that seems to be a bug to me. The UI should stay on when not connected and not try to reconnect by restarting the app...

mbfab, 2020-05-14:
It do flicker if its not connected to the printer, something for Octoscreen
to resolve

God way is to shut down screen when not in use

JeffB42, 2021-01-02:
@radimsejk and @radimsejk Please update to 2.6.1 (https://github.com/Z-Bolt/OctoScreen/releases/tag/2.6.1) and let me know if this is still an issue.  i'm not sure 2.6.1 fixes the issue you were seeing, but I did fix some issues in the Print panel and so this issue might have been fixed.  If not, let me know and I'll try to help out & see what's going wrong.

waywit, 2021-01-18:
Hi, I have the same issue.... and actually no active 3D printer.

I have turned on debug logging and maybe this helps?

This is repeated all the time:
```
time="2021-01-18T12:33:12+01:00" level=info msg="ui.update() - thus.UIState is:  splash" file=ui.go func="ui.(*UI).update" line=372
time="2021-01-18T12:33:12+01:00" level=debug msg="entering ui.checkNotification()" file=ui.go func="ui.(*UI).checkNotification" line=265
time="2021-01-18T12:33:12+01:00" level=debug msg="leaving ui.checkNotification()" file=ui.go func="ui.(*UI).checkNotification" line=284
time="2021-01-18T12:33:12+01:00" level=debug msg="entering ui.verifyConnection()" file=ui.go func="ui.(*UI).verifyConnection" line=152
time="2021-01-18T12:33:12+01:00" level=debug msg="entering ui.sdNotify()" file=ui.go func="ui.(*UI).sdNotify" line=406
time="2021-01-18T12:33:12+01:00" level=debug msg="leaving ui.sdNotify()" file=ui.go func="ui.(*UI).sdNotify" line=415
time="2021-01-18T12:33:12+01:00" level=info msg="ui.verifyConnection() - newUIState equals ui.UIState and is: \"splash\"" file=ui.go func="ui.(*UI).verifyConnection" line=233
time="2021-01-18T12:33:12+01:00" level=debug msg="leaving ui.verifyConnection()" file=ui.go func="ui.(*UI).verifyConnection" line=234
time="2021-01-18T12:33:12+01:00" level=debug msg="leaving ui.update()" file=ui.go func="ui.(*UI).update" line=402
time="2021-01-18T12:33:22+01:00" level=debug msg="entering ui.update()" file=ui.go func="ui.(*UI).update" line=363
```

Then after a while the screen flickers jumps back to blinking cursor and this is logged:
```
time="2021-01-18T12:33:22+01:00" level=debug msg="leaving ui.update() - connectionAttempts > 8" file=ui.go func="ui.(*UI).update" line=368
time="2021-01-18T12:33:24+01:00" level=debug msg="OctoScreen - entering main.init()" file=main.go func=main.init.0 line=55
time="2021-01-18T12:33:24+01:00" level=debug file=main.go func=main.readConfig line=169
time="2021-01-18T12:33:24+01:00" level=debug file=main.go func=main.readConfig line=170
time="2021-01-18T12:33:24+01:00" level=debug msg="entering main.readConfig()" file=main.go func=main.readConfig line=171
time="2021-01-18T12:33:24+01:00" level=info msg="Path to OctoPrint's config file: \"/home/pi/.octoprint/config.yaml\"" file=main.go func=main.readConfig line=180
time="2021-01-18T12:33:24+01:00" level=info msg="main.readConfig() - ReadFile() succeeded" file=main.go func=main.readConfig line=187
time="2021-01-18T12:33:24+01:00" level=info msg="main.readConfig() - YAML config file was decoded" file=main.go func=main.readConfig line=193
time="2021-01-18T12:33:24+01:00" level=info msg="main.readConfig() - server host is: \"localhost\"" file=main.go func=main.readConfig line=200
time="2021-01-18T12:33:24+01:00" level=info msg="main.readConfig() - server port is: 5000" file=main.go func=main.readConfig line=206
time="2021-01-18T12:33:24+01:00" level=debug msg="leaving main.readConfig()" file=main.go func=main.readConfig line=209
time="2021-01-18T12:33:24+01:00" level=debug file=main.go func=main.readConfig line=210
time="2021-01-18T12:33:24+01:00" level=debug file=main.go func=main.readConfig line=211
time="2021-01-18T12:33:24+01:00" level=debug msg="OctoScreen - entering main.setApiKey()" file=main.go func=main.setApiKey line=128
time="2021-01-18T12:33:24+01:00" level=debug msg="main.setApiKey() - APIKey is \"320---3B5\"" file=main.go func=main.setApiKey line=141
time="2021-01-18T12:33:24+01:00" level=debug msg="OctoScreen - leaving main.setApiKey()" file=main.go func=main.setApiKey line=144
time="2021-01-18T12:33:24+01:00" level=info msg="main.setBaseUrl() - using \"http://localhost:5000\" as server address" file=main.go func=main.setBaseUrl line=124
time="2021-01-18T12:33:24+01:00" level=debug msg="OctoScreen - leaving main.init()" file=main.go func=main.init.0 line=78
time="2021-01-18T12:33:24+01:00" level=debug msg="OctoScreen - entering main.main()" file=main.go func=main.main line=29
time="2021-01-18T12:33:24+01:00" level=info file=environment.go func=utils.DumpEnvironmentVariables line=79
time="2021-01-18T12:33:24+01:00" level=info file=environment.go func=utils.DumpEnvironmentVariables line=80
time="2021-01-18T12:33:24+01:00" level=info msg="Environment variables..." file=environment.go func=utils.DumpEnvironmentVariables line=81
time="2021-01-18T12:33:24+01:00" level=info msg="Required environment variables:" file=environment.go func=utils.DumpEnvironmentVariables line=84
time="2021-01-18T12:33:24+01:00" level=info msg="key: \"OCTOPRINT_HOST\", value: \"http://localhost:5000\"" file=environment.go func=utils.dumpEnvironmentVariable line=109
time="2021-01-18T12:33:24+01:00" level=info msg="key: \"OCTOPRINT_APIKEY\", value: \"320---3B5\"" file=environment.go func=utils.dumpObfuscatedEnvironmentVariable line=118
time="2021-01-18T12:33:24+01:00" level=info msg="key: \"OCTOSCREEN_STYLE_PATH\", value: \"/opt/octoscreen/styles/z-bolt/\"" file=environment.go func=utils.dumpEnvironmentVariable line=109
time="2021-01-18T12:33:24+01:00" level=info file=environment.go func=utils.DumpEnvironmentVariables line=90
time="2021-01-18T12:33:24+01:00" level=info msg="Optional environment variables:" file=environment.go func=utils.DumpEnvironmentVariables line=91
time="2021-01-18T12:33:24+01:00" level=info msg="key: \"OCTOPRINT_CONFIG_FILE\", value: \"/home/pi/.octoprint/config.yaml\"" file=environment.go func=utils.dumpEnvironmentVariable line=109
[... cut ...]
time="2021-01-18T12:33:25+01:00" level=debug msg="leaving ui.verifyConnection()" file=ui.go func="ui.(*UI).verifyConnection" line=261
time="2021-01-18T12:33:25+01:00" level=debug msg="leaving ui.update()" file=ui.go func="ui.(*UI).update" line=402
time="2021-01-18T12:33:34+01:00" level=debug msg="entering ui.update()" file=ui.go func="ui.(*UI).update" line=363
```

Perhaps this happens because of the not connected printer? The UI shows "Printer is Offline", but the rest is working: System Information and Network Information...

radimsejk, 2021-01-27:
> @radimsejk Please update to 2.6.1 (https://github.com/Z-Bolt/OctoScreen/releases/tag/2.6.1) and let me know if this is still an issue. i'm not sure 2.6.1 fixes the issue you were seeing, but I did [...]

Hello, sorry for the long silence from my side. I will try to update it today and report back in couple of days how it si working. Thanks.

jjgerber01, 2021-01-28:
|  
|  
|    |

  |
|  
|  
 
J

J.J. Gerber43 Hollyberry DriveParadise, NL, A1L 4A1CanadaTel: +1 (709) 749-1612Email: jjgerber01@yahoo.ca
 

  |

  |
|  
|    |

  |

  |
|  
|           Yahoo Mail Stationery     |

  |

radimsejk, 2021-02-01:
@JeffB42 I do not see any flickering now so for me it is solved.

JeffB42, 2021-02-04:
@Mearman is this still an issue?

Tobias-Keller, 2021-03-11:
> @Mearman is this still an issue?

got this problem at this moment..any solution?

jkuusama, 2021-03-13:
Yes, this is still an issue. Printing is not affected as far as I can tell, and the flicker happens only every few minutes, s not serious, only somewhat annoying. I happened to see something about x resources running out before the octoscreen came back. I'll be happy to provide you with logs etc if those would help.

JeffB42, 2021-03-27:
@Mearman I think this was fixed in version 2.7.0, which was just released.  Please update to 2.7.0 and let me know if this is still an issue.

RollingGecko, 2021-04-08:
Hi
I'm running Version 2.7.1 on fresh installed system with Klipper on Octopi 5.4.79 with an Raspberry 3b+. I've also a flickering screen.
Here is my logfile:
`time="2021-04-08T23:35:43+01:00" level=debug msg="entering OctoScreen - main.init()" file=logger.go func=logger.TraceEnter line=101
time="2021-04-08T23:35:43+01:00" level=debug msg="    entering main.readConfig()" file=logger.go func=logger.TraceEnter line=101
time="2021-04-08T23:35:43+01:00" level=info msg="        Path to OctoPrint's config file: \"/home/pi/.octoprint/config.yaml\"" file=logger.go func=logger.Infof line=150
time="2021-04-08T23:35:43+01:00" level=info msg="        main.readConfig() - server host is: \"localhost\"" file=logger.go func=logger.Infof line=150
time="2021-04-08T23:35:43+01:00" level=info msg="        main.readConfig() - server port is: 5000" file=logger.go func=logger.Infof line=150
time="2021-04-08T23:35:43+01:00" level=debug msg="    leaving main.readConfig()" file=logger.go func=logger.TraceLeave line=115
time="2021-04-08T23:35:43+01:00" level=debug msg="    entering main.setApiKey()" file=logger.go func=logger.TraceEnter line=101
time="2021-04-08T23:35:43+01:00" level=debug msg="        main.setApiKey() - APIKey is \"729---306\"" file=logger.go func=logger.Debugf line=141
time="2021-04-08T23:35:43+01:00" level=debug msg="    leaving main.setApiKey()" file=logger.go func=logger.TraceLeave line=115
time="2021-04-08T23:35:45+01:00" level=error msg="                        Client.doRequest() - this.handleResponse() returned an error: \"unexpected status code: 404\"" file=logger.go func=logger.LogError line=121
time="2021-04-08T23:35:45+01:00" level=error msg="                    Client.doJsonRequest() - this.doRequest() returned an error: \"unexpected status code: 404\"" file=logger.go func=logger.LogError line=121
`
I've no idea how to solve it...

JeffB42, 2021-04-09:
@RollingGecko I think the error you are seeing is a false-positive.  When OctoScreen starts, it attempts to call OctoPrint and get OctoScreen's plug-in's settings, and if the plug-in isn't present, it returns a 404.

(oddly enough, I checked in a guard against this the other day, and this no longer reports an error)

Unfortunately, what you posted isn't enough to go on.  Please set the log level to debug, clear the log file, reboot, and then post what's logged.

Thanks,

Jeff

RollingGecko, 2021-04-09:
That is quite a lot: 
[logfile.txt](https://github.com/Z-Bolt/OctoScreen/files/6288323/logfile.txt)

JeffB42, 2021-04-10:
Thanks @RollingGecko, that helps.  Please try this for me: ssh into your RPi, then run `sudo nano /lib/systemd/system/octoscreen.service`.  Look for `WatchdogSec`.  it's probably set to 20s.  Change WatchdogSec to 40s, save the file, then reboot your RPi.  After doing this, please let me know if this fixed the issue or if it still happens.

Thanks,

Jeff

RollingGecko, 2021-04-10:
Hi
Unfortunatly that didn't help.
Regard
Andy

JeffB42, 2021-04-10:
OK, next, please SSH into your RPi and run the following:
`curl -H "Content-Type: application/json" -H "X-Api-Key:YOURAPIKEY" -X GET http://localhost:5000/api/connection`
(and obviously substitute your API key... the one that starts with 729 and ends with 306)

When you run this curl command...
1) using a timer, approximately how many seconds does it take to run?
2) What are the results back?

Also, how do you have your API key set up?  Is the value ("729...") the global API key?
(OctoPrint > Settings > API > Global API Key)

Or is the API key that you are using defined in OctoPrint > Settings > API > Application Keys ?

Thanks,

Jeff

JeffB42, 2021-04-10:
Also, I just release 2.7.2.

RollingGecko, 2021-04-14:
Hi,
The result comes in less a second
The result is:

{"current":{"baudrate":115200,"port":"/tmp/printer","printerProfile":"_default","state":"Operational"},"options":{"baudratePreference":null,"baudrates":[250000,230400,115200,57600,38400,19200,9600],"portPreference":"/tmp/printer","ports":["/dev/ttyACM0","/tmp/printer"],"printerProfilePreference":"_default","printerProfiles":[{"id":"_default","name":"Default"}]}}

Yes the Api is a "729..." . And it is the gobal API Key.

I tried 2.7.2 without success.

RollingGecko, 2021-04-20:
Is there any further aproach?

JeffB42, 2021-04-21:
I've been looking into it but still no solution.  I'm busy with work and at home a plumbing issue came up, so I won't be posting an update/details for a few days.

JeffB42, 2021-04-25:
@RollingGecko I have some info, but not really a fix.  Here's the latest:

I looked over your log file, and the part that's interesting is this:
```
.
.
.
time="2021-04-09T21:25:39+01:00" level=debug msg="    entering ui.update()" file=logger.go func=logger.TraceEnter line=101
time="2021-04-09T21:25:39+01:00" level=debug msg="        entering ui.sdNotify()" file=logger.go func=logger.TraceEnter line=101
time="2021-04-09T21:25:39+01:00" level=debug msg="        leaving ui.sdNotify()" file=logger.go func=logger.TraceLeave line=115
time="2021-04-09T21:25:39+01:00" level=info  msg="        ui.update() - this.UIState is: \"idle\"" file=logger.go func=logger.Infof line=150
time="2021-04-09T21:25:39+01:00" level=debug msg="        entering ui.verifyConnection()" file=logger.go func=logger.TraceEnter line=101
time="2021-04-09T21:25:39+01:00" level=debug msg="            entering ConnectionRequest.Do()" file=logger.go func=logger.TraceEnter line=101
time="2021-04-09T21:25:39+01:00" level=debug msg="                entering Client.doJsonRequest()" file=logger.go func=logger.TraceEnter line=101
time="2021-04-09T21:25:39+01:00" level=debug msg="                    entering Client.doRequest()" file=logger.go func=logger.TraceEnter line=101
time="2021-04-09T21:25:39+01:00" level=debug msg="                        method: GET" file=logger.go func=logger.Debugf line=141
time="2021-04-09T21:25:39+01:00" level=debug msg="                        target: /api/connection" file=logger.go func=logger.Debugf line=141
time="2021-04-09T21:25:39+01:00" level=debug msg="                        contentType: application/json" file=logger.go func=logger.Debugf line=141
time="2021-04-09T21:25:39+01:00" level=debug msg="                        url: http://localhost:5000/api/connection" file=logger.go func=logger.Debugf line=141
time="2021-04-09T21:25:39+01:00" level=debug msg="                        userAgent: go-octoprint/0.1" file=logger.go func=logger.Debugf line=141
>>> this is were OctoScreen reboots
time="2021-04-09T21:25:41+01:00" level=debug msg=- file=logger.go func=logger.Debug line=137
time="2021-04-09T21:25:41+01:00" level=debug msg=- file=logger.go func=logger.Debug line=137
time="2021-04-09T21:25:41+01:00" level=debug msg="entering OctoScreen - main.init()" file=logger.go func=logger.TraceEnter line=101
time="2021-04-09T21:25:41+01:00" level=debug msg="    entering main.readConfig()" file=logger.go func=logger.TraceEnter line=101
time="2021-04-09T21:25:41+01:00" level=info  msg="        Path to OctoPrint's config file: \"/home/pi/.octoprint/config.yaml\"" file=logger.go func=logger.Infof line=150
time="2021-04-09T21:25:41+01:00" level=info  msg="        main.readConfig() - server host is: \"localhost\"" file=logger.go func=logger.Infof line=150
time="2021-04-09T21:25:41+01:00" level=info  msg="        main.readConfig() - server port is: 5000" file=logger.go func=logger.Infof line=150
time="2021-04-09T21:25:41+01:00" level=debug msg="    leaving main.readConfig()" file=logger.go func=logger.TraceLeave line=115
time="2021-04-09T21:25:41+01:00" level=debug msg="    entering main.setApiKey()" file=logger.go func=logger.TraceEnter line=101
time="2021-04-09T21:25:41+01:00" level=debug msg="        main.setApiKey() - APIKey is \"729---306\"" file=logger.go func=logger.Debugf line=141
time="2021-04-09T21:25:41+01:00" level=debug msg="    leaving main.setApiKey()" file=logger.go func=logger.TraceLeave line=115
time="2021-04-09T21:25:41+01:00" level=info  msg="    main.setBaseUrl() - using \"http://localhost:5000\" as server address" file=logger.go func=logger.Infof line=150
time="2021-04-09T21:25:41+01:00" level=debug msg="leaving OctoScreen - main.init()" file=logger.go func=logger.TraceLeave line=115
.
.
.
```

The "entering OctoScreen - main.init()" is what's logged when OctoScreen starts, and here are some comments about this (when running on your machine):
1. This is happening when the REST API http://localhost:5000/api/connection is called
2. It's failing at this spot, but looking through the log, it's called two other times before, and calls #1 & #2 pass, but the third call is failing for some reason.  At this point, I don't know why it's succeeding some times but failing the third time.
3. The last log entry (before the restart) was at 2021-04-09T21:25:39+01:00.  After the restart the next message logged was at 2021-04-09T21:25:41+01:00.  That's only a 2 second difference, so it doesn't seem to be an issue of the API taking too long and the watchdog restarting the service.
4. I was thinking that calling api/connection was taking too log, but when you ran curl, you reported that it only took a second or two, and in the log file you posted it appears only a few seconds go by.

I release v2.7.3 and added some additional logging in this area.  Perhaps something inside of `Client.doRequest()` is causing an exception.  Looking through the code I didn't see anything substantial, but I added more logging to this area.  Please do the following:
1. Uninstall v2.7.2
2. Install v2.7.3
3. Edit OctoScreen's config file.  Turn logging on (by specifying a path to a log file) and set the level to debug.
3. Clear your log file (either delete it and then recreate it, or remove all of its contents)
4. Reboot
5. Post the log file

After you post the new log file (assuming the crash still happens), I'd like you to try out a new config setting I just added, named `EXPERIMENTAL_UPDATE_FREQUENCY`.  This hasn't been officially announced or d [... cut ...]
The value here is for the linux watchdog service.  The default value is 20 (seconds), and you might try increasing it to 30 or 40.  If you do change it, you will also need to edit `WatchdogSec` in octoscreen.service.  Edit octoscreen.service...
```
sudo nano /lib/systemd/system/octoscreen.service
```

...and then set the value for WatchdogSec.  The value of `WatchdogSec` needs to be 2x that of `EXPERIMENTAL_UPDATE_FREQUENCY` (eg, if `EXPERIMENTAL_UPDATE_FREQUENCY` is 30, `WatchdogSec` should be `60s`).  Also, the value for `EXPERIMENTAL_UPDATE_FREQUENCY` is just the number (eg `30`) but the value for `WatchdogSec` also has a trailing 's' (eg `60s`)

JeffB42, 2021-05-01:
@RollingGecko just a reminder, please install 2.7.3 and then post the log file that was generated.

Thanks,

Jeff

RollingGecko, 2021-05-03:
Sorry.At the moment I'm bussy with some other things. Will try to test it soon. Thx

JeffB42, 2021-05-24:
@RollingGecko any update on this?

Kyrunix, 2021-05-26:
(At first, i'm sorry. My english is not really good)
I noticed this issue and noticed it on my system a few days ago 
(prusa i3 mk3s; rasp pi 4 2GB; octopi 0.17.0; octoprint 1.6.1; octoscreen 2.7.3)
If @RollingGecko don't response til saturday or i find some time before, i will do the changes of your post (25. april) on my system.
Please wait these few days with closing this issue

RollingGecko, 2021-05-26:
Sorry that I didn't answer till now. I managed it to smash my screen and I wait for a new one. I will be now on holidays for 2 weeks. I hope that during this time the screen will arrive. I'll keep you updated.

JeffB42, 2021-08-08:
Closed due to lack of response.

closed 2021-08-08 by JeffB42

Kyrunix, 2021-08-19:
Sorry for no response. I was injured and after this i had a lot to do at work. It looks like, that update 2.7.4 fix the problem. I didn't noticed screen crashes after update.

Thanks a lot and sorry again.