Z-Bolt/OctoScreen #117: Not able to get octoscreen to the screen

daktylus, 2020-04-06:
Hi I have done a fresh install of octopi and also my 3.5" GPIO LCD driver followed by Octoscreen installation. However after installing or rebooting the display showed me only the command line login even after octoscreen.service was active. After I used raspi-config to set the boot option to desktop autologin, the display shows some part of the boot log and when I login via putty i get this.

[image]

from here on nothing happens. Octoprint itself is usable via the browser.

What did I wrong and how can get octoscreen to work proberly?

mdaneman, 2020-05-27:
Have you tried typing just "sudo systemctl start octoscreen.service"?  If this makes Octoscreen come up, then you just need to set it up as you window manager:
sudo dpkg-reconfigure octoscreen
and it should come up automatically when you boot to the desktop enviroment.
There's also a way to get it to come when you boot just to console, but try the above first.

daktylus, 2020-05-27:
Yes I tried this several times without any success. :-(

JeffB42, 2020-12-04:
@daktylus Could you follow the steps listed in the wiki (located at https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen) and then post a reply to this issue whether the steps listed work and the issue is resolved?  I'd like to start closing out all the display-related issues.

If it doesn't work,
1) make sure you use a Raspberry Pi 3B or Raspberry Pi 4
2) start with a fresh install of OctoPi
3) please take a picture of the screen

Thanks, 

Jeff

daktylus, 2020-12-04:
After severals iterations, I got it to work. The Problem was the missing fbturbo during driver installation. The following worked for me (Raspberry Pi 3B + 3.5" RPi Display)

- Fresh Install Octopi (buster lite 0.17.0) 
- Install fbturbo. Needs to be installed before display driver!!
- Install display drivers
- Install Octodash 2.1.1

Tobias

closed 2020-12-04 by daktylus