Z-Bolt/OctoScreen #119: Wrong size resolution

Cakekong, 2020-04-15:
Hello to all, 

I realy don't understand. I have this screen (https://www.amazon.fr/gp/product/B01CNLYL1C/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)
I think i have a godd configuration too, but no ...
[image]

Could you help me please ? Thank you !

floridaservices, 2020-04-15:
I had the same experience with OctoScreen with my 3.5" LCD  - After struggling to get it working like you, I switched over to OctoDash  and it's working as expected.

Cakekong, 2020-04-15:
Thank you! Do I need to remove Octoscreen before? If yes comment? Thank you :)

floridaservices, 2020-04-15:
yes, always good practice.  I just wiped the sd card and rebuilt but I am not as proficient with linux as i would like to be

Cakekong, 2020-04-15:
Installation very easy of OctoDash , but it see no file :'(

tutdeveloper, 2020-05-05:
Hi
Your screen does not seem to be 800x533. The Amazon posting shows it as 320x480. Not sure if that might be your problem.

closed 2020-05-06 by Cakekong