Z-Bolt/OctoScreen #120: Wrong resolution in "print screen"

onit12, 2020-04-22:
Hi!
I have a hyperpixel 4 touch, which works fine except the screen in "print mode" which goes over the right edge:

[image]

Home screen and others are fine. 
Anybody an idea?

Thanks
Tino

FsFalcon, 2020-04-22:
I also have this same issue; I'm using the Adafruit PiTFT Plus 480x320 3.5" screen

mkenngott, 2020-05-07:
i have the same issue with a 5" touchscreen with 800x480

dirkvranckaert, 2020-05-13:
Had the same issue only once, after the very first print with Octoscreen connected. But it was the only time so far that I started the print from Octoscreen, maybe that is the issue?! I noticed after the night print that the screen was on the main screen but to wide...

alexkychen, 2020-05-17:
> I also have this same issue; I'm using the Adafruit PiTFT Plus 480x320 3.5" screen

I got a 3.5" screen (UCTRONICS brand) from Amazon, and also had a similar resolution issue when set up. I managed to fix it by changing the default 480x320 to 800x533 pixels in /boot/config.txt

Specifically,  hdmi_cvt=800 533 60 6 0 0 0

rwaspe, 2020-05-24:
Looks like the same issue here: https://github.com/Z-Bolt/OctoScreen/issues/126

marbet-zz, 2020-08-17:
I have the same problem with the 7" "official" raspberry pi screen.
Looks nice till you start printing, then the screen moves a couple of 100 pixels to the right (half way out of view)

rwaspe, 2020-08-17:
@marbet-zz Does it happen with short file names (under 8 characters)?

JeffB42, 2020-08-18:
@onit12 and @marbet-zz - which version of OctoScreen are you using?  there was a known bug where, if the filename was long, it would print past the bounds of the screen, and the screen would be resized, however that bug was fixed and I think the current released version (2.5.1) has the fix in it.

AdrianTwaTR, 2020-08-18:
@JeffB42 - It looks like the clipping fix is in a [commit ](https://github.com/Z-Bolt/OctoScreen/commit/238a76063bbf03c026c2e72c230c7e313a4d1fde) from July, but not released yet?

FsFalcon, 2020-08-18:
@JeffB42 - It looks like the clipping fix is in a [commit ](https://github.com/Z-Bolt/OctoScreen/commit/238a76063bbf03c026c2e72c230c7e313a4d1fde) from July, but not released yet?

JeffB42, 2020-08-19:
@FsFalcon @onit12 and @marbet-zz  - sorry, I got my wires crossed.  The change got checked into the master branch, but FsFalcon is correct, that was in July and was after the 2.5.1 release was built.  There hasn't been a new build though (and I'm not planning on publishing a build until 2.6).  If you are you are adventuresome, there are build steps on the main page, (but I haven't tried building that way, yet).  If you pull the latest from master and build, you'll pick up the change.

marbet-zz, 2020-08-19:
No, I tried that today, and with really short names it works.

It did not work for long filenames with spaces, since octoprint seems to
replace space with underscore internally.

rwaspe, 2020-08-19:
Follow the solution in #126 till a new version comes out.

JeffB42, 2020-12-04:
@onit12 Could you follow the steps listed in the wiki (located at https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen) and then post a reply to this issue whether the steps listed work and the issue is resolved?  I'd like to start closing out all the display-related issues.

If it doesn't work,
1) make sure you use a Raspberry Pi 3B or Raspberry Pi 4
2) start with a fresh install of OctoPi
3) please take a picture of the screen

Thanks, 

Jeff

JeffB42, 2021-01-02:
This has been fixed with the 2.6.1 release.

closed 2021-01-02 by JeffB42