Z-Bolt/OctoScreen #121: missing dependencies

Charlemagne65, 2020-04-22:
Hello. Would like to use this software but installation ends in dependencies missing. I am new to linux/raspbian so please forgive ignorance. 

I issued command like stated: sudo apt-get install libgtk-3-0 xserver-xorg xinit x11-xserver-utils
output: 
Reading package lists... Done
Building dependency tree
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 libgtk-3-0 : Depends: adwaita-icon-theme but it is not going to be installed
              Depends: libatk-bridge2.0-0 (>= 2.5.3) but it is not going to be installed
              Depends: libatk1.0-0 (>= 2.15.1) but it is not going to be installed
              Depends: libcairo-gobject2 (>= 1.10.0) but it is not going to be installed
              Depends: libcolord2 (>= 0.1.10) but it is not going to be installed
              Depends: libcups2 (>= 1.6.0) but it is not going to be installed
              Depends: libepoxy0 (>= 1.4.3) but it is not going to be installed
              Depends: libjson-glib-1.0-0 (>= 0.12.0) but it is not going to be installed
              Depends: librest-0.7-0 (>= 0.7) but it is not going to be installed
              Depends: libsoup2.4-1 (>= 2.4.0) but it is not going to be installed
              Depends: libxcomposite1 (>= 1:0.3-1) but it is not going to be installed
              Depends: libgtk-3-common (>= 3.24.5-1+rpt2) but it is not going to be installed
              Recommends: libgtk-3-bin but it is not going to be installed
 octoscreen : Depends: libatk1.0-0 (>= 1.12.4) but it is not going to be installed
              Depends: libcairo-gobject2 (>= 1.10.0) but it is not going to be installed
 x11-xserver-utils : Depends: libxaw7 but it is not going to be installed
                     Depends: libxmu6 but it is not going to be installed
                     Depends: libxt6 but it is not going to be installed
 xserver-xorg : Depends: xserver-xorg-core (>= 2:1.17.2-2) but it is not going to be installed
                Depends: xserver-xorg-video-all but it is not going to be installed or
                         xorg-driver-video
                Depends: x11-xkb-utils but it is not going to be installed
                Recommends: xserver-xorg-legacy but it is not going to be installed
 xserver-xorg-input-evdev : Depends: libevdev2 (>= 1.2.2+dfsg-1~) but it is not going to be installed
                            Depends: libmtdev1 (>= 1.1.0) but it is not going to be installed
                            Depends: xorg-input-abi-24
                            Depends: xserver-xorg-core (>= 2:1.18.99.901) but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).

How can I fix this?

Dario-Ciceri, 2020-04-23:
> Hello. Would like to use this software but installation ends in dependencies missing. I am new to linux/raspbian so please forgive ignorance.
>
> I issued command like stated: sudo apt-get install [...]

try before installation to run this two commands: `sudo apt-get update` and `sudo apt-get upgrade`

Charlemagne65, 2020-04-24:
I solved this not by following the suggested "apt --fix-broken install" but by using the command "sudo apt --fix-broken install". The remaing problem is that octoscreen is showing up on my hdmi connection but not on my tft screen. It seems these two are seperate on linux. I don't know if I can switch the two somehow or should install using only the tft screen. Will octoscreen show on the first screen found on initial boot? If I have further questions I will post another question. Many thanks for now.

closed 2020-04-24 by Charlemagne65