Z-Bolt/OctoScreen #125: Octoscreen service is running but no interface on the screen

rcsez, 2020-04-30:


pjratl, 2020-05-01:
having the same issue with this screen https://www.amazon.com/gp/product/B07N38B86S/
during boot the text shows on the screen but then nothing

pjratl, 2020-05-02:
Ok I fixed my issue not sure of the exact steps but my issues was FBCP was not running I ended up editing /etc/rc/local and fully qualifying it in my case /usr/local/bin/fbcp/fbcp after installing/reinstalling it...

rcsez, 2020-05-04:
> Ok I fixed my issue not sure of the exact steps but my issues was FBCP was not running I ended up editing /etc/rc/local and fully qualifying it in my case /usr/local/bin/fbcp/fbcp after installing/reinstalling it...

Thanks! For some reason FBCP wasn't running, and starting it up did the trick.

Morcegolas, 2020-05-08:
Hello, my service is also running but I have no interface on the screen, what is FBCP and how do I install it?

Thanks.

pjratl, 2020-05-08:
hope this helps https://github.com/tasanakorn/rpi-fbcp

hrqiang, 2020-05-09:
I get the screen showing properly, but the touch is not working. 
Anyone know how to fix?

JeffB42, 2020-12-04:
@rcsez Could you follow the steps listed in the wiki (located at https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen) and then post a reply to this issue whether the steps listed work and the issue is resolved?  I'd like to start closing out all the display-related issues.

If it doesn't work,
1) make sure you use a Raspberry Pi 3B or Raspberry Pi 4
2) start with a fresh install of OctoPi
3) please take a picture of the screen

Thanks, 

Jeff

JeffB42, 2021-01-02:
@rcsez Is this still an issue?

JeffB42, 2021-02-04:
Closed due to inactivity.

closed 2021-02-04 by JeffB42