Z-Bolt/OctoScreen #126: after a few prints, octoscreen enlarges and change the resolution

cryptodonniedxb, 2020-05-01:
I'm running the latest version of Octoscreen on a pi 3b+, with a kuman 3,5 inches 320*480 display.
After a few prints, octoprins kinda "enlarges" the resolution and moves on the right, therefore unsetting to the resolution.
Here is a picture of what happens during print:
[image]
and in the worst cases, when back to home is like this
[image]

reboots fix, but after the first print, it go back to the same problem

does anybody had the same issue?

FsFalcon, 2020-05-02:
Look similar to (or is the same as) #120

agentsmith29, 2020-05-15:
Hi! I finally found the solution (or a least the problem): If the filename is too large for the screen so OctoScreen UI enlarges. To avoid this, use shorter filenames or as I suggest change the source code:

1.  Download the source code: [Releses](https://github.com/Z-Bolt/OctoScreen/releases)
(the repository is not working properly, I've also added an issue for this)
2. Open the file `ui/print_status.go` and change  line 262 to e.g.
`file = strEllipsisLen(file, 20)` the number **20** defines the length a string should get shorten to. By default it's set to 35 which is to long for my screen (Hyperpixel 4)
3. Compile the project using `make build`
4. Install it using `system dpkg -i octoscreen_2.5-1*.deb`

Tested with a Hyperpixel 4 (800x480) and a Rapsberry Pi 4.

rwaspe, 2020-05-23:
> Hi! I finally found the solution (or a least the problem): If the filename is too large for the screen so OctoScreen UI enlarges. To avoid this, use shorter filenames or as I suggest change the [...]

Unfortunately I'm not quite following how to change that line of code and get it installed.  I downloaded the source on my computer and can edit the file but then don't know how to compile it and install it on the Raspberry.  Would anyone be able to do a quick step by step?  Thank you.

agentsmith29, 2020-05-24:
For clarification: Obviously, I gave you the wrong installation command (I wrote `system` instead of `sudo`). 

For the answer it's assumed that you followed along the description on how to setup Octoscreen to use Hyperpixel's resolution. And I'm assuming that the screen resolution is **correct**, but only enlarges when accessing the the *job-status*-screen.

----
I don't know where you got stuck, you need to tell us what didn't work. Actually, there are **no** further steps to perform. 
If you have problems following my steps, have a look at the official [guidance](https://github.com/Z-Bolt/OctoScreen#install-from-source) on how to compile this project. Please go through it. 
If you have troubles using *docker* or *make*, please share the specific error message with us, otherwise we won't be able to help you.

PS: You need to download the source files using the 'Release'-Tab, because this project is straight up broken when using the committed files from branch master. This should be fixed as well.

rwaspe, 2020-05-24:
@agentsmith29 Please understand that I am new to Linux and am figuring it out as I go like how in the Install from source section, it says "assuming that you already cloned this repository" means I had to search how to do that and then figure out that I needed to go to the OctoScreen directory to do `make build`. This took a while of figuring everything out but I finally got it.  Also, the instructions to install from source say `make build` and that didn't work until I changed it to `sudo make build`.  Things like that are going to trip up a lot of newbies like myself.

Ok.  Step by step for anyone else having problems because they are brand new to the Linux/Raspberry system:

In a terminal window to your Pi (Putty for example):
Install Docker First:
`curl -fsSL get.docker.com -o get-docker.sh`
`sh get-docker.sh`  (Can take a while and will look frozen.)

OctoScreen File Name Length Fix (Can be done if OctoScreen is already installed):

`git clone https://github.com/Z-Bolt/OctoScreen` (Just to be absolutely certain that I had the files from the "release tab", I downloaded the source code from [here](https://github.com/Z-Bolt/OctoScreen/releases/tag/v2.5.1), unzipped it then copied it to the Pi by FTP.  It may not have been necessary but I didn't know what was causing errors.  Turns out the OctoScreen Install from Source instructions were missing a command most likely.)

`sudo nano OctoScreen/ui/print_status.go`
Press Ctrl and Underscore( _ ) then enter 262 for the line you want to go to.
Change the 35 after file to 20.
Press Ctrl and X then Y and Enter to save.

`cd ./OctoScreen`

`sudo make build`

`cd build/stretch`

`sudo dpkg -i octoscreen_2.5-1_armhf.deb`

`sudo reboot`

Hope that helps!  Maybe in a future release the file name could just scroll or at the very least that 35 to 20 name length change is in the code.

JeffB42, 2020-07-31:
I just pushed this change.  The length is now clipped at 20.

closed 2020-07-31 by JeffB42

chrkov, 2021-02-03:
I just installed the lastest Octoscreen and I am seeing this issue.  Has this somehow been reintroduced on the latest build somehow?