Z-Bolt/OctoScreen #127: Orange PI One & 3.5" GPIO LCD

savenko-egor, 2020-05-02:
Hello!
Anybody use this hardware?
I have Orange PI One and  [3.5" GPIO LCD](https://aliexpress.ru/item/32771565544.html) Chinese Waveshare copy from AliExpress.
I installed all software: Drivers for LCD, for touchscreen, OctoPrint and OctoScreen.
All works, but I see buttons and text are out of place. I was trying use other resolution, but this not take any effect. I think problem is not in resolution. Background image place is normal.
[image]

Maybe anybody can help me? Or maybe anybody have any think how fix it?
Thanks!

P.S. Sorry for my bad English.

mariusdp, 2020-06-05:
Any solution???
I have a GPI KeiDei 480x320. Works fine with octoprint-tft but found no solution for octoscreen.

savenko-egor, 2020-06-05:
I changed OctoScreen on OctoDash.

closed 2021-01-02 by JeffB42