Z-Bolt/OctoScreen #131: FEATURE REQUEST: Add support for Prusa Thumbnails plugin

ruralaccent, 2020-05-03:
Prusa Thumbnails is an awesome plugin that displays a thumbnail of the STL file in the file list (and also the selected file whilst printing). 

It would be amazing if Octoscreen (which is one of my absolute favourite plugins) could support this, just as the Printoid app has just done.

Many thanks! :)

dirkvranckaert, 2020-05-13:
That would be a really nice addition to the UI :-)

skibum-za, 2020-05-27:
Just installed the plugin and I agree it would be great.

noxhirsch, 2020-05-27:
It seems like OctoScreen isn't maintained anymore. I switched to OctoDash (https://github.com/UnchartedBull/OctoDash). It already supports thumbnails (displayed rather small) and with the next version coming soon it supports themes for which I created one with big thumbnails:
https://github.com/noxhirsch/OctoDash/tree/master/themes/NOX

dirkvranckaert, 2020-05-27:
Great tip, will give it a try one of these days!

ruralaccent, 2020-05-29:
Is there any chance of getting the preview thumbnails displaying a bit bigger, do you think? They are really small when the job is printing. Many thanks!

noxhirsch, 2020-05-29:
@ruralaccent Do you mean in OctoDash or in my theme for OctoDash?

JeffB42, 2021-02-04:
This will be in the 2.7 release.

JeffB42, 2021-03-27:
Boom!  Done!

closed 2021-03-27 by JeffB42