Z-Bolt/OctoScreen #132: FEATURE REQUEST: Screen sleep / wake on touch?

ubermick, 2020-05-04:
It's a good lookin' plugin, prettier than TouchUI, and while I had a helluva time getting it up and running, its really nice. But... the constantly "on" screen has me worried that my lovely little Hyperpixel screen will suffer from being on constantly. Any chance we can get a screen sleep/wake on touch added?

spawnrider, 2020-05-11:
+1 for this feature

Bigfellas, 2020-07-18:
+1 100% yes please

JeffB42, 2020-07-31:
Run `sudo nano /etc/octoscreen/disablescreenblank.sh`

Then edit the following and add # before each line (commenting the line out)
```
#/usr/bin/xset -display :$screen s off          # deactivate screen saver
#/usr/bin/xset -display :$screen -dpms          # disable DPMS
#/usr/bin/xset -display :$screen s noblank      # disable screen blanking
```

closed 2020-07-31 by JeffB42

madd0gcom, 2023-01-18:
Hey Jeff, I am fully new to all this, and just installed Octoscreen, which is amazing add-on for my printer, however, the screen does not go dark after 10 Seconds.... Installed the latest version, and checked that the # were added in the lines you mentioned.
A quick help would be really apreciated! Thanks for building this great feature! By the way, my display is an original Pi 7 inch screen.