Z-Bolt/OctoScreen #133: Feature Request: Restart print

cpyarger, 2020-05-05:
I would love to see the print screen have a restart print button, Similar to the Octoprint Web interface When a print is paused.

Mearman, 2020-05-11:
I think the same as I was intending with this issue https://github.com/Z-Bolt/OctoScreen/issues/122

dirkvranckaert, 2020-05-13:
Would be really nice! I often print something more than once and just hitting a reprint button would be a nice addition!

cpyarger, 2020-05-13:
I often use the feature in octoprint for doing bed leveling tests
Warmest Regards,

Christopher P. Yarger

Phone: 802-505-7574
Skype: cpyarger

/************************************
Based upon email reliability concerns,
please send an acknowledgment in response to this note.