Z-Bolt/OctoScreen #137: Where is the Menu configuration file?

lienbacher, 2020-05-12:
hello! 
I currently feel stupid, I've been trying for an hour to find the configuration file that correlates to this: https://github.com/Z-Bolt/OctoScreen/blob/master/docs/README.md

I only have a single extruder, like 99% of all 3d printers. But I fail removing the toolchanger from the OctoScreen Menu, did not even get to try adding other commands. Reading the documentation leaves me clueless. 

Any help would be vastly appreciated!

rwaspe, 2020-05-18:
Also unable to update the menu.  I have tried editing it through the OctoPrint plugin as well as the /etc/octoscreen/default_menu.json and /opt/octoscreen/styles/z-bolt/default_menu.json files.  I confirmed that the edits are reflected in the /home/pi/.octoprint/config.yaml file.

Protean-Man, 2020-05-26:
Agreed. How do we edit the menus?
Buttons Images, Button Actions, Button Locations, and more?

totalitarian, 2020-07-16:
Same question

JeffB42, 2020-07-19:
I'm not the developer or maintainer of OctoScreen, but a search revealed that it's in OctoScreen-master/ui/menu.go

JeffB42, 2020-07-19:
I did some digging.  I don't know Go so the source code is a little cryptic to me, but here's what (I think) I've found:

/ui/menu.go has:
```
func getDeafultMenu() []octoprint.MenuItem {

	default_menu := `[
	.
	.
	.

	var items []octoprint.MenuItem

	json.Unmarshal([]byte(default_menu), &items)

	return items
}
```

getDeafultMenu() is called in ui/idle_status.go, line 45:
```
func (m *idleStatusPanel) initialize() {
	defer m.Initialize()

	var menuItems []octoprint.MenuItem

	Logger.Info(m.UI.Settings)

	if m.UI.Settings == nil || len(m.UI.Settings.MenuStructure) == 0 {
		Logger.Info("Loading default menu")
		menuItems = getDeafultMenu()
	} else {
		Logger.Info("Loading octo menu")
		menuItems = m.UI.Settings.MenuStructure
	}

	.
	.
	.
}
```

So, if m.UI.Settings.MenuStructure doesn't exist, getDeafultMenu() is called.
Else, if m.UI.Settings.MenuStructure does exist, use it.

Searching for MenuStructure, I found in vendor/github.com/mcuadros/go-octoprint/zbolt.go, line 115:
```
type GetSettingsResponse struct {
	// Job contains information regarding the target of the current print job.
	FilamentInLength  float64    `json:"filament_in_length"`
	FilamentOutLength float64    `json:"filament_out_length"`
	ToolChanger       bool       `json:"toolchanger"`
	ZAxisInverted     bool       `json:"z_axis_inverted"`
	MenuStructure     []MenuItem `json:"menu_structure"`
	GCodes            struct {
		AutoBedLevel string `json:"auto_bed_level"`
	} `json:"gcodes"`
}

func (cmd *GetSettingsRequest) Do(c *Client) (*GetSettingsResponse, error) {
	cmd.Command = "get_settings"

	params := bytes.NewBuffer(nil)
	if err := json.NewEncoder(params).Encode(cmd); err != nil {
		return nil, err
	}

	b, err := c.doJSONRequest("POST", URIZBoltOctoScreenRequest, params, ConnectionErrors)
	if err != nil {
		return nil, err
	}

	r := &GetSettingsResponse{}
	if err := json.Unmarshal(b, r); err != nil {
		return nil, err
	}

	return r, err
}
```

On line 129, there's:
```
b, err := c.doJSONRequest("POST", URIZBoltOctoScreenRequest, params, ConnectionErrors)
```

...and URIZBoltOctoScreenRequest is defined on line 9 as:
```
const URIZBoltOctoScreenRequest = "/api/plugin/zbolt_octoscreen"
```

So, from what I can tell, it looks like the function:
```
func (cmd *GetSettingsRequest) Do(c *Client) (*GetSettingsResponse, error)
```
...is making a JSON request ("doJSONRequest") using POST (line 129... why POST for a get operation?... why not use GET?) to "/api/plugin/zbolt_octoscreen"

Now, what I don't know is, is /api/plugin/zbolt_octoscreen meant to be an API request to OctoPrint, and there's supposed to be a plugin in OctoPrint named zbolt_octoscreen?  Or if OctoScreen has its own server, and requests to /api/plugin/zbolt_octoscreen is to a service/server OctoScreen is running.

Like I said, I'm not the developer of this repro, and I don't know Go, so this might be ovious to someone more familiar with this codebase or with Go.  Perhaps others can chime in.

JeffB42, 2020-07-19:
I just filed https://github.com/Z-Bolt/OctoPrint-Z-Bolt-OctoScreen/issues/5

JeffB42, 2020-07-29:
@here (@lienbacher @rwaspe @Protean-Man @totalitarian ) - good news, I contacted Aleksei and he granted me moderator rights.  I just fixed the CSS issue I reported earlier.

closed 2020-07-29 by JeffB42

Protean-Man, 2020-07-29:
@jeffB42 you are on a roll! 
I will definitely be messing with this over the next few days. Thanks and I hope you are able to work through some of the requested changes and updates to Octoscreen now that you are moderator.