Z-Bolt/OctoScreen #138: Build of source code succeeds but nothing has changed in the UI

agentsmith29, 2020-05-13:
I' currently trying to play a little bit with OctoScreen and tried to build the files from source. But unfortunately I'm got stuck while compiling them. 

Steps I performed:
1. Clone repo to home folder.
2. Edit a file (e. g. (_./ui/system.go_). I've just changed a text of a button. So I didn't mess up anything and the file should compile just fine.
4. Remove OctoScreen using `sudo dpkg -r octoscreen`
3. Call `make build`
4. Wait **very long** (the `go install` command in the docker container takes very long, is it possible to speed this up?)
5. Install my new deb-file by calling `sudo dpkg -i build/stretch/octoscreen_2.5-1_armhf.deb`.

Issues:
1. My OctoScreen starts up but nothing has changed... In fact the UI tells me I'm running "2.2(unknown)" - I don't know if it's just a bug, but I've never had any other version than the current one (2.5-1) installed. This was a completely fresh Raspian image.
2. Just to be said: I don't have a glue about `dpkg-deb`, but I even had troubles when changing the version for testing purposes to 2.6. I've done this by setting the variable `VERSION` it in the Makefile, but my Docker Container fails with the error:

`dpkg-source: error: can't build with source format '3.0 (quilt)': no upstream tarball found at ../octoscreen_2.5.orig.tar.{bz2,gz,lzma,xz}`

JeffB42, 2020-12-30:
@agentsmith29 I don't know what to tell you, other than it works for me.  The only other thing to mention is to remove the files in build/stretch first before building.  Here are the steps I take.
1. ssh into RPi
2. cd into git clone on OctoScreen.  The path needs to be ~/go/src/github.com/Z-Bolt/OctoScreen
3. edit a file
4. sudo rm build/stretch/*
5. make build
6. cd build/stretch
6. ls -l (to verify the files were built)
7. sudo dpkg -r octoscreen
8. sudo dpkg -i {name of the deb package that was built}
9. sudo reboot now

JeffB42, 2020-12-30:
Let me know if you still have issues, of if you can now build,. please close this ticket.

closed 2021-03-27 by JeffB42