Z-Bolt/OctoScreen #139: Folder content not displaying

dirkvranckaert, 2020-05-13:
I have some nested folders and there are some issues with that.
First of all the folder size is always 0 bytes... But that not really a blocker to me, however the content of my folder is shown for the root and 1 level down. 

ROOT
- GcodesFolder1
 - testA.gcode
 - SubFolder1_1
  - test.gcode
  - test1.gcode
  - test2.gcode
 - SubFolder1_2
- GcodesFolder2
 - testB.gcode

In the the example above the testA and testB are selectable as prints, the folders SubFolder1_1 and SubFolder1_2 also. But content inside these subfolder not... So it seems the recursive lookup of files is not working. Tried restarting Ocotscreen as well as restarting the entire RPi but no luck...

cwevenson, 2020-12-01:
I upgraded my Octoprint to version 1.5 and found that certain functions did not work anymore with Octoscreen.  I have a number of print files organized in folders and within the folder there is no listing of files now after to update. I am going to look into this issue and see if I need to change any permissions or if a refresh of the Octoscreen code will help.  Seems kind of fishy.

JeffB42, 2020-12-27:
@dirkvranckaert , @trainspotta , and @cwevenson I looked into this and found the issue.  The fix will be in the upcoming 2.6 release.

closed 2020-12-27 by JeffB42