Z-Bolt/OctoScreen #140: Display shows octopi login screen

duennerd, 2020-05-16:
Hello 
I installed all dependecies and installed octoscreen 2.5.1 and tried also with 2.4.9 from https://github.com/Z-Bolt/OctoScreen/releases. 

When I type:  systemctl status octoscreen.service
I get this message: 

```
octoscreen.service - OctoScreen
   Loaded: loaded (/lib/systemd/system/octoscreen.service; enabled; vendor prese
   Active: active (running) since Sat 2020-05-16 15:49:01 BST; 23s ago
  Process: 1032 ExecStartPost=/bin/bash /etc/octoscreen/disablescreenblank.sh 0
 Main PID: 1011 (xinit)
    Tasks: 19 (limit: 2200)
   Memory: 23.2M
   CGroup: /system.slice/octoscreen.service
           ├─1011 /usr/bin/xinit /usr/bin/OctoScreen -- :0 -nolisten tcp -nocurs
           ├─1012 /usr/lib/xorg/Xorg :0 -nolisten tcp -nocursor
           └─1020 /usr/bin/OctoScreen

May 16 15:48:50 octopi xinit[1011]:         (WW) warning, (EE) error, (NI) not i
May 16 15:48:50 octopi xinit[1011]: (==) Log file: "/var/log/Xorg.0.log", Time:
May 16 15:48:50 octopi xinit[1011]: (==) Using config directory: "/etc/X11/xorg.
May 16 15:48:50 octopi xinit[1011]: (==) Using system config directory "/usr/sha
May 16 15:48:50 octopi xinit[1011]: time="2020-05-16T15:48:50+01:00" level=info
May 16 15:48:50 octopi xinit[1011]: time="2020-05-16T15:48:50+01:00" level=info
May 16 15:48:50 octopi xinit[1011]: time="2020-05-16T15:48:50+01:00" level=info
May 16 15:48:51 octopi xinit[1011]: time="2020-05-16T15:48:51+01:00" level=debug
May 16 15:48:51 octopi xinit[1011]: time="2020-05-16T15:48:51+01:00" level=error
May 16 15:49:01 octopi systemd[1]: Started OctoScreen.
```

But on the Screen I see the octopi login. How can I start octoscreen?

agentsmith29, 2020-05-19:
Did you activate automatic log-in?

duennerd, 2020-05-20:
No, I didnt activate it. 
When I activated it, I had an error where xorg did not find a display. Altought the display displayed the console. 
I followed a bunch of articels and also issue #43 helped a bit.
I've come this far that octoprint is displayed on the display, but not in the correct resolution.
The octoscreen interface is going off screen. 
I tried to set to the resolution from the screen which is 480x320. 

When I change the resolution i change it in 
/boot/config.txt 
```
framebuffer_width=480
framebuffer_height=320
hdmi_cvt 480 320 60 6 0 0 0 
```
and in /etc/octoscreen/config
OCTOSCREEN_RESOLUTION=480x320

But that didnt seem to have any effect. 

When I changed the resolution to 800x533 it was bigger (more zoomed) so I tried to set it to 240x160. But then it was like in the beginning. The overscan option in /boot/config.txt did not seem to have any affection.

Any ideas?

mdaneman, 2020-05-27:
I had a similar issue and, in my case, is that the octoscreen.service is configured to only start if there's a graphical environment. This means you need to have lightdm installed and use raspi-config to setup booting to desktop.
However, you don't have to do this.  I found the answer in another issue item ont this board.
All you have to do is edit the octoscreen.service config file: 
sudo nano /lib/systemd/system/octoscreen.service
Then find the [Install} section and change line
"WantedBy=graphical.target" to
"WantedBy=graphical.target multi-user.target"
and save the file.

Then disable and re-enable the service:
sudo systemctl disable octoscreen.service
sudo systemctl enable octoscreen.service

Now if you reboot, Octoscreen should come up automatically.

Protean-Man, 2020-05-27:
Brilliant. Thank you this worked. @mdaneman

JeffB42, 2020-12-04:
@duennerd Could you follow the steps listed in the wiki (located at https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen) and then post a reply to this issue whether the steps listed work and the issue is resolved?  I'd like to start closing out all the display-related issues.

If it doesn't work,
1) make sure you use a Raspberry Pi 3B or Raspberry Pi 4
2) start with a fresh install of OctoPi
3) please take a picture of the screen

Thanks, 

Jeff

duennerd, 2020-12-06:
Hey Jeff
Thank you for getting back to me. 
I installed it and the screen is displayed correctly. :-)
Althought the Touchscreen Input is wrong.
It seems that X and Y is reversed. 
I tried it with this calibration tool: 
sudo DISPLAY=:0.0 xinput_calibrator
As the tool sugests you need to input the result manually into the calibration file. 

I then tried to add the swap axes option inside those 3 files inside /usr/share/X11/xorg.conf.d
99-calibration.conf
99-fbturbo.conf
99-fbtrbo.conf.save

I tried "SwapAxes" "0" and "1". Both didnt work.
Do you happen to know how to solve that issue?
Maybe I have to add that option into another file? 
It is not so bad if you dont know. In the end I ended up not using octoprint at all. I was just curious if the screen was correctly displayed or not, which is the case. And of course if it would work I would use it of course. :-)

Protean-Man, 2020-12-06:
It is based on your screens installation files in your Raspberry Pi. Octoprint doesn't necessarily control the touch calibrations. 
It depends on your screen as well, so hopefully you can find info from the manufacturers website? Including raspberry pi install drivers and setup options.

Try this video, it goes through that scenario and a bit more: https://www.youtube.com/watch?v=OJ59hXSyBoI

Here is a Doc I made last year for people to install Octoscreen as well as other options that may help you:
https://docs.google.com/document/d/1VqGRo5G0RhrmktN6hTTUKQ6RQ4XaNL3gvYlSf5ktHAo/edit?usp=sharing

JeffB42, 2020-12-06:
@Protean-Man which screen do you use?  Do you have an Amazon link?  I'm consolidating and making a list of known working displays.

Thanks,

Jeff

Protean-Man, 2020-12-06:
> @Protean-Man which screen do you use? Do you have an Amazon link? I'm consolidating and making a list of known working displays.
> 
> Thanks,
> 
> Jeff

I use this: https://www.aliexpress.com/item/33024647905.html?spm=a2g0s.9042311.0.0.27424c4dfhwZGs
I think the DSI connector is important and is becoming more common. There are many more on aliexpress.com that have DSI so I a direct link may be outdated too soon?

Most screens on Amazon still use HDMI and USB connectors which aren't ideal.

JeffB42, 2020-12-06:
Thanks @Protean-Man - I'm in the process of updating the install and setup documents:

https://github.com/Z-Bolt/OctoScreen/wiki/Setting-Up-OctoScreen-and-Your-Display

https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen

I added a link to your setup guide, and I added a link to the display you posted.

Thanks!

Jeff

Protean-Man, 2020-12-06:
Thanks! Looks great. Really appreciate the work you're putting into this. I know many others do as well.
I'll make that guide look better then.

Protean-Man, 2020-12-07:
@JeffB42 
Ok I just did a total update on that Install & Trouble Shoot Doc. 
https://docs.google.com/document/d/1VqGRo5G0RhrmktN6hTTUKQ6RQ4XaNL3gvYlSf5ktHAo/edit?usp=sharing

Hope it helps someone.

JeffB42, 2021-01-02:
@duennerd is this still an issue?

Protean-Man, 2021-01-02:
I don't think it is.

duennerd, 2021-01-03:
@JeffB42 I wasn't able to resolve the problem, but I decided to not use Octoprint or Octoscreen in the end, so therefore it's not an issue anymore.

closed 2021-01-03 by JeffB42