Z-Bolt/OctoScreen #143: Feature Request: Turn screen off if TP-Link Smartplug is off

skibum-za, 2020-05-21:
I use a TP-Link Smartplug and the octoprint plugin. This turns off the power to my printer once the nozzle temp is below 50c and the printer has been idle for a time (30mins I think). Anyway, at this point the OctoScreen plugin just loops around trying to reconnect.

Any chance you could enhance Octoscreen, so that if the Smartplug is off, then it just turns the screen off and you have to touch the screen to re-enable it. 

And if possible, could you add the toggle that the TP-Link Smartplug plugin has to switch on/off the printer as well?

xunleii, 2020-09-15:
What about switching off Octoscreen when the printer is disconnected (after 15s for example) and switching on after a touch signal ? IMO, it could be easier to manage than creating dependencies between plugins

Hardestatd, 2021-01-13:
> What about switching off Octoscreen when the printer is disconnected (after 15s for example) and switching on after a touch signal ? IMO, it could be easier to manage than creating dependencies between plugins

I second this, would be a pretty good solution.