Z-Bolt/OctoScreen #144: No Octoscreen shown, but the Debian-Promt

freak1051, 2020-05-23:
Hello, 

i´m not that Linux-Junkie :) i Have installted like in Wiki were dokumented. At the point make build he told me :"make: *** No rule to make target 'build'.  Stop."
The anser on ls-l build/ is:
"ls: cannot access 'build/': No such file or directory"

i dont know, if i have doing something wrong. but unfortunately after rebooting in display is the Octoprin.promt (Terminal) (" pi@octopi:~$"

what am i doing wrong?

AtomicRas, 2020-05-25:
Same issue with this and TouchUI

rwaspe, 2020-05-25:
Try this:

cd ./OctoScreen

sudo make build

cd build/stretch

sudo dpkg -i octoscreen_2.5-1_armhf.deb

sudo reboot

Protean-Man, 2020-05-27:
Having the same Problem 

> Try this:
> 
> cd ./OctoScreen
> 
> sudo make build
> 
> cd build/stretch
> 
> sudo dpkg -i octoscreen_2.5-1_armhf.deb
> 
> sudo reboot

Having the same problem. Having to call each time I start:
systemctl start octoscreen

geoffrothman, 2020-05-27:
All you have to do  to have it autoload upon boot each time is edit the /etc/rc.local file and throw this in right before the last line:
"systemctl start octoscreen &"

JeffB42, 2020-12-04:
@freak1051 Could you follow the steps listed in the wiki (located at https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen) and then post a reply to this issue whether the steps listed work and the issue is resolved?  I'd like to start closing out all the display-related issues.

If it doesn't work,
1) make sure you use a Raspberry Pi 3B or Raspberry Pi 4
2) start with a fresh install of OctoPi
3) please take a picture of the screen

Thanks, 

Jeff

JeffB42, 2021-01-06:
@freak1051 Is this still an issue?  Did you get your display working?

JeffB42, 2021-02-04:
Closed due to inactivity.

closed 2021-02-04 by JeffB42