Z-Bolt/OctoScreen #145: Is this project abandoned?

mdaneman, 2020-05-27:
Is this project still being supported by Alex?   I like this app, but it still has a few annoying bugs. In particular the screen resizing when the filename is too long issue is quite a pain.  Is there still a plan to keep updating Octoscreen?

floridaservices, 2020-05-27:
I suspect you are correct.  I wanted to love octoscreen but it has some serious functionality weaknesses.  I have moved to octodash and installed it without too much fuss and it's been rock solid.  Maybe give it a try

mdaneman, 2020-05-27:
I actually tried OctoDash recently and wasn't a big fan.  I'm using a 3.5" resistive LCD and the touch response for OctoDash is pretty bad (much worse than OctoScreen). It also lacks some simple features like pushing a button to toggle bed or nozzle heating on and off (There are buttons to start heating, but to turn heating off you have to push those buttons, press "minus" multiple times to set temperature to zero, and press "set").  I also long ago tried TouchUI, which has the issue of not being designed for 3.5" touch screens and has poor touch response as well. I'd really prefer for OctoScreen to keep being debugged so it can be a more reliabiel interface.  If Alex no long has time for it, perhaps someone else can take it over?

fmatray, 2020-06-01:
Hello,
@CyrilleSouchet, @sercangunonu and other developers seems to continu this project since december into two different branches. Rather than having many forks, it would be nice (but not easy) to merge all work into one official package.

mdaneman, 2020-06-01:
I see those, but the @sercangunonu doesn't really clarify what it has changed/fixed, while @CyrilleSouchet seems to be a french translation.

sercangunonu, 2020-06-07:
@fmatray @mdaneman Hello guys. I customized OctoScreen for my companies product.I changed Turkish translation and some minor UI changes.

clue3030, 2020-06-08:
I wish they would just put this on the plugin manager in octoprint cause it’sa passion in the Ass to get this to work and install.

Protoncek, 2020-06-16:
@mdaneman : octodash has custom shortcuts, currently 6 of them and in there you can put your g code (or more of them, separated by ; ) i use those shourtcuts as presets for preheating PLA, PETG, cooldown, power off Pi.... check out wiki on that github site.
Version 2 is just about to come out any day now (as author promises) and it will be even better (so he says...).

clue3030, 2020-06-19:
I hope it’s a lot easier to install than what it is now. I still can’t get it to install and run. I did for about a day and I had to reboot and now can’t get the actoscreen to load or anything. So I killed it and went back to Repeiter-server.

Protoncek, 2020-06-19:
If i remember correctly you must enable auto login in console in ‘sudo raspi-config’ then it works.
I’m not sure if xorg is installed or you must install it manually... look at wiki pages:[https://github.com/UnchartedBull/OctoDash/wiki/Installation](url)

JeffB42, 2020-07-29:
@here well, good news, I contacted Aleksei and he granted me moderator rights.  I"m a develop and plan to work on this and Z-Bolt/OctoPrint-Z-Bolt-OctoScreen.  I can't work on this full time, and can't promise anything, but I plan to work on this a little.

floridaservices, 2020-07-29:
that is good news, thanks for taking up the cause!

closed 2020-07-31 by JeffB42