Z-Bolt/OctoScreen #147: Cancel/Stop Print Button Resets System

Protean-Man, 2020-05-31:
I am finding that when I cancel or stop a print from the Octoscreen command buttons the system resets and goes to the splash screen - detecting baud.. etc.

LizRidley, 2020-07-27:
I've only just started using this and have already seen the same.

If it does work without the reset then the screen doesn't re-display properly, its shifted to the right.

[image]

Protean-Man, 2020-07-28:
@LizRidley how long are your gcode names?
Try this, make two gcode files for the same exact print. One is a short name like "1" and the other super long - 6+ words and spaces. Print both and see if the screen shifts for both.
I has the same problem and found that long gcode names cause the shift. Short don't. 
I don't have a fix, yet. Maybe in the near future if I fork this and work with it.

rwaspe, 2020-07-28:
@Protean-Man I posted the fix for the file name length in #126.  Hope that helps.
https://github.com/Z-Bolt/OctoScreen/issues/126

LizRidley, 2020-07-28:
Brilliant thanks, my filename was really long and using a short one the screen shift doesn't happen.

It has though restarted Octoprint and the first time it cycled a few times before it connected again, the second it seemed to just disconnect and reconnect - it dropped my browser window on my Mac too.  Doesn't stop me using it though.

Will have a look at redefining the string length when I get some more time.

I'm glad its working as I tried another similar and the browser one and I didn't like either and so was a bit disheartened and wondered why I'd bothered with the screen.  This has enough features to be useful and good interface and makes the screen worthwhile.

Protean-Man, 2020-07-28:
Happy to hear at least one thing is solved. 
This project seems to be stuck or at least the owner hasn't updated for quite some time. I'm hoping there is something more in the works but your guess is as good as mine. 
Maybe someone will fork this and fix all the current issues or take over as owner...

JeffB42, 2020-08-18:
@Protean-Man I ran into the bug you described here and posted a workaround in https://github.com/Z-Bolt/OctoScreen/issues/164.  This will be fixed ion the next release.

JeffB42, 2020-09-17:
This is fixed in the upcoming 2.6 release.

closed 2020-09-17 by JeffB42