Z-Bolt/OctoScreen #149: Unit octoscreen.service is masked.

clue3030, 2020-06-08:
what does this mean, Unit octoscreen.service is masked?

JeffB42, 2021-03-30:
This issue was also reported here: 
https://github.com/Z-Bolt/OctoScreen/issues/235

I'm not familiar with masking, @thebeline does this sound familiar?  Any ideas as to what's going on?

@clue3030 it sounds like something is misconfigured pretty badly.  Maybe you used `sudo` when you didn't need to, or maybe you didn't use `sudo` when you needed to (maybe, I'm just guessing here).  What I would do is take a new SD card (or reuse your current one), flash the latest OctoPi, install and configure OctoPrint, and then run through the steps at https://github.com/Z-Bolt/OctoScreen/wiki/Setting-Up-OctoScreen-and-Your-Display and 
https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen

If after having installed clean and you get this same error, I did a google search and came across
https://medium.com/@codingmaths/service-masking-in-linux-f7265d9b2181 and https://unix.stackexchange.com/questions/308904/systemd-how-to-unmask-a-service-whose-unit-file-is-empty and they might help

thebeline, 2021-03-30:
So... This is literally explained in the deb files. `rules` I believe. IIRC it means it is linked to null.

thebeline, 2021-03-30:
https://www.github.com/Z-Bolt/OctoScreen/tree/master/debian%2Foctoscreen.postinst

TBF, we could just unlink it and re-link it. But I am not sure how that would effect other things. I will have to double-check.

Broad-strokes, though: i don't think it is an indication of something "super-broke", moreso I think something else that was installed just added the link and didn't clean it up.

@clue3030 - it's a few months out now, but do you recall adding anything prior to the build that may have done something like that?

thebeline, 2021-03-30:
Hmmm....  So, this is curious. 

I was able to reproduce the masked status by running `$ sudo apt remove octoscreen`, which is the only way (beyond explicitly calling `$ systemctl mask octoscreen.service`) to end up with OctoScreen being masked.

This means you "began" the process of Uninstalling OctoScreen.  I quote "began" because `remove` doesn't "finish the job," it just marks a package for removal.  To "complete" the job, you must call `purge`.

This also indicates that OctoScreen is not "installed".

This leads me to believe you HAD it installed, and were attempting an upgrade.  At some point during the install process, something must have gone wrong.

This should be able to be resolved by running the install process again.  Follow the [UpdateInstructions](https://github.com/Z-Bolt/OctoScreen/releases/tag/v2.7.0), you will not loose your settings.

If you still have issues, try running `systemctl unmask octoscreen.service` prior to following the upgrade instructions.

That being said, @JeffB42: The main `README` contains a link to the last version (`2.6.1`), and is also broken.

JeffB42, 2021-05-01:
@clue3030 any update on this?

closed 2021-05-24 by JeffB42