Z-Bolt/OctoScreen #152: No screen found

duennerd, 2020-06-21:
I'm trying to install octoscreen: (sorry for the long post - I tried to provide enough information)

I'm using this [screen](https://www.banggood.com/Geekcreit-3_5-inch-TFT-LCD-Touch-Screen-Protective-Case-Heatsink-Touch-Pen-Kit-For-Raspberry-Pi-32Model-p-1269846.html?rmmds=myorder&cur_warehouse=CN) and this [driver](url). 

I partly followed this [tutorial](https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-user-mods-octoprint-enclosures-nozzles-.../3-5-gpio-touchscreenrpi-3band-octoscreen/). 

First: burn 2019-09-26-octopi-buster-lite-0.17.0.img on SD Card with balenaEtcher

In putty:
```
sudo apt update
sudo apt full-upgrade
sudo apt-get install git build-essential xorg-dev xutils-dev x11proto-dri2-dev
sudo apt-get install libltdl-dev libtool automake libdrm-dev
sudo apt-get install libgtk-3-0 xserver-xorg xinit x11-xserver-utils
```

Then: 
```
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
sudo ./LCD35-show
```
I also tried with the [waveshare driver](https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A)) and the [xf86-video-fbturbo driver](https://github.com/ssvb/xf86-video-fbturbo), but with those two  I was unable to get the touchscreen working. That's why I wanted to try also with this one. 

Then: Octoscreen Installation:
```
wget https://github.com/Z-Bolt/OctoScreen/releases/download/v2.5.1/octoscreen_2.5-1_armhf.deb
sudo dpkg -r octoscreen
sudo dpkg -i octoscreen_2.5-1_armhf.deb
sudo systemctl restart octoscreen.service
```
I also installed lightdm and set the autologin to desktop and chose octoscreen in raspi-config. 

When I type 
`systemctl status octoscreen.service`
i get a "no screen found" error: 

```
pi@octopi:~ $ systemctl status octoscreen.service
● octoscreen.service - OctoScreen
   Loaded: loaded (/lib/systemd/system/octoscreen.service; enabled; vendor preset: enabled)
   Active: activating (start) since Sun 2020-06-21 13:35:51 BST; 11s ago
 Main PID: 1102 (xinit)
    Tasks: 2 (limit: 2200)
   Memory: 360.0K
   CGroup: /system.slice/octoscreen.service
           └─1102 /usr/bin/xinit /usr/bin/OctoScreen -- :0 -nolisten tcp -nocursor

Jun 21 13:35:52 octopi xinit[1102]: (EE)
Jun 21 13:35:52 octopi xinit[1102]: Fatal server error:
Jun 21 13:35:52 octopi xinit[1102]: (EE) no screens found(EE)
Jun 21 13:35:52 octopi xinit[1102]: (EE)
Jun 21 13:35:52 octopi xinit[1102]: Please consult the The X.Org Foundation support
Jun 21 13:35:52 octopi xinit[1102]:          at http://wiki.x.org
Jun 21 13:35:52 octopi xinit[1102]:  for help.
Jun 21 13:35:52 octopi xinit[1102]: (EE) Please also check the log file at "/var/log/Xorg.0.log" for addit
Jun 21 13:35:52 octopi xinit[1102]: (EE)
Jun 21 13:35:52 octopi xinit[1102]: (EE) Server terminated with error (1). Closing log file.
```
I can see the linux prompt. (pi@octopi:~$ _) on the screen. 

When I check the Xorg.0.log i get: 
```
X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
[  1679.984] Build Operating System: Linux 4.15.0-48-generic armv8l Raspbian
[  1679.985] Current Operating System: Linux octopi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l
[  1679.985] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_head$
[  1679.985] Build Date: 05 June 2019  12:49:54PM
[  1679.985] xorg-server 2:1.20.4-1+rpt1 (https://www.debian.org/support)
[  1679.985] Current version of pixman: 0.36.0
[  1679.985]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[  1679.985] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[  1679.986] (==) Log file: "/var/log/Xorg.0.log", Time: Sun Jun 21 13:49:05 2020
[  1679.987] (==) Using config directory: ""
[  1679.987] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[  1679.989] (==) No Layout section.  Using the first Screen section.
[  1679.989] (==) No screen section available. Using defaults.
[  1679.989] (**) |-->Screen "Default Screen Section" (0)
[  1679.989] (**) |   |-->Monitor "<default monitor>"
[  1679.992] (==) No device specified for screen "Default Screen Section".
        Using the first device section listed.
[  1679.992] (**) |   |-->Device "Allwinner A10/A13/A20 FBDEV"
[  1679.992] (**) |   |-->GPUDevice "touchscreen"
[  1679.992] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[  1679.992] (==) Automatically adding devices
[  1679.992] (==) Automatically enabling devices
[  1679.992] (==) Automatically adding GPU devices
[  1679.992] (==) Max clients allowed: 256, resource mask: 0x1fffff
[  1679.992] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[  1679.992]    Entry deleted from font path.
[  1679.993] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[  1679.993]    Entry deleted from font path.
[  1679.993] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[  1679.993]    Entry deleted from font path.
[  1679.993] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[  1679.993]    Entry deleted from font path.
[  1679.993] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[  1679.993]    Entry deleted from font path.
[  1679.993] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[  1679.993]    Entry deleted from font path.
[  1679.993] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
```
I tried a autoreconf from https://www.x.org/wiki/ConfigurationHelp/:
`sudo X -configure`
which didnt work. 

Inside /etc/X11/xorg.conf.d I just have the file 99-calibration.conf:
```
Section "InputClass"
        Identifier      "calibration"
        MatchProduct    "ADS7846 Touchscreen"
        Option  "Calibration"   "3936 227 268 3880"
        Option  "SwapAxes"      "1"
EndSection

Section "Device"
        Identifier "touchscreen"
        Driver "fbdev"
        Option "fbdev" "/dev/fb0"
EndSection
```
The second section "Device" I added by hand from the [tutorial. ](https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-user-mods-octoprint-enclosures-nozzles-.../3-5-gpio-touchscreenrpi-3band-octoscreen/)

Inside /usr/share/X11/xorg.conf.d I get the following files:
```
10-amdgpu.conf  10-quirks.conf  40-libinput.conf  70-wacom.conf
10-evdev.conf   10-radeon.conf  45-evdev.conf     99-fbturbo.conf
```

Is there a way to configure X to the correct screen instead of the default one or the first one in the list?

SConaway, 2020-07-21:
You could always remove the incorrect files. I'd guess I'd remove the files that start with `10-`.

In your `/etc/X11/xorg.conf.d/99-calibration.conf`, you might give `fb1` a try.

I don't have this screen so I can't try it myself, but I hope this helps.

JeffB42, 2020-07-31:
I followed the steps Chis listed in his video - https://www.youtube.com/watch?v=OJ59hXSyBoI&t=432s
If possible, use another memory card, flash a new/fresh image of OctoPi onto it, and then follow Chris's video.

Hope this helps,

Jeff

duennerd, 2020-08-09:
@SConaway 
its workin with fb1 :-) 
But the interface does still not fit the screen. (The interface is bigger than the screen and gets cut off.)
I tried with differenc screen size settings in /boot/config.txt and /etc/octoscreen/config

Here my /boot/config.txt: 
```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
framebuffer_width=800
framebuffer_height=533

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
#max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
hdmi_force_hotplug=1
dtparam=i2c_arm=on
dtparam=spi=on
enable_uart=1
dtoverlay=tft35a:rotate=270
hdmi_cvt 800 533 60 6 0 0 0
```

And my /etc/octoscreen/config:

```
# Location of the OctoPrint's config.yaml file. If empty the file will
# be search at the `pi` home folder or the current user. Only used for locally
# installed OctoPrint servers.
OCTOPRINT_CONFIG_FILE=/home/pi/.octoprint/config.yaml

# OctoPrint HTTP address, example `http://localhost:5000`, if OctoPrint is
# locally installed will be read from the config file.
OCTOPRINT_HOST=http://localhost:5000

# OctoScreen expects an API key to be supplied. This API key can be either
# the globally configured one or a user specific one if  ^`^|Access Control ^`^}.
# http://docs.octoprint.org/en/master/api/general.html, if OctoPrint is
# locally installed will be read from the config file.
OCTOPRINT_APIKEY=

# Location of the application theme.
OCTOSCREEN_STYLE_PATH=/opt/octoscreen/styles/z-bolt/

# Resolution of the application, should be configured to the resolution of your
# screen, for example 800x480.
OCTOSCREEN_RESOLUTION=800x533

# Location of file for logging (optional)
OCTOSCREEN_LOG_FILE=
```

Any suggestions what to change?

@JeffB42 Chris uses a different screen in his video. I tried with his tutorial but had no success.

JeffB42, 2020-08-09:
@SConaway - I just replied to that other thread, but here's what I found out... I have some bad news - I ran some experiments and from what I've seen, it's not possible on that resolution.  It appears that the absolute minimum resolution OctoScreen needs is 584x348.  OctoScreen is built on GTK, and GTK has an annoying habit where if there's too much content to display, and not enough pixels, it resizes the window.  In our case the window is the entire size of the screen, and when the screen/window expands, it give the appearance of shifting things around.

I myself have a Pimoroni Hyperpixel 4, which has a resolution of 800x480 and has capacitive touch.  I know this isn't the answer you were wanting, but maybe use your current screen for another project, and buy a Hyperpixel (or another display that's 800x480, or at least greater than 584x348)

Jeff

SConaway, 2020-08-09:
@duennerd I'm not sure. I too have a 320x480 display and use those settings. 

What do you mean by 
> But the interface does still not fit the screen. (The interface is bigger than the screen and gets cut off.)

Could you please post an image? 

You also could try changing the resolution configured in `/boot/config.txt` and `/etc/octoscreen/config` to `480x320` and see if that helps.

SConaway, 2020-08-09:
@JeffB42 in response to the comment that I received via email (I guess you deleted it):

I think that's why we set the resolution as `800x533` and let it scale to the screen's native resolution. 

It could be that the driver doesn't support scaling, and instead, just chops off anything around the edge.

@duennerd just thought of something:
Change `overscan_right` and `overscan_bottom` in `/boot/config.txt`. You may be able to "force" the content on the display (leave `800x533` everywhere though)

duennerd, 2020-08-09:
@SConaway 
I tried to play a bit with the resolution settings in both file between 800x533 and 480x320. In one setting, I think it was 480x320 in both files, it was a bit better as you can see in the images, but still not good.
Here the images:

[image]
[image]
[image]

I will try with the overscan settings next week. 
What do you mean by "force"? Which setting are you refering to?

JeffB42, 2020-08-09:
@SConaway and @duennerd - duennerd already posted some images, but here's some  info which might shed some light.  I have a RPi4+Hyperpixel running OctoScreen, but for development I'm running locally off my computer.  OctoScreen is written using GTK (a GUI library) and I've witnessed that if there aren't enough pixels to display all of the UI elements, GTK will resize the screen.

(I've already encountered this, where I added a new image, it was slightly too large and caused the window/screen to expand, and I had to go back and resize the image to a smaller size)

Running on my computer's desktop, when there isn't enough space to draw everything, the window is resized.  On a RPi running with a touchscreen display, the screen/window/display are practically the same, so when there aren't enough pixels, GTK increases the window (AKA drawing area) and to us, it looks like things are shifted and are offscreen.

I ran some tests, and when I set the screen size to 480x320, I initially (and briefly) see this:
[image]

...and then, once the UI has rendered, the window is resized and I see this:
[image]

This looks to be around 548x348 (+/- a pixel or two), and this tells me that OctoScreen requires a minimum resolution of 548x348.

SConaway - re. deleted message: yes, sorry about that.  I replied to a few issues this morning, saying basically "from what I've seen 548x348 (ish) is the minimum resolution needed, and won't run on 480x320".  However... 1) I read through a few messages and some people reported that they _were_ able to get OctoScreen to run in 480x320, and 2) the main page for OctoScreen explicitly states that is does support, and runs in 480x320, so I decided to delete my messages.

I don't own a 480x320 display (just my 800x640 Hyperpixel) so it's impossible for me to troubleshoot it.  The information on this is a mess... lots of tips, posted across many tickets, with some reporting it works and others reporting it doesn't.  I wish someone would create a guide on how to set up for 480x320 and post it to the wiki (or send me it and I'll post to the wiki)

JeffB42, 2020-08-09:
@duennerd - one other comment/suggestion (and you aren't going to like it), is if this is becoming too much of a time suck, you might want to buy a Hyperpixel or some other 800x640 display, and use the Hyperpixel for OctoScreen and use your current display for another project.  Not the best advice, I know, but a Hyperpixel display (and presumably others) aren't too expensive, and at some point time and frustration vs money need to be considered.

Jeff

PS.  I still would love it if you got your current 480x320 display to work, and then posted the fix.
PPS.  I talk up the Pimori Hyperpixel a lot.  I'm not affiliated with them in any way, it's just that I followed Chris Riley's set up video, and it worked w/o too much fuss, and I've been happy with it.

SConaway, 2020-08-09:
@JeffB42 no worries. 

I much rather would have bought a Hyperpixel too, but Adafruit was out of stock of them at the time. Purchasing one could be an option for duennerd if he'd like.

@duennerd  —

If you play around with the `overscan_*` settings, you might be able to get this to work. Try increasing those values (bottom and right), and it might cause the display to scale successfully. Sorry I can't test it: I don't have one of those exact displays.

duennerd, 2020-08-12:
@JeffB42 @SConaway 
So I did some testing the last two days and I haven't been able to go smaller than shown above in the pictures. Even with the overscan settings. I tried with numerous settings and also tried with different hdmi_group=1 and hdmi_mode=1 but with no luck. 
So I tried to install it on a new sd card and try with a clean install. I wasnt able to get shown anything on the screen. 
Just the backlight comes up and the screen stayed blank/white.
So I inserted the old/other SD card with the "old" installation, with the same result (white screen). (This SD card should at least show the linux terminal.)
Sooo ... the screen is probably for the trash. :-(
(I will maybe try to install raspbian as a last try to be sure that the screen really is broken.)
Thanks for the support. 
(If I will by any chance be able to get it working I will post the solution here.) 
-> Off to buying a hyperpixel :-D

closed 2020-08-16 by duennerd

pyrophreak, 2021-11-04:
> 
> 
> I followed the steps Chis listed in his video - https://www.youtube.com/watch?v=OJ59hXSyBoI&t=432s If possible, use another memory card, flash a new/fresh image of OctoPi onto it, and then follow Chris's video.
> 
> Hope this helps,
> 
> Jeff

Chris Video is outdated and links dont work. I posted on twitter asking him to update or enable comments to allow people to ask questions. im having similar issues with Octoscreen not finding any screen.

fenixryan, 2024-02-26:
Howdy I know the thread is quite old but wanted to put in my fix. Got the cheapo boards my self
I used a combination of
https://forum.prusa3d.com/forum/original-prusa-i3-mk3s-mk3-user-mods-octoprint-enclosures-nozzles/3-5-gpio-touchscreenrpi-3band-octoscreen/
and
https://cryptpad.fr/code/#/2/code/view/h7L6PdPs76CsEPwykNJ1rTI+w3p+2ZlpLwfhCxnuLIc/

Rebooted, the screen came up but then sat at a command prompt.  

After running sudo systemctl restart octoscreen it falled to reload suggesting I look at  "systemctl status octoscreen.service" and "journalctl -xe" for details.

And there it was line 7 - /etc/X11/xorg.conf.d/99-calibration.conf

Feb 26 19:10:38 jtbLCD xinit[1278]: (==) Using config directory: "/etc/X11/xorg.conf.d"
Feb 26 19:10:38 jtbLCD xinit[1278]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Feb 26 19:10:38 jtbLCD xinit[1278]: Parse error on line 7 of section InputClass in file /etc/X11/xorg.conf.d/99-calibration.conf
Feb 26 19:10:38 jtbLCD xinit[1278]:         The Section keyword requires a quoted string to follow it.
Feb 26 19:10:38 jtbLCD xinit[1278]: (EE)

Problem with copy and paste from website, the quotation marks were garabge. Replaced them in nano with correct ones and good to go. Screen came up.

Now off to fix me klipper screens :)