Z-Bolt/OctoScreen #154: "Octoscreen starting" and then nothing happens

cryptodonniedxb, 2020-06-25:
My octoscreen worked as a charm, but since yesterday night, it doesn't start

this is what happens on the screen
nothing more
[image]

any ideas?

Moorviper, 2020-06-26:
[image]

cryptodonniedxb, 2020-06-26:
sorry i'm a super noob.
how do i get logs?

Moorviper, 2020-06-27:
https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299

JeffB42, 2020-07-31:
I followed the steps Chis listed in his video - https://www.youtube.com/watch?v=OJ59hXSyBoI&t=432s
If possible, use another memory card, flash a new/fresh image of OctoPi onto it, and then follow Chris's video.

Hope this helps,

Jeff

JeffB42, 2021-02-04:
@cryptodonniedxb is this still an issue?

cryptodonniedxb, 2021-02-04:
unfortunately yes
it was also very difficult to make octoscreen start, therefore i wish not to reinstall octopi and octoscreen
logs, it's difficult for me to extract

JeffB42, 2021-02-04:
I suggest reading through https://github.com/Z-Bolt/OctoScreen/wiki/Setting-Up-OctoScreen-and-Your-Display and https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen.  Then, flash the latest version of OctoPi onto a new SD card, and then follow the steps listed.  I've gone over the steps several times with several installs and if you follow the steps exactly, I think they work and you shouldn't have any issues.  Please try installing on a new SD card and let me know how it goes.

closed 2021-03-27 by JeffB42