Z-Bolt/OctoScreen #155: Octoscreen shifting to the right

hayder9445, 2020-07-03:
Hi, octoscreen seems to shift at different rates to the right, sometimes hidding whole buttons, the only way to restore everything is just to unplug the rbp. How can I fix this issue?

rwaspe, 2020-07-03:
Is it when you print things with longer file names?

https://github.com/Z-Bolt/OctoScreen/issues/126#issuecomment-633234573

hayder9445, 2020-07-03:
I'm now selecting prints directly from octoscreen and it happens more often yes, but before I use to always send my prints from cura, and it still happened.

hayder9445, 2020-07-03:
ok so I followed instructions for my case ( already installed ), but when I use sudo make build I get this error:

mkdir -p /home/pi/OctoScreen/build
docker build \
        --build-arg IMAGE=golang:1.9-stretch \
        --build-arg GO_TAGS=gtk_3_22 \
        -t mcuadros/octoprint-tft-build:stretch . \
        && \
docker run -it --rm \
        -v /home/pi/OctoScreen/build/stretch:/build \
        mcuadros/octoprint-tft-build:stretch \
        make build-internal
/bin/sh: 1: docker: not found
make: * [Makefile:51: STRETCH] Error 127

rwaspe, 2020-07-03:
How many characters are the file names you are printing? That is what is causing the bug that I linked above.

hayder9445, 2020-07-03:
The largest is 25, but can the bug appear even if it's sent from CURA? But anyway, how can I solve the error 127? I understand that it's an error when the path is inaccurate, but I did all the previous commands successfully

hayder9445, 2020-07-03:
sorry nevermind, the mistake was mine, I'm making the build now, let's see if it fixes the issue

closed 2020-07-06 by hayder9445