Z-Bolt/OctoScreen #156: Gcode files from inserted sd card printed via Octoprint doesn't let z -axis move

Kefier, 2020-07-07:
Hi Guys,

i am currently experiencing a crazy issue with octoprint.

I want to print from a gcode file. The gcode tells the printer at first to heat everything up, home the x axis, then the y axis and then to move the bed near the nozzle so the print can start.

When i put this gcode file onto a sd card and insert it into the printers card slot, then choose the file via the build in touchscreen of the Tronxy, everything works perfectly. It also works perfectly when i open Octoprints Webinterface with a Windows PC and directly upload the file from my computer to octoprint.

However, when i put the gcode file onto the sd, put it in my printer's card slot and then choose it from the filemanager and print the file via Octoprints Webinterface with the PC, it corrupts the whole print. At first, the printer heats up and homes the x and y axis, but the bed won't move. The motors of the z - axis are not working. The printer ignores that and resumes the print in the air. After that, when i am canceling the print and then try to restart it the same way, the printer just keeps resuming on the last position. But i don't know why, i checked all settings and there is nothing telling the printer to resume. this issue is portrayed by the message in the terminal of octoprint as:

Recv: //restore file position:0 - 0 = 0

I can move the z -axis with no problems manually with octoprint. It just seems that something is overriding the commands for the z- axis motors or something like that.

Here is a list of things i already did to fix this problem:

Reinstalled Octoprint on different SD cards several times

Downgraded to older versions

Deactived plugins / started Octoprint in safemode

Installed Fix CBD Firmware

Changed all Cables (USB and power cable of Raspberry)

Changed the SD Cards

Changed the Raspberry Pi

I also checked the settings of octoprints to see if something is adressing the motors for the z axis but i can't see anything in terms of that. I hope someone has experienced the same strange issue.

Using:

Tronxy X5SA Pro

Raspberry 3 A+

Octoprint Version: 1.4.0

OctoPi: 0.17.0

Protean-Man, 2020-07-20:
First - you should find your answers here: https://community.octoprint.org/
This is for Octoscreen which I don't think is part of the Tonxy printers. You can even try Tronxy forums.

I was also a little confused when reading your description: when using the Octoprint it is usually standard to upload the GCode file through the web interface, which saves it to your raspberry pi SD card. If you do this and print from the web interface does it print normally?

Also, is this a single GCode file? That keeps causing the ussues? 
Have you tried re-slicing the file and printing from the new GCode file? There have been times when the slicer has a hiccup when creating the file. If you can move the printer through octoprint and the printer screen normally then it is probably the GCode file.

JeffB42, 2021-03-27:
@Kefier this doesn't appear to be an OctoScreen issue.  If I'm wrong, please install the latest (2.7.0) and try to isolate the issue.

closed 2021-03-27 by JeffB42