Z-Bolt/OctoScreen #157: 480x320 display

niklask52t, 2020-07-07:
I have a 3,5 Inch display with the resolution 480x320, but when i change the resolution in the config, the picture is still shifted.

Protean-Man, 2020-07-20:
Shifted in what way?

SConaway, 2020-07-21:
I'd guess he's referring to overscan.

@niklask52t disable overscan by running `sudo raspi-config`, then navigate to `7 Advanced Options` > `A2 Overscan` and turn it off by choosing now, then finally reboot the pi. 

Also, make sure your `/boot/config.txt` is configured correctly. 
- it should have a display size specified in `. For my 480x320 display, I have `hdmi_cvt 800 533 60 6 0 0 0` at the bottom. 
- it should have a framebuffer set. Again, I use this placed near line 20:
```
framebuffer_width=800
framebuffer_height=533
```

Hope this helps.

rwaspe, 2020-07-28:
Also check #126.   https://github.com/Z-Bolt/OctoScreen/issues/126

JeffB42, 2020-08-09:
@niklask52t I have some bad news - I ran some experiments and from what I've seen, it's not possible on that resolution.  It appears that the absolute minimum resolution OctoScreen needs is 584x348.  OctoScreen is built on GTK, and GTK has an annoying habit where if there's too much content to display, and not enough pixels, it resizes the window.  In our case the window is the entire size of the screen, and when the screen/window expands, it give the appearance of shifting things around.

I myself have a Pimoroni Hyperpixel 4, which has a resolution of 800x480 and has capacitive touch.  I know this isn't the answer you were wanting, but maybe use your current screen for another project, and buy a Hyperpixel (or another display that's 800x480, or at least greater than 584x348)

Jeff

JeffB42, 2020-12-04:
@niklask52t  Could you follow the steps listed in the wiki (located at https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen) and then post a reply to this issue whether the steps listed work and the issue is resolved?  I'd like to start closing out all the display-related issues.

If it doesn't work,
1) make sure you use a Raspberry Pi 3B or Raspberry Pi 4
2) start with a fresh install of OctoPi
3) please take a picture of the screen

Thanks, 

Jeff

pearlythepirate, 2020-12-24:
@JeffB42 I was able to get the screen to show up correctly using the steps in the wiki, however, the touch interface did not work correctly. When pressing the screen, the touch registers in a completely different part of the screen. Do you have any recommendations for this issue?

SConaway, 2020-12-24:
@pearlythepirate is your touchscreen a resistive or capacitive one? If it's resistive, you can follow steps similar to the ones in this guide: https://learn.adafruit.com/adafruit-pitft-3-dot-5-touch-screen-for-raspberry-pi/touchscreen-install-and-calibrate

Protean-Man, 2020-12-24:
Which screen are you using?
Typically the manufacturer will give instructions on adjusting the touch point matrix.
If you check this DOC you can find a section about adjusting the touch matrix: https://docs.google.com/document/d/1VqGRo5G0RhrmktN6hTTUKQ6RQ4XaNL3gvYlSf5ktHAo/edit#

JeffB42, 2020-12-26:
@pearlythepirate , other than what @SConaway and @Protean-Man have already mentioned, the only other advise I have would be:
* Chris Riley's (Chris's basement) video on how to set up OctoScreen with a Pimoroni Hypixel.  The video is specific to the Hypixel, but he did run into an issue with rotation and getting the display to match the touch interface, and I suggest watching his video and trying some of the things that he suggests.  The link to his video is https://www.youtube.com/watch?v=OJ59hXSyBoI&t=1s, and the link to his set up doc is https://docs.google.com/document/d/1VfDyTq252wkcG9_D4Nh2lTRCSq51iltMDkG5AJ3J9jQ/edit
* Charlie Steenhagen's (Modern Hobbyist) video on setting up OctoScreen with an Anycubic i3.  He also lists steps on setting up the display along with screen rotation.  The link to his video is https://www.youtube.com/watch?v=1thsrIO64Ig, and the link to his set up doc is https://www.modhobbyist.com/projects/17
* Seek out the manufacturer's web site and see if you can find any documentation for your display.
* And if all else fails, contact the manufacturer.

Please post your progress, whether you find a solution, and what the solution is.

Jeff

pearlythepirate, 2020-12-28:
@SConaway @Protean-Man @JeffB42 thank you all for your replies. You all pointed me down the right path!
For my Adafruit PiTFT 3.5", the missing ingredient was found in Chris Riley's work:
```
Edit the touch screen matrix to match our 90 degree change:
sudo nano /usr/share/X11/xorg.conf.d/40-libinput.conf

Changes I made to the file:

Section "InputClass"
        Identifier "libinput touchscreen catchall"
        MatchIsTouchscreen "on"
        MatchDevicePath "/dev/input/event*"
        Driver "libinput"
        Option "TransformationMatrix" "0 -1 1 1 0 0 0 0 1"
EndSection
```
The TransformationMatrix line fixed it immediately!! Thanks again.

closed 2021-01-05 by JeffB42