Z-Bolt/OctoScreen #159: Issues creating a new OctoScreen

K1W13D, 2020-07-16:
Hi people, I am new to Raspberry Pi and struggling to setup an OctoScreen for my sons Prusa MK3

I religiously followed the below instructions, with the exception that my LCD is a 3.5" Geekcreit TFT

https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-user-mods-octoprint-enclosures-nozzles-.../3-5-gpio-touchscreenrpi-3band-octoscreen/

pi@octopi:~ $ octodash

(octodash:1022): Gtk-WARNING **: 10:40:36.393: cannot open display: 
pi@octopi:~ $ OctoScreen
INFO[0000] OctoPrint's config file found: "/home/pi/.octoprint/config.yaml"  file=main.go func=main.readConfig line=102
INFO[0000] Using "http://localhost:5000" as server address  file=main.go func=main.init.0 line=52
INFO[0000] Found API key at "/home/pi/.octoprint/config.yaml" file  file=main.go func=main.init.0 line=58
Unable to init server: Could not connect: Connection refused

(OctoScreen:1034): Gtk-WARNING **: 10:40:50.220: cannot open display: 

It's getting more than a little frustrating trying to et this up and running

Happy to provide what ever files / info you guys need to help me .....looking forward to getting this running

Cheers

Ray

Protean-Man, 2020-07-20:
Does the display work on the Pi with a fresh install? Showing cmd line and startup scripts?

JeffB42, 2020-07-29:
I followed the steps Chis listed in his video - https://www.youtube.com/watch?v=OJ59hXSyBoI&t=432s
If possible, use another memory card, flash a new/fresh image of OctoPi onto it, and then follow Chris's video.

Hope this helps,

Jeff

JeffB42, 2020-08-18:
Ok, it's been 20 days now.  I'm going to close this ticket.

closed 2020-08-18 by JeffB42