Z-Bolt/OctoScreen #162: When I install octoscreen I get this error and my HDMI screen blinks on and off every 5 seconds while the TFT remains all white.

scifiaholic, 2020-08-08:
(Reading database ... 60361 files and directories currently installed.)
Preparing to unpack octoscreen_2.5-1_armhf.deb ...
Unpacking octoscreen (2.5-1) over (2.5-1) ...
Setting up octoscreen (2.5-1) ...
Job for octoscreen.service failed because the service did not take the steps required by its unit configuration.
See "systemctl status octoscreen.service" and "journalctl -xe" for details.

JNSmith1717, 2020-08-16:
I have the same issue, been for a while i thought it was first with the tft i was using through the GPIO and now got me a 800x480 Hdmi lcd and the same thing

JeffB42, 2020-08-18:
"...HDMI screen blinks on and off every 5 seconds while the TFT..."

It sounds like you are using two displays at the same time (?)  If so, for the sake of trouble shooting, try using just the 800x480 TFT.  Perhaps the dual display is causing the issue.  Don't know if that's the case or not, but just a thought.

JeffB42, 2021-01-02:
@scifiaholic is this still an issue?

JNSmith1717, 2021-01-04:
Nope gave up

closed 2021-01-04 by JeffB42