Z-Bolt/OctoScreen #165: Packages not found while downloading

tiagomaricate, 2020-08-18:
Tring to install and....

Err:12 http://raspbian.raspberrypi.org/raspbian buster/main armhf dbus-user-session armhf 1.12.16-1
  404  Not Found [IP: 93.93.128.193 80]
Err:14 http://raspbian.raspberrypi.org/raspbian buster/main armhf glib-networking-common all 2.58.0-2
  404  Not Found [IP: 93.93.128.193 80]
Err:15 http://raspbian.raspberrypi.org/raspbian buster/main armhf glib-networking-services armhf 2.58.0-2
  404  Not Found [IP: 93.93.128.193 80]
Err:17 http://raspbian.raspberrypi.org/raspbian buster/main armhf glib-networking armhf 2.58.0-2
  404  Not Found [IP: 93.93.128.193 80]
Obter:10 http://archive.raspberrypi.org/debian buster/main armhf libgtk-3-common all 3.24.5-1+rpt2 [3.680 kB]
Err:21 http://raspbian.raspberrypi.org/raspbian buster/main armhf libcups2 armhf 2.2.10-6+deb10u1
  404  Not Found [IP: 93.93.128.193 80]
Err:39 http://raspbian.raspberrypi.org/raspbian buster/main armhf libinput-bin armhf 1.12.6-2                                                                         
  404  Not Found [IP: 93.93.128.193 80]
Err:41 http://raspbian.raspberrypi.org/raspbian buster/main armhf libinput10 armhf 1.12.6-2                                                                           
  404  Not Found [IP: 93.93.128.193 80]
Obter:18 http://mirror.nbtelecom.com.br/raspberryPI/raspbian buster/main armhf libatk1.0-data all 2.30.0-2 [145 kB]                                                   
Err:43 http://raspbian.raspberrypi.org/raspbian buster/main armhf libunwind8 armhf 1.2.1-9                                                                            
  404  Not Found [IP: 93.93.128.193 80]
E: Falhou ao buscar http://raspbian.raspberrypi.org/raspbian/pool/main/d/dbus/dbus-user-session_1.12.16-1_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Falhou ao buscar http://raspbian.raspberrypi.org/raspbian/pool/main/g/glib-networking/glib-networking-common_2.58.0-2_all.deb  404  Not Found [IP: 93.93.128.193 80]
E: Falhou ao buscar http://raspbian.raspberrypi.org/raspbian/pool/main/g/glib-networking/glib-networking-services_2.58.0-2_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Falhou ao buscar http://raspbian.raspberrypi.org/raspbian/pool/main/g/glib-networking/glib-networking_2.58.0-2_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Falhou ao buscar http://raspbian.raspberrypi.org/raspbian/pool/main/c/cups/libcups2_2.2.10-6+deb10u1_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Falhou ao buscar http://raspbian.raspberrypi.org/raspbian/pool/main/libi/libinput/libinput-bin_1.12.6-2_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Falhou ao buscar http://raspbian.raspberrypi.org/raspbian/pool/main/libi/libinput/libinput10_1.12.6-2_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Falhou ao buscar http://raspbian.raspberrypi.org/raspbian/pool/main/libu/libunwind/libunwind8_1.2.1-9_armhf.deb  404  Not Found [IP: 93.93.128.193 80]

JeffB42, 2020-08-21:
Are you trying to install Rasbian?  Or are you trying to install OctoScreen?  From your log, it looks like you are still trying to install Rasbian.  What are the commands that you are running?

savenko-egor, 2020-08-21:
I think you need run `apt update` before `apt install`.

closed 2020-09-11 by JeffB42