Z-Bolt/OctoScreen #170: Turn off TFT after some time, reactivate on touch

OevreFlataeker, 2020-08-25:
I am using OctoScreen with a Waveshare HDMI LCD(B) 800x480 touch panel and it works great. 
The only disadvantage with this display is, that is has no power button so it cannot be turned off. From the README I see that you per default turn screenblanking explicitly off. Why is this? Wouldn't it be better to let the display blank out after 5 min of inactivity so to conserve the display and protect from burn-ins as well as to safe power?

I'd like to be able to blank the screen and it should be possible to reactivate it on a touch of the screen. 

I am not sure if this feature request is something that needs to be implemented in OctoScreen itself or whether the same can be achieved just by configuring the X server appropriately. Please advise the best approach. Thank you!

NeoGee, 2020-08-26:
I would like to see an ability to blank the screen too.
Configuring X server doesn't look to work, as OctoScreen keeps polling for the printer, and updating the screen, so it never times out.

JeffB42, 2020-09-08:
Have you tried the solutions already posted?  https://github.com/Z-Bolt/OctoScreen/issues/132 or https://github.com/Z-Bolt/OctoScreen/issues/96

JeffB42, 2020-09-08:
I'm going to close this ticked since this topic has already been answered.  if neither of those solutions work, please open this ticket back up and post what happened after trying either of those solutions.

closed 2020-09-08 by JeffB42