Z-Bolt/OctoScreen #173: Configure polling interval / stop after too many failures

OevreFlataeker, 2020-09-02:
Please refer to https://github.com/OctoPrint/OctoPrint/issues/3532 for a topic already known it seems.
I'd welcome a solution which would make the following activities be configurable on the OctoScreen side in order to remedy this issue:

* Configure polling interval in general
* Cease attempts to read data after X consecutive failures, require user to press a button to restart the connection attempts

Knowing the 'behind the scenes details' now from the various issue reports, I have to say that this currently is hindering me from using OctoScreen which is a pity, because I REALLY like the idea and presentation and know spend 100 bucks for a nice LCD...

Thank you for considering my feature request.

OevreFlataeker, 2020-09-02:
... I never developed in Go but have cloned the repo and looked at it statically in VS Code...

Is it possible that these references are the actual parts where the whole stuff is setup?

ui.go, line 81: ui.b = NewBackgroundTask(time.Second*2, ui.update)    --> Setup a 2 second poll interval

This code looks to me as if there was actually already some kind of error handling included?
```Go
func (m *UI) update() {
	if m.connectionAttempts > 8 {
		m.s.putOnHold()
		return
	} else if m.UIState == "splash" {
		m.connectionAttempts += 1
	} else {
		m.connectionAttempts = 0
	}

	if m.OctoPrintPlugin {
		m.checkNotification()
		m.loadSettings()
	}

	m.verifyConnection()
}
```

splash.go
```Go
func (m *SplashPanel) putOnHold() {
	m.RetryButton.Show()
	ctx, _ := m.RetryButton.GetStyleContext()
	ctx.RemoveClass("hidden")
	m.Label.SetText("Cannot connect initialize the printer. Tap \"Retry\" to try again.")
}
``` 

So the question is... is that code that seems to be needed already present? Also in the latest package for the RPI?

JeffB42, 2020-09-08:
@OevreFlataeker Is the issue that it polls to frequently, and should be changed from 2 seconds to 10 seconds?

JeffB42, 2021-02-04:
Closed due to inactivity.

closed 2021-02-04 by JeffB42