Z-Bolt/OctoScreen #175: After installation screen opening and closing repeatedly.

eksglox, 2020-09-14:
I followed installation guide on Github after everything installed OctoScreen didn't start. Screen opening and closing repeatedly and OctoScreen UI didn't open. When I install deb. package console give that response : 

```
pi@octopi:~ $ sudo dpkg -i octoscreen_2.5-1_armhf.deb
Selecting previously unselected package octoscreen.
(Reading database ... 58402 files and directories currently installed.)
Preparing to unpack octoscreen_2.5-1_armhf.deb ...
Unpacking octoscreen (2.5-1) ...
Setting up octoscreen (2.5-1) ...
Created symlink /etc/systemd/system/display-manager.service → /lib/systemd/system/octoscreen.service.
Created symlink /etc/systemd/system/graphical.target.wants/octoscreen.service → /lib/systemd/system/octoscreen.service.
Job for octoscreen.service failed because the service did not take the steps required by its unit configuration.
See "systemctl status octoscreen.service" and "journalctl -xe" for details.
```

I use Raspberry Pi Zero and 3.5 inc. Waveshare touch screen (480x320).

the1snm, 2020-09-17:
For me when this happened during the install I needed to change fb1 to fb0 in the 99-fbdev.conf and it worked for me. I am using an SPI hat versus a HDMI so your mileage may vary. What ever you have setup it's worth a shot flipping them to see.

sudo nano /etc/X11/xorg.conf.d/99-fbdev.conf
Change: Option "fbdev" "/dev/fb1" to Option "fbdev" "/dev/fb0"

JeffB42, 2021-01-02:
@eksglox is this still an issue?

HonskiDesign, 2021-01-03:
I'm currently having this issue. Using an Elecrow 5 Inch 800x480 TFT Display, so not sure if that has anything to do with it.

Also could not get xserver to install without doing an apt upgrade.

The above '99-fbdev.conf' file is not present on my system.

Might try and redo the whole thing from a fresh flash of OctoPrint as I may have done some things out of order just to get the installs repaired and working. Will keep you posted.

HonskiDesign, 2021-01-03:
I keep getting 'x11-server-utils' has no installation candidate.

JeffB42, 2021-01-03:
Hmm.. I'm able to install w/o any issues.  Here are the steps I took:
1. Downloaded the latest OctoPi image, 0.17.0 from https://octoprint.org/download/
2. Flashed the image onto a new SD card
3. On the newly-flashed SD card, I opened /boot/octopi-wpa-supplicant.txt and edited the WPA network settings
4. Ejected the SD card, inserted it into my RPi 4, and then booted the RPi
5. SSH'd into the RPi
6. ran `sudo raspi-config`
7. (back on my desktop machine) I opened a browser window and went to http://octopi.local
8. I then ran through OctoPrint's new configuration wizard
9. Back in my SSH session, I ran `apt-get update`
10. I then ran `sudo apt-get install libgtk-3-0 xserver-xorg xinit x11-xserver-utils`
    -> everything installed fine
10. I then download OctoScreen 2.6.1, via `wget https://github.com/Z-Bolt/OctoScreen/releases/download/2.6.1/octoscreen_2.6.1_armhf.deb`
11. And then installed the package via `sudo dpkg -i octoscreen_2.6.1_armhf.deb`

@HonskiDesign If you run apt-get update and then install the video libraries, do you still run into issues?  If so, I suggest flashing a new SD card with the latest version of OctoPi and run the steps I listed above.

Jeff

HonskiDesign, 2021-01-03:
Everything seemed to go fine until step 12 where I recieved the error "Job for octoscreen.service failed because the control process exited with error code."

However, I put a step between 9 and 10 where I installed the touch screen software from here. https://github.com/Elecrow-RD/Elecrow-LCD5

Thanks for your quick replies!

HonskiDesign, 2021-01-04:
Ok, finally have it up and running without the screen jumping back and forth between Bash and Octoscreen every 15 seconds or so. I went through the whole process yet again and installed the Elecrow Driver package between steps 6 and 7. It still happens every ~2 mins or so. I find these messages on the Octoprint Terminal when this happens. I'm assuming it's because I don't have my printer plugged into the Pi yet.

`Changing monitoring state from "Offline" to "Detecting serial port"
Serial port list: []
Changing monitoring state from "Detecting serial port" to "Error: Failed to autodetect serial port, please set it manually."
Failed to autodetect serial port, please set it manually.`

Thanks for your help!

JeffB42, 2021-01-05:
>  I don't have my printer plugged into the Pi yet.

This is a known issue, in that, if the printer isn't turned on and connected, OctoPrint ends up restarting.  Yea, please connect it and have it turned on and then try troubleshooting the issue.

HonskiDesign, 2021-01-05:
Thanks again, Jeff. I hooked it up and that specific issue was resolved.

closed 2021-01-06 by JeffB42