Z-Bolt/OctoScreen #177: Icons and text too small

turbatus, 2020-09-16:
Hi, First of all, thank you for this amazing GUI!
Now, back to work: how the icon and text sizes can be changed? I have a 5"screen and the text is unreadable, icons are really tiny.
Thank you.

JeffB42, 2020-09-19:
@turbatus there isn't a setting to control the size - instead, the images and fonts are supposed to scale.  I'm not sure how much I can help out on this, but I have a few questions for you:
1) Could you provide a link to the touchscreen you are using?
2) What is the resolution of the touchscreen?
3) What is the resolution of the screen set to?  (this is set your configs for your system)
4) What is OCTOSCREEN_RESOLUTION set to?
5) Could you also take a picture of the display and post it here so I can see what you are seeing?

turbatus, 2020-09-20:
@JeffB42 Well, I feel like a noob, now! Once you requested that info, I have checked and looks like I had the resolution set to a different one than the native LCD's resolution! :(
It was 1280x720 in both places, changed to 800x480 fixed the size issue I had. 
You can close this, now. Was my bad. 
FYI: this is how it looked:
[image]

closed 2020-09-20 by turbatus