Z-Bolt/OctoScreen #180: 3.5 inch monitor

reggielee31, 2020-09-24:
Hi 
I just setup a octoscreen. I am using 3.5inch monitor and followed other post change below setting. but still not work. Please help.
framebuffer_width=800
framebuffer_height=533
and
hdmi_cvt 800 533 60 6 0 0 0
[image]

JeffB42, 2020-09-26:
@reggielee31 This is a known issue, and is mentioned in the FAQ.  At the moment I don't have a fix, but I have a few questions for you:
1) The size of your screen is 3.5", what is the resolution?  (I'm guessing 480x320, but could you please reply back and confirm)
2) Did you set OCTOSCREEN_RESOLUTION in the config file to your resolution?
3) For the display you are using, does it use HDMI, or does it use the GPIO pins?  Chris Riley recently posted a video on OctoDash, but at the beginning he talks about HDMI on 3.5" screens work for OctoScreen and OctoDash, but screens which use GPIO don't.  https://www.youtube.com/watch?v=kwo3HMBnqC4

reggielee31, 2020-09-27:
> @reggielee31 This is a known issue, and is mentioned in the FAQ. At the moment I don't have a fix, but I have a few questions for you:
>
> 1. The size of your screen is 3.5", what is the resolution? [...]

Hi,  Thanks for your reply

1. Yes,  it is 480x320.

2. OCTOSCREEN_RESOLUTION=800x533 it is followed by other post.

3. http://www.lcdwiki.com/MHS-3.5inch_RPi_Display  i am using this lcd mon

JeffB42, 2020-09-27:
OK, I have some bad news.  I don't think OctoScreen will work with a non-HDMI 480x320 screen.  The minimum resolution needed is 548x348 (regardless of screen).  Chris Riley recently posted a video, where he mentioned that he was able to run OotoDash and OctoScreen on his 480x320 HDMI screen, but couldn't run either using his 480x320 GPIO screen.
(see https://www.youtube.com/watch?v=kwo3HMBnqC4)

I just looked over the original developer's instructions/readme (at https://github.com/Z-Bolt/OctoScreen) and just noticed something.  In the instructions, he writes...

_"...if the physical resolution of your screen is 480x320, it's recommended to set the software resolution 800x533. If you are using Raspbian you can do it by changing hdmi_cvt param in /boot/config.txt file."_

Notice that he says one needs to **change the software resolution** by **changing hdmi_cvt**.
hmmm... hdmi...

I am not an expert in configuring Linux hardware, but this leads me to believe that if one has a screen that doesn't meet the minimum pixel requirements, one needs it to be an HDMI screen and not GPIO.

Like I said, I am not an expert in configuring Linux hardware, so please try experimenting yourself and see if there's a workaround.  If you find one, please post the solution.

reggielee31, 2020-09-27:
> OK, I have some bad news. I don't think OctoScreen will work with a non-HDMI 480x320 screen. The minimum resolution needed is 548x348 (regardless of screen). Chris Riley recently posted a video, [...]

Finally,  I have done.  I am follow this link.
https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-user-mods-octoprint-enclosures-nozzles-.../3-5-gpio-touchscreenrpi-3band-octoscreen/
[image]

jpokorny12, 2020-11-15:
https://badmin.co.uk/2020/07/07/octoscreen-installation/

JeffB42, 2020-11-18:
@jpokorny12 thanks for the link.  I just tried, using the steps @mattglg posted, with the kuman 3.5: display I have  (see https://www.amazon.com/gp/product/B01CNJVG8K/) but wasn't able to get the screen to resize correctly.  I reached out to Matt via his blog and hopefully he will contact me back, as I have some questions for him.

mattglg, 2020-11-18:
@JeffB42
To be honest I don't use Octoprint anymore and the pi and screen are currently a pi-hole with PADD for the display.
I found out an space SD card and ran through my old instruction and it didn't seem to work for me either - after a bit of fiddling I did get it running on my hardware though.
I will mention that I'm not a Linux guy at all! 

My kit is a Pi 3 with this display:
https://www.amazon.co.uk/dp/B01MRQTMTD/ref=cm_sw_em_r_mt_dp_NKrTFbWPVSKKW 

This will be a quick write up as I'm on my lunch hour so please forgive any typos.

The steps I took were as follows:

Flash the OctoPrint 0.17.0 image to the card with Etcher, put my wireless details into the octopi-wpa-supplicant.txt file and boot the pi from it.

**sudo apt-get update**

Connect to octoprint web site and run through wizard until you are at the normal OctoPrint home page

**sudo apt-get install -f
git clone https://github.com/waveshare/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
chmod +x LCD35-show
sudo ./LCD35-show 180**

The Pi should reboot and the screen should display the console (omit the 180 if you dont need to flip the display)

**sudo nano /boot/config.txt**

There should be an existing hdmi_cvy line at the bottom, amend this to:
_hdmi_cvt 800 533 60 6 0 0 0_

**sudo reboot**

The Pi should come back up and when the console if fully loaded the test should be visibly smaller (its still big when loading)

**sudo apt-get install libgtk-3-0 xserver-xorg xinit x11-xserver-utils
sudo reboot**

I rebooted between each step just to be safe

**sudo apt-get install git build-essential xorg-dev xutils-dev x11proto-dri2-dev
sudo apt-get install libltdl-dev libtool automake libdrm-dev
git clone https://github.com/ssvb/xf86-video-fbturbo.git
cd xf86-video-fbturbo
autoreconf -vi
./configure --prefix=/usr
make
sudo make install
sudo cp xorg.conf /etc/X11/xorg.conf

sudo reboot**

Connect the Pi to your 3D printer if not already

**wget https://github.com/Z-Bolt/OctoScreen/releases/download/v2.5.1/octoscreen_2.5-1_armhf.deb
sudo dpkg -i octoscreen_2.5-1_armhf.deb**

OctoScreen should load but with a black bar underneath

**sudo nano /etc/octoscreen/config**

Amend the existing OCTOSCREEN_RESOLUTION line to the following:

_OCTOSCREEN_RESOLUTION=800x533_

**sudo reboot**

This was enough to get it working for me, res looked good and touch screen appeared to work but I just moved the print head around a bit.

Hopefully this will get you up and running, let me know if it work and I'll tart up these steps a bit.

Cheers,
Matt

denics, 2020-11-19:
Dear all, I have the same problem with this model: http://www.lcdwiki.com/3.5inch_RPi_Display
Does it means that we cannot use OctoScreen on GPio based 3.5' TFT screens ?

mattglg, 2020-11-19:
@denics i just followed the instructions linked in the octoprint install guide for fbturbo.

DanielP5433, 2020-11-20:
I have gotten mine working now after following the newer video from Chris Riley, he got a 4inch gpio screen running. It's actually not that bad to do. Biggest thing to remember is to set the resolution in both boot config and octoscreen config to match!

denics, 2020-11-20:
hei @DanielP5433 can you post here the link to the video? thanks!

DanielP5433, 2020-11-20:
Hi, 

So just to go over it, 
I installed the adafruit drivers for my screen obviously.
I have set the boot config and octoscreen resolution to 800x533 
Set the hdmi_cvt to: hdmi_cvt 800 533 60 6 0 0 0 in boot config 
In the google docs linked in the video i didn't touch: Changes I made to the config.txt file or Check Octoscreen service and anything below that.

Video:: https://www.youtube.com/watch?v=OJ59hXSyBoI
Google Docs with video: https://docs.google.com/document/d/1VfDyTq252wkcG9_D4Nh2lTRCSq51iltMDkG5AJ3J9jQ/edit

JeffB42, 2020-11-20:
@denics and @DanielP5433 - that won't work for the discussion that's going on here.  There are three groups of screens:
1) Using a Pimoroni Hyperpiel.  This is the screen in Chris's video.  This display works, and Chris's video is excellent if you have one.  This is the display I have.
2) 3.5" 480x320 HDMI screen.  The postings I've seen report that this works.
3) 3.5" 480x320 GPOI screen.  This doesn't work.  Or maybe it does.  The postings that report the the screen doesn't scale properly all seem to have one thing in common - a 3.5" 480x320 screen that's connected via GPIO.  Some users however have reported that they have been able to get their GPIO screen to work.

I ordered a 3.5" GPIO screen and I'm getting the issue.  I'm currently working through some settings some people have reported that work, but at the moment, this is still a WIP for me as I haven't been able to get my 3.5" GPIO screen to scale correctly.

DanielP5433, 2020-11-20:
I have a 3.5" GPIO screen from adafruit that is now working. I used the general guidelines from his video to help. The only difference between his screen and another gpio screen is the size and where you get the drivers. The overall process should get it working.

JeffB42, 2020-11-23:
@DanielP5433 My apologies.  From your post, it looked to me that you had a HyperPixel and was reporting that Chis's steps worked, but you were talking about using his steps with a HyperPixel, and not with a 3.5" 480x320 TFT.  What's odd to me, is that most of the steps he listed (like screen rotation, the drivers, the matrix transform, etc...) are specific to the HyperPixel, and after removing the HyperPixel-specific steps, there wasn't much left that I wasn't already doing, but yet it wasn't displaying correctly.  Well, no matter... 1) thanks for the help! and 2) I got it working thanks to the steps @mattglg posted, but more on that in my next post...

JeffB42, 2020-11-23:
Well UREKA!!  Thanks @mattglg!  I started over with a fresh OctoPrint image and ran through your steps, and I got it to display correctly.  I'm going to reach out to another person and see if they also have success, and if they report they can now run correctly, I'll post a guide in the Wiki and post announcements in all the open tickets.  Thanks again @mattglg and @DanielP5433!!!

denics, 2020-11-24:
Finally! @JeffB42 here instructions for making OctoScreen work with ``3. 3.5" 480x320 GPOI screen``.

```
git clone https://github.com/waveshare/LCD-show.git
cd LCD-show/
chmod +x LCD35-show
sudo ./LCD35-show
sudo reboot
```

As soon as your RPi reboot, you will see the console on your screen! This is the begin of the adventure. Let's follow the great wiki here: https://github.com/notro/fbtft/wiki/FBTFT-on-Raspian#framebuffer-copy .
Compile the package, install it and make sure to enable at boot with:

```
sudo systemctl enable fbcp
```

At this point you must make sure that ``/usr/share/X11/xorg.conf.d/99-fbturbo.conf`` looks like:

```
Section "Device"
        Identifier      "Allwinner A10/A13/A20 FBDEV"
        Driver          "fbturbo"
        Option          "fbdev" "/dev/fb0"

        Option          "SwapbuffersWait" "true"
EndSection
```

Please note that our Device is at ``/dev/fb0``. After following these steps you are syncing /dev/fb0 with /dev/fb1 . One last touch: ``sudo systemctl enable octoscreen``. At this point I am a bit confused, fbcp should be launched before octoscreen, but this is not happening. Maybe someone with a better knowledge of systemd can help here.

Reboot, launch ``sudo service octoscreen start`` and enjoy your TFT3.5 touchscreen.

JeffB42, 2020-11-24:
@denics though not the same steps, I posted the steps that worked for me in the wiki, at https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen

denics, 2020-11-24:
This is the result when following my "guide".
[image]

I wanted to re-run the instructions found here to remove any useless or redundant steps.

denics, 2020-11-24:
I have just realized that not all the screens show the same way. Like the network panel does not scale correctly :(

mattglg, 2020-11-24:
@denics my screen had that black bar on the bottom until I set OCTOSCREEN_RESOLUTION in the config file, I can’t see that step in your setup so might be worth a go

denics, 2020-11-24:
You are right, setting this in ``/etc/octoscreen/config``

```
OCTOSCREEN_RESOLUTION=640x480
``` 

and this in ``/boot/config.txt``

```
hdmi_cvt 640 480 60
```

remove the black bar and everything is fine. However, the network panel keeps giving problems. 

I recompiled Octoprint-TFT and submitted a pull request here https://github.com/darksid3r/OctoPrint-TFT/pull/2 . Octoprint-TFT is more suitable for small screens I found.

JeffB42, 2020-12-04:
@reggielee31 Could you follow the steps listed in the wiki (located at https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen) and then post a reply to this issue whether the steps listed work and the issue is resolved?  I'd like to start closing out all the display-related issues.

If it doesn't work,
1) make sure you use a Raspberry Pi 3B or Raspberry Pi 4
2) start with a fresh install of OctoPi
3) please take a picture of the screen

Thanks, 

Jeff

JeffB42, 2021-01-02:
I'm closing this issue b/c I haven't heard anything back.

closed 2021-01-02 by JeffB42