Z-Bolt/OctoScreen #181: Button/gcode documentation

lanman1, 2020-09-24:
U'm trying to find documentation of how pressing a button performs a function.  I see the menu and panel config, but where is a function mapped to pressing a button?  For instance, I can perform a G29 and do an ABL, how do i make OctoPrint issue a G29 when the Auto Z Calibration button is pressed?

JeffB42, 2020-09-26:
@lanman1 Most of the code uses the OctoPrint API (see https://docs.octoprint.org/en/master/api/).  There are some places where the gcode is hard coded into the source code though.  The best way to tell what's sent is by looking through the project's source code.  The master branch has the latest release, and the 2.6.0-dev branch has the latest version that I'm working on (and is still in development).

Having customized buttons is the main reason why I decided to work on OctoScreen... and it's coming, it's just that there are some bugs that need to get worked out first (like the whole "screen being resized when running in 480x320" issue)

lanman1, 2020-09-28:
Ah, sorry to be dense, i 'assumed' that since the menu structure was defined in .json that function to button mappings would be as well.  I'll take a look at the source and see if i can figure out why I can issue a G26 and do ABL but the Auto Z calibration button doesn't work.  Thanks and i love this project.

JeffB42, 2020-09-28:
No, not dense at all.  When I first heard about OctoScreen, I think I read somewhere that it was customizable, and that got me interested in using it.  After installing and using it, I discovered that it's really more customizable in the sense that you can command what panels are displayed, and not customize the buttons.

This might help:  In the current 2.5.1 codebase, the home panel is at:
https://github.com/Z-Bolt/OctoScreen/blob/master/ui/home.go

In the initialize() function, there is:
```
func (m *homePanel) initialize() {
	defer m.Initialize()

	m.AddButton(m.createMoveButton("Home All", "home.svg",
		octoprint.XAxis, octoprint.YAxis, octoprint.ZAxis,
	))

	m.AddButton(m.createMoveButton("Home X", "home-x.svg", octoprint.XAxis))
	m.AddButton(m.createMoveButton("Home Y", "home-y.svg", octoprint.YAxis))
	m.AddButton(m.createMoveButton("Home Z", "home-z.svg", octoprint.ZAxis))
}
```

...and the createMoveButton() is:
```
func (m *homePanel) createMoveButton(label, image string, axes ...octoprint.Axis) gtk.IWidget {
	return MustButtonImageStyle(label, image, "color2", func() {
		cmd := &octoprint.PrintHeadHomeRequest{Axes: axes}
		Logger.Warningf("Homing the print head in %s axes", axes)
		if err := cmd.Do(m.UI.Printer); err != nil {
			Logger.Error(err)
			return
		}
	})
}
```

octoprint.PrintHeadHomeRequest() is in https://github.com/Z-Bolt/OctoScreen/blob/master/vendor/github.com/mcuadros/go-octoprint/printer.go
...and it sends an API message to OctoPrint (OctoPrint exposes it functionality through a REST API)

I forget where, but there's also some other code in the 2.5.1 codebase which sends gcode (also through REST API) to OctoPrint, and it's sending G28.  (so essentially, there are two different implementations of home all in the 2.5.1 code)

The current code (the release I'm working on), I consolidated everything.  For reference, there's:
https://github.com/Z-Bolt/OctoScreen/blob/2.6.0-dev/ui/HomePanel.go
https://github.com/Z-Bolt/OctoScreen/blob/2.6.0-dev/uiWidgets/HomeButton.go
https://github.com/Z-Bolt/OctoScreen/blob/2.6.0-dev/uiWidgets/HomeAllButton.go

In HomeAllButton.go, I left a comment about using PrintHeadHomeRequest for home all, and of if users report issues, there's the G28 version which is currently comment out.

I've tested this, using PrintHeadHomeRequest (and passing all three axis), and this works on my printer, which is a Prusa MKS3.

lanman1, 2020-09-29:
Thanks JeffB42!

closed 2020-09-29 by lanman1