Z-Bolt/OctoScreen #182: Job for octoscreen.service failed because the control process exited with error code

willgarner42, 2020-10-11:
(Reading database ... 56081 files and directories currently installed.)
Preparing to unpack octoscreen_2.5-1_armhf.deb ...
Unpacking octoscreen (2.5-1) over (2.5-1) ...
Setting up octoscreen (2.5-1) ...
Job for octoscreen.service failed because the control process exited with error code.
See "systemctl status octoscreen.service" and "journalctl -xe" for details.

set up is (raspberry pi 3 b+, xpt2046 raspberry pi touch 5" display,) updated to 1.4.2 Octoprint .

pdsavas, 2020-11-23:
I am getting the same error with a similar setup.

JeffB42, 2021-01-02:
@willgarner42 and @pdsavas I just published release 2.6.1.  Try following the install steps listed on the main page, including the dependencies (https://github.com/Z-Bolt/OctoScreen), or if you are upgrading an existing (working) installation, follow the steps listed at https://github.com/Z-Bolt/OctoScreen/releases/tag/2.6.1.

JeffB42, 2021-02-04:
@willgarner42 is this still an issue?

willgarner42, 2021-02-04:
The set up is a raspberry pi 3 B+ with a 3.5 RPI 480x320 pixel 16 bit
xpt2046 display.

I use a 32g SD card and cable internet to the pi (I do set up wifi and it
does work) however I flash the card start octoprint update load display
drivers and I get no error or download problems I go to hi touch instal
reboot and that's as far as it ever goes.

JeffB42, 2021-02-05:
OK, install steps are listed at https://github.com/Z-Bolt/OctoScreen/wiki/Setting-Up-OctoScreen-and-Your-Display and https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen.  Look these pages over and see if any of the information there helps.  If not, turn on logging, set the log level to debug, then look at the log file, and look for the first error listed.  Also, update to 2.6.1 as that's the latest.

willgarner42, 2021-02-05:
Ok, I'll check it out.

willgarner42, 2021-02-24:
I 100% restart on this project and now have one working great Raspberry pi
3 B + 3.5 touch screen . The other. Same type of raspberry pi with HDMI 5in
touch is giving me problems but believe it's the drivers for the screen not
octoprint

So thank you for the help
Be safe

JeffB42, 2021-02-27:
Glad to hear it's working for you.  Also version 2.7 will be released in a few weeks.  It won't fix display configuration issues, but it does fix a lot of the rebooting issues that have been reported.

closed 2021-02-27 by JeffB42