Z-Bolt/OctoScreen #183: 2.6.0 Update?

mustafacobanoglu55, 2020-10-21:
Hello now i am using the master version. İs there any hay to update this to 2.6.0-dev version? @alexkv

JeffB42, 2020-10-24:
It's in development and hasn't been released yet.  I'm working out a few bugs and writing documentation at the moment.  Before it's released however, I want to get the display issue resolved.  I have a 3.5" 480x320 (GPIO) screen on order and waiting for it to arrive.

mustafacobanoglu55, 2020-10-24:
Thank you for the information. Waiting Impatiently. Now I am revising the OctoScreen interface to work effectively with Klipper and BLV. but it is very tiring for me to compile it to see the changes every time. because it takes too long. There is definitely an easier way. But the only way I know to see the changes is to compile.

JeffB42, 2020-12-30:
@mustafacobanoglu55 It's been released.

closed 2020-12-30 by JeffB42