Z-Bolt/OctoScreen #190: Problem install 2.6.0

nelsonov, 2020-12-29:
[Every attempt to install 2.6.0, following the instructions on the releases page, fails:

```
sudo dpkg -i octoscreen_2.6.0_armhf.deb
Selecting previously unselected package octoscreen.
(Reading database ... 101352 files and directories currently installed.)
Preparing to unpack octoscreen_2.6.0_armhf.deb ...
Unpacking octoscreen (2.6.0-1) ...
Setting up octoscreen (2.6.0-1) ...
Installing service /lib/systemd/system/lightdm.service
Failed to preset unit: File /etc/systemd/system/display-manager.service already exists and is a symlink to /lib/systemd/system/lightdm.service.                         
/usr/bin/deb-systemd-helper: error: systemctl preset failed on octoscreen.service: No such file or directory    
```                                                        
I have verified that /lib/systemd/system/display-manager.service does not exist before the install, but does exist after the install.

nelsonov, 2020-12-30:
Any attempt to install 2.5.1 now reports the same error, so I have no way to have a working version of octoscreen.

JeffB42, 2020-12-30:
@nelsonov did you this?
`sudo dpkg -r octoscreen`

This is in the release notes.  It's after the wget step, and before the `sudo dpkg -i octoscreen_2.6.0_armhf.deb` step.

nelsonov, 2020-12-30:
Yes, I did the uninstall step before installing 2.6.0.

nelsonov, 2020-12-30:
I removed lightdm and I am now able to install 2.5.1.  2.6.0 installs but hangs and eventually times out on startup.

nsa666666, 2020-12-30:
i can't actually download the .deb-package. it gives me a 404

nelsonov, 2020-12-30:
From journalctl for 2.6.0:

```
Dec 29 16:47:28 octopi systemd[1]: Starting OctoScreen...                                                                                                                
-- Subject: A start job for unit octoscreen.service has begun execution                                                                                                  
-- Defined-By: systemd                                                                                                                                                   
-- Support: https://www.debian.org/support                                                                                                                               
--                                                                                                                                                                       
-- A start job for unit octoscreen.service has begun execution.                                                                                                          
--                                                                                                                                                                       
-- The job identifier is 1453.                                                                                                                                           
Dec 29 16:47:28 octopi xinit[1935]: X.Org X Server 1.20.4                                                                                                                
Dec 29 16:47:28 octopi xinit[1935]: X Protocol Version 11, Revision 0                                                                                                    
Dec 29 16:47:28 octopi xinit[1935]: Build Operating System: Linux 5.4.0-54-generic armv8l Raspbian                                                                       
Dec 29 16:47:28 octopi xinit[1935]: Current Operating System: Linux octopi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l                                     
Dec 29 16:47:28 octopi xinit[1935]: Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_h
Dec 29 16:47:28 octopi xinit[1935]: Build Date: 01 December 2020  05:59:57PM                                                                                             
Dec 29 16:47:28 octopi xinit[1935]: xorg-server 2:1.20.4-1+rpt2+deb10u2 (https://www.debian.org/support)                                                                 
Dec 29 16:47:28 octopi xinit[1935]: Current version of pixman: 0.36.0                                                                                                    
Dec 29 16:47:28 octopi xinit[1935]:         Before reporting problems, check http://wiki.x.org                                                                           
Dec 29 16:47:28 octopi xinit[1935]:         to make sure that you have the latest version.                                                                               
Dec 29 16:47:28 octopi xinit[1935]: Markers: (--) probed, (**) from config file, (==) default setting,                                                                   
Dec 29 16:47:28 octopi xinit[1935]:         (++) from command line, (!!) notice, (II) informational,                                                                     
Dec 29 16:47:28 octopi xinit[1935]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Dec 29 16:47:28 octopi xinit[1935]: (==) Log file: "/var/log/Xorg.0.log", Time: Tue Dec 29 16:47:28 2020                                                                 
Dec 29 16:47:28 octopi xinit[1935]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"                                                                      
Dec 29 16:47:28 octopi xinit[1935]: modprobe: FATAL: Module g2d_23 not found in directory /lib/modules/5.4.51-v7l+                                                      
Dec 29 16:47:29 octopi xinit[1935]: 2020/12/29 16:47:29 logger.SetLogLevel() - the log level is now set to: "debug"                                                     
Dec 29 16:47:29 octopi xinit[1935]: time="2020-12-29T16:47:29-08:00" level=info msg="logger.init() - logFilePath is: \"logs/logfile.txt\"" file=logger.go func=utils.init
Dec 29 16:47:29 octopi xinit[1935]: time="2020-12-29T16:47:29-08:00" level=error msg="logger.init() - OpenFile() FAILED!  err is: \"open logs/logfile.txt: no such file o
Dec 29 16:47:29 octopi xinit[1935]: time="2020-12-29T16:47:29-08:00" level=error msg="Failed to log to file, using default stderr" file=logger.go func=utils.init.0 line=
Dec 29 16:47:29 octopi xinit[1935]: time="2020-12-29T16:47:29-08:00" level=debug msg="entering main.init()" file=main.go func=main.init.0 line=55                       
Dec 29 16:47:29 octopi xinit[1935]: time="2020-12-29T16:47:29-08:00" level=error msg="main.init() - RequiredEnvironmentVariablesAreSet() returned false" file=main.go fun
Dec 29 16:47:29 octopi xinit[1935]: time="2020-12-29T16:47:29-08:00" level=debug msg="leaving main.init()" file=main.go func=main.init.0 line=60                        
Dec 29 16:47:29 octopi xinit[1935]: time="2020-12-29T16:47:29-08:00" level=debug msg="entering main.main()" file=main.go func=main.main line=29                         
[... cut ...]
Dec 29 16:47:29 octopi xinit[1935]: time="2020-12-29T16:47:29-08:00" level=info file=environment.go func=utils.DumpEnvironmentVariables line=81 
```                        

And this is what I get when I try to start it, along with "time" to show the length of timeout:

```
pi@octopi:~ $ time sudo systemctl start octoscreen
Job for octoscreen.service failed because a timeout was exceeded.
See "systemctl status octoscreen.service" and "journalctl -xe" for details.

real    0m18.984s
user    0m0.063s
sys     0m0.023s

```

JeffB42, 2020-12-30:
> I removed lightdm and I am now able to install 2.5.1. 2.6.0 installs but hangs and eventually times out on startup.

I checked my system, and lightdm is not present.  I'm not familiar with lightdm.  This page https://wiki.ubuntu.com/LightDM says it's part of Ubuntu.  Are you running Ubuntu?

nelsonov, 2020-12-30:
@nsa666666 I think he has fixed the link now.  I used:

`wget https://github.com/Z-Bolt/OctoScreen/releases/download/2.6.0/octoscreen_2.6.0_armhf.deb`

JeffB42, 2020-12-30:
> i can't actually download the .deb-package. it gives me a 404

@nsa666666 I took it down b/c cause of the issue Lee reports.  I posted it back up, an also posted a slightly updated version.

@nelsonov I posted a slightly updated version of octoscreen_2.6.0_armhf.deb.  It probably won't fix the issue you are seeing, but since you are installing, uninstalling, and troubleshooting this issue on your system, you might want to grab the updated version.  I ran into some issues where, on my development machine everything appeared fine, but on only of my RPi test machines, some of the graphics were enlarging the window.  I know I should updated the version, but I haven't actually announced the release (I was planning on posting an announcement in the Issues section), and updating the release version with this project is somewhat of a pain, as there are several places where the version number needs to be incremented.  If doing a major or minor release, it's not too bad, but for an incremental bug fix, it's a slight pain.

nsa666666, 2020-12-30:
Thank you. It works now. Installed without problmes.

JeffB42, 2020-12-30:
@nelsonov thanks for posting the log.  Looking at the log file, I see a few things:

```
Dec 29 16:47:29 octopi xinit[1935]: time="2020-12-29T16:47:29-08:00" level=info msg="logger.init() - logFilePath is: \"logs/logfile.txt\"" file=logger.go func=utils.init
Dec 29 16:47:29 octopi xinit[1935]: time="2020-12-29T16:47:29-08:00" level=error msg="logger.init() - OpenFile() FAILED!  err is: \"open logs/logfile.txt: no such file o
```
...try creating logs/logfile.txt

```
Dec 29 16:47:29 octopi xinit[1935]: time="2020-12-29T16:47:29-08:00" level=info msg="key: \"OCTOPRINT_APIKEY\", value: \">>MISSING<<\"" file=environment.go func=utils.du
```
...the OCTOPRINT_APIKEY value needs to be set

```
Dec 29 16:47:29 octopi xinit[1935]: time="2020-12-29T16:47:29-08:00" level=info msg="key: \"OCTOSCREEN_LOG_LEVEL\", value: \"Warn\"" file=environment.go func=utils.dumpE
```
...you might want to set the level to debug.  It's very chatty, so you won't want to regularly run at that level, but for debugging and troubleshooting, you might want to set it to the deeper. more verbose level of debug.

nelsonov, 2020-12-30:
I wonder how it works with 2.5.1....  I have never set the API key, yet 2.5.1 is running right now.  It is presumably being read from octoprint's config.yaml.  There must be something weird with paths.  What is the full paths where octoscreen is attempting to find config.yaml and logs/logfile.txt?  The paths in the output above give relative paths, but what are they relative to?

Mulle00, 2020-12-30:
I had the same problem, now its working:

- You need to fill in the APIKEY in the config file
- You need to install this plugin in octoprint : https://github.com/Z-Bolt/OctoPrint-Z-Bolt-OctoScreen
- Restart everything

JeffB42, 2020-12-31:
> What is the full paths where octoscreen is attempting to find config.yaml and logs/logfile.txt?

For the log file path, try setting it to `/home/pi/logs/logfile.txt`

JeffB42, 2021-01-02:
@nelsonov please update to 2.6.1. (https://github.com/Z-Bolt/OctoScreen/releases/tag/2.6.1)

If you see any issues, let me know, otherwise, please close this issue and I'll know that the issues have been resolved.

nelsonov, 2021-01-04:
Thank you.  This works.  I really like the look and feel of 2.6.  Thanks for the continuing work!

closed 2021-01-04 by nelsonov