Z-Bolt/OctoScreen #192: Swapping between X11 display for Octoscreen and console.

boydhako, 2020-12-30:
For your eyes to asses what I'm talking about... 
https://youtu.be/77XMy30ctVE

Using the SmartiPi Touch v1 for the display.

root@octopi:/home/pi# cat /proc/device-tree/model
Raspberry Pi 3 Model B Rev 1.2

root@octopi:/home/pi# cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

== The log from `journalctl -xf`
`-- Logs begin at Thu 2019-02-14 00:11:59 HST. --
Dec 29 19:45:04 octopi xinit[2230]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Dec 29 19:45:04 octopi xinit[2230]: (==) Log file: "/var/log/Xorg.0.log", Time: Tue Dec 29 19:45:04 2020
Dec 29 19:45:04 octopi xinit[2230]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Dec 29 19:45:05 octopi xinit[2230]: 2020/12/29 19:45:05 logger.SetLogLevel() - the log level is now set to: "debug"
Dec 29 19:45:05 octopi xinit[2230]: time="2020-12-29T19:45:05-10:00" level=info msg="logger.init() - logFilePath is: \"logs/logfile.txt\"" file=logger.go func=utils.init.0 line=67
Dec 29 19:45:05 octopi xinit[2230]: time="2020-12-29T19:45:05-10:00" level=error msg="logger.init() - OpenFile() FAILED!  err is: \"open logs/logfile.txt: no such file or directory\"" file=logger.go func=utils.init.0 line=76
Dec 29 19:45:05 octopi xinit[2230]: time="2020-12-29T19:45:05-10:00" level=error msg="Failed to log to file, using default stderr" file=logger.go func=utils.init.0 line=77
Dec 29 19:45:05 octopi xinit[2230]: time="2020-12-29T19:45:05-10:00" level=debug msg="entering main.init()" file=main.go func=main.init.0 line=55
Dec 29 19:45:05 octopi xinit[2230]: 2020/12/29 19:45:05 logger.SetLogLevel() - the log level is now set to: "warning"
Dec 29 19:45:05 octopi xinit[2230]: 2020/12/29 19:45:05 main.SetLogLevel() - logLevel is now set to: "warn"
Dec 29 19:45:07 octopi xinit[2230]: time="2020-12-29T19:45:07-10:00" level=error msg="ui.checkNotification() - Do(GetNotificationRequest) returned an error: \"unexpected status code: 404\"" file=logger.go func=utils.LogError line=86
Dec 29 19:45:07 octopi xinit[2230]: time="2020-12-29T19:45:07-10:00" level=error msg="ui.loadSettings() - Do(GetSettingsRequest) returned an error: \"unexpected status code: 404\"" file=logger.go func=utils.LogError line=86
Dec 29 19:45:07 octopi xinit[2230]: time="2020-12-29T19:45:07-10:00" level=error msg="leaving ui.loadSettings() - Do(GetSettingsRequest) returned an error" file=ui.go func="ui.(*UI).loadSettings" line=293
Dec 29 19:45:07 octopi xinit[2230]: panic: runtime error: invalid memory address or nil pointer dereference
Dec 29 19:45:07 octopi xinit[2230]: [signal SIGSEGV: segmentation violation code=0x1 addr=0x18 pc=0x48e324]
Dec 29 19:45:07 octopi xinit[2230]: goroutine 1 [running]:
Dec 29 19:45:07 octopi xinit[2230]: github.com/Z-Bolt/OctoScreen/ui.(*idleStatusPanel).initialize(0x11d2a640)
Dec 29 19:45:07 octopi xinit[2230]:         /go/src/github.com/Z-Bolt/OctoScreen/obj-arm-linux-gnueabihf/src/github.com/Z-Bolt/OctoScreen/ui/IdleStatusPanel.go:47 +0x10c
Dec 29 19:45:07 octopi xinit[2230]: github.com/Z-Bolt/OctoScreen/ui.IdleStatusPanel(0x11cae600, 0x11c47940)
Dec 29 19:45:07 octopi xinit[2230]:         /go/src/github.com/Z-Bolt/OctoScreen/obj-arm-linux-gnueabihf/src/github.com/Z-Bolt/OctoScreen/ui/IdleStatusPanel.go:31 +0x16c
Dec 29 19:45:07 octopi xinit[2230]: github.com/Z-Bolt/OctoScreen/ui.(*UI).verifyConnection(0x11cae600)
Dec 29 19:45:07 octopi xinit[2230]:         /go/src/github.com/Z-Bolt/OctoScreen/obj-arm-linux-gnueabihf/src/github.com/Z-Bolt/OctoScreen/ui/ui.go:245 +0x548
Dec 29 19:45:07 octopi xinit[2230]: github.com/Z-Bolt/OctoScreen/ui.(*UI).update(0x11cae600)
Dec 29 19:45:07 octopi xinit[2230]:         /go/src/github.com/Z-Bolt/OctoScreen/obj-arm-linux-gnueabihf/src/github.com/Z-Bolt/OctoScreen/ui/ui.go:372 +0x140
Dec 29 19:45:07 octopi xinit[2230]: github.com/Z-Bolt/OctoScreen/ui.(*UI).(github.com/Z-Bolt/OctoScreen/ui.update)-fm()
Dec 29 19:45:07 octopi xinit[2230]:         /go/src/github.com/Z-Bolt/OctoScreen/obj-arm-linux-gnueabihf/src/github.com/Z-Bolt/OctoScreen/ui/ui.go:98 +0x1c
Dec 29 19:45:07 octopi xinit[2230]: reflect.Value.call(0x516010, 0x11c6ec78, 0x13, 0x5f9503, 0x4, 0xcb9e3c, 0x0, 0x0, 0x59d228, 0x69e01, ...)
Dec 29 19:45:07 octopi xinit[2230]:         /usr/local/go/src/reflect/value.go:434 +0x678
Dec 29 19:45:07 octopi xinit[2230]: reflect.Value.Call(0x516010, 0x11c6ec78, 0x13, 0xcb9e3c, 0x0, 0x0, 0x0, 0x11cdd490, 0x11cdd4a0)
Dec 29 19:45:07 octopi xinit[2230]:         /usr/local/go/src/reflect/value.go:302 +0x74
Dec 29 19:45:07 octopi xinit[2230]: github.com/Z-Bolt/OctoScreen/vendor/github.com/gotk3/gotk3/glib.sourceAttach.func2()
Dec 29 19:45:07 octopi xinit[2230]:         /go/src/github.com/Z-Bolt/OctoScreen/obj-arm-linux-gnueabihf/src/github.com/Z-Bolt/OctoScreen/vendor/github.com/gotk3/gotk3/glib/glib.go:311 +0x160
Dec 29 19:45:07 octopi xinit[2230]: reflect.Value.call(0x516010, 0x11d1a000, 0x13, 0x5f9503, 0x4, 0xcb9e3c, 0x0, 0x0, 0x59d228, 0xffffff01, ...)
Dec 29 19:45:07 octopi xinit[2230]:         /usr/local/go/src/reflect/value.go:434 +0x678
Dec 29 19:45:07 octopi xinit[2230]: reflect.Value.Call(0x516010, 0x11d1a000, 0x13, 0xcb9e3c, 0x0, 0x0, 0x11c00128, 0x5a35c, 0x5a3ec)
Dec 29 19:45:07 octopi xinit[2230]:         /usr/local/go/src/reflect/value.go:302 +0x74
Dec 29 19:45:07 octopi xinit[2230]: github.com/Z-Bolt/OctoScreen/vendor/github.com/gotk3/gotk3/glib.goMarshal(0x72d019c0, 0x7ec03a08, 0x0, 0x0, 0x0, 0x0)
Dec 29 19:45:07 octopi xinit[2230]:         /go/src/github.com/Z-Bolt/OctoScreen/obj-arm-linux-gnueabihf/src/github.com/Z-Bolt/OctoScreen/vendor/github.com/gotk3/gotk3/glib/glib.go:216 +0x3d8
Dec 29 19:45:07 octopi xinit[2230]: github.com/Z-Bolt/OctoScreen/vendor/github.com/gotk3/gotk3/glib._cgoexpwrap_88078d459311_goMarshal(0x72d019c0, 0x7ec03a08, 0x0, 0x0, 0x0, 0x0)
Dec 29 19:45:07 octopi xinit[2230]:         github.com/Z-Bolt/OctoScreen/vendor/github.com/gotk3/gotk3/glib/_obj/_cgo_gotypes.go:3771 +0x44
Dec 29 19:45:07 octopi xinit[2230]: github.com/Z-Bolt/OctoScreen/vendor/github.com/gotk3/gotk3/gtk._Cfunc_gtk_main()
Dec 29 19:45:07 octopi xinit[2230]:         github.com/Z-Bolt/OctoScreen/vendor/github.com/gotk3/gotk3/gtk/_obj/_cgo_gotypes.go:11943 +0x30
Dec 29 19:45:07 octopi xinit[2230]: github.com/Z-Bolt/OctoScreen/vendor/github.com/gotk3/gotk3/gtk.Main()
Dec 29 19:45:07 octopi xinit[2230]:         /go/src/github.com/Z-Bolt/OctoScreen/obj-arm-linux-gnueabihf/src/github.com/Z-Bolt/OctoScreen/vendor/github.com/gotk3/gotk3/gtk/gtk.go:884 +0x14
Dec 29 19:45:07 octopi xinit[2230]: main.main()
Dec 29 19:45:07 octopi xinit[2230]:         /go/src/github.com/Z-Bolt/OctoScreen/obj-arm-linux-gnueabihf/src/github.com/Z-Bolt/OctoScreen/main.go:48 +0xfc
Dec 29 19:45:07 octopi xinit[2230]: /usr/bin/xinit: connection to X server lost
Dec 29 19:45:07 octopi xinit[2230]: [1B blob data]
[... 87 lines cut ...]
Dec 29 19:45:20 octopi xinit[2272]:         /go/src/github.com/Z-Bolt/OctoScreen/obj-arm-linux-gnueabihf/src/github.com/Z-Bolt/OctoScreen/ui/ui.go:98 +0x1c
Dec 29 19:45:20 octopi xinit[2272]: reflect.Value.call(0x516010, 0x12a6ec78, 0x13, 0x5f9503, 0x4, 0xcb9e3c, 0x0, 0x0, 0x59d228, 0x69e01, ...)
Dec 29 19:45:20 octopi xinit[2272]:         /usr/local/go/src/reflect/value.go:434 +0x678
Dec 29 19:45:20 octopi xinit[2272]: reflect.Value.Call(0x516010, 0x12a6ec78, 0x13, 0xcb9e3c, 0x0, 0x0, 0x0, 0x12add4a0, 0x12add4b0)
Dec 29 19:45:20 octopi xinit[2272]:         /usr/local/go/src/reflect/value.go:302 +0x74
Dec 29 19:45:20 octopi xinit[2272]: github.com/Z-Bolt/OctoScreen/vendor/github.com/gotk3/gotk3/glib.sourceAttach.func2()
Dec 29 19:45:20 octopi xinit[2272]:         /go/src/github.com/Z-Bolt/OctoScreen/obj-arm-linux-gnueabihf/src/github.com/Z-Bolt/OctoScreen/vendor/github.com/gotk3/gotk3/glib/glib.go:311 +0x160
Dec 29 19:45:20 octopi xinit[2272]: reflect.Value.call(0x516010, 0x12b1a000, 0x13, 0x5f9503, 0x4, 0xcb9e3c, 0x0, 0x0, 0x59d228, 0xffffff01, ...)
Dec 29 19:45:20 octopi xinit[2272]:         /usr/local/go/src/reflect/value.go:434 +0x678
Dec 29 19:45:20 octopi xinit[2272]: reflect.Value.Call(0x516010, 0x12b1a000, 0x13, 0xcb9e3c, 0x0, 0x0, 0x12a00128, 0x5a35c, 0x5a3ec)
Dec 29 19:45:20 octopi xinit[2272]:         /usr/local/go/src/reflect/value.go:302 +0x74
[... cut ...]
Dec 29 19:45:20 octopi xinit[2272]:         /go/src/github.com/Z-Bolt/OctoScreen/obj-arm-linux-gnueabihf/src/github.com/Z-Bolt/OctoScreen/vendor/github.com/gotk3/gotk3/gtk/gtk.go:884 +0x14
Dec 29 19:45:20 octopi xinit[2272]: main.main()
Dec 29 19:45:20 octopi xinit[2272]:         /go/src/github.com/Z-Bolt/OctoScreen/obj-arm-linux-gnueabihf/src/github.com/Z-Bolt/OctoScreen/main.go:48 +0xfc
Dec 29 19:45:20 octopi xinit[2272]: /usr/bin/xinit: connection to X server lost
Dec 29 19:45:20 octopi xinit[2272]: [1B blob data]
Dec 29 19:45:20 octopi xinit[2272]: waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.
`

JeffB42, 2020-12-30:
@boydhako Thanks for posting the video and the log.  I see there potentially being a few issues here.  First, when you are running OctoScreen, I see that OctoScreen starts and displays the status, "Printer is offline" and doesn't get to the connected state.
1. When runing OctoScreen, is your printer on and connected to your Raspberry Pi?  If not, turn your printer on and then start OctoPrint and OctoScreen.  There's a know issue (see https://github.com/Z-Bolt/OctoScreen/issues/114) where, if the printer isn't on and OctoPrint can't connect, OctoScreen crashes and causes a reboot.
2. Is the API key set in your config file?  Edit /etc/octoscreen/config and make sure that OCTOPRINT_APIKEY and OCTOPRINT_HOST are set.

kquinsland, 2020-12-31:
Heh, came here to file this bug as well. I've had this 'issue' for about a year now (been busy! now getting time to write it up). The constant octodash restarts opens a connection to octoprint and eventually the python process will run out of sockets / start behaving weirdly. The simple fix is just yank power to the pi... but that's a different problem that will go away when octodash stops crash-cycleing.

Here's my `journalctl` dump:
 (this repeats every few seconds...)

```log
Dec 31 09:54:46 octoprint systemd[1]: Starting OctoScreen...
-- Subject: A start job for unit octoscreen.service has begun execution
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit octoscreen.service has begun execution.
--
-- The job identifier is 4547.
Dec 31 09:54:46 octoprint xinit[2385]: X.Org X Server 1.20.4
Dec 31 09:54:46 octoprint xinit[2385]: X Protocol Version 11, Revision 0
Dec 31 09:54:46 octoprint xinit[2385]: Build Operating System: Linux 5.4.0-54-generic armv8l Raspbian
Dec 31 09:54:46 octoprint xinit[2385]: Current Operating System: Linux octoprint 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l
Dec 31 09:54:46 octoprint xinit[2385]: Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=533 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:0D:C5:2C vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=tty1 root=PARTUUID=738a4d67-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Dec 31 09:54:46 octoprint xinit[2385]: Build Date: 01 December 2020  05:59:57PM
Dec 31 09:54:46 octoprint xinit[2385]: xorg-server 2:1.20.4-1+rpt2+deb10u2 (https://www.debian.org/support)
Dec 31 09:54:46 octoprint xinit[2385]: Current version of pixman: 0.36.0
Dec 31 09:54:46 octoprint xinit[2385]:         Before reporting problems, check http://wiki.x.org
Dec 31 09:54:46 octoprint xinit[2385]:         to make sure that you have the latest version.
Dec 31 09:54:46 octoprint xinit[2385]: Markers: (--) probed, (**) from config file, (==) default setting,
Dec 31 09:54:46 octoprint xinit[2385]:         (++) from command line, (!!) notice, (II) informational,
Dec 31 09:54:46 octoprint xinit[2385]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Dec 31 09:54:46 octoprint xinit[2385]: (==) Log file: "/var/log/Xorg.0.log", Time: Thu Dec 31 09:54:46 2020
Dec 31 09:54:46 octoprint xinit[2385]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Dec 31 09:54:47 octoprint xinit[2385]: 2020/12/31 09:54:47 logger.SetLogLevel() - the log level is now set to: "debug"
Dec 31 09:54:47 octoprint xinit[2385]: time="2020-12-31T09:54:47-08:00" level=info msg="logger.init() - logFilePath is was not defined.  Now using just the standard console output." file=logger.go func=utils.init.0 line=64
Dec 31 09:54:47 octoprint xinit[2385]: time="2020-12-31T09:54:47-08:00" level=debug msg="entering main.init()" file=main.go func=main.init.0 line=55
Dec 31 09:54:47 octoprint xinit[2385]: 2020/12/31 09:54:47 logger.SetLogLevel() - the log level is now set to: "error"
Dec 31 09:54:47 octoprint xinit[2385]: 2020/12/31 09:54:47 main.SetLogLevel() - logLevel is now set to: "error"
Dec 31 09:54:47 octoprint xinit[2385]: time="2020-12-31T09:54:47-08:00" level=error msg="ui.checkNotification() - Do(GetNotificationRequest) returned an error: \"unexpected status code: 404\"" file=logger.go func=utils.LogError line=86
Dec 31 09:54:47 octoprint xinit[2385]: time="2020-12-31T09:54:47-08:00" level=error msg="ui.loadSettings() - Do(GetSettingsRequest) returned an error: \"unexpected status code: 404\"" file=logger.go func=utils.LogError line=86
Dec 31 09:54:47 octoprint xinit[2385]: time="2020-12-31T09:54:47-08:00" level=error msg="leaving ui.loadSettings() - Do(GetSettingsRequest) returned an error" file=ui.go func="ui.(*UI).loadSettings" line=293
Dec 31 09:54:48 octoprint xinit[2385]: panic: runtime error: invalid memory address or nil pointer dereference
Dec 31 09:54:48 octoprint xinit[2385]: [signal SIGSEGV: segmentation violation code=0x1 addr=0x18 pc=0x48e324]
Dec 31 09:54:48 octoprint xinit[2385]: goroutine 1 [running]:
Dec 31 09:54:48 octoprint xinit[2385]: github.com/Z-Bolt/OctoScreen/ui.(*idleStatusPanel).initialize(0x12a0c540)
Dec 31 09:54:48 octoprint xinit[2385]:         /go/src/github.com/Z-Bolt/OctoScreen/obj-arm-linux-gnueabihf/src/github.com/Z-Bolt/OctoScreen/ui/IdleStatusPanel.go:47 +0x10c
Dec 31 09:54:48 octoprint xinit[2385]: github.com/Z-Bolt/OctoScreen/ui.IdleStatusPanel(0x1299e5a0, 0x12947940)
Dec 31 09:54:48 octoprint xinit[2385]:         /go/src/github.com/Z-Bolt/OctoScreen/obj-arm-linux-gnueabihf/src/github.com/Z-Bolt/OctoScreen/ui/IdleStatusPanel.go:31 +0x16c
Dec 31 09:54:48 octoprint xinit[2385]: github.com/Z-Bolt/OctoScreen/ui.(*UI).verifyConnection(0x1299e5a0)
Dec 31 09:54:48 octoprint xinit[2385]:         /go/src/github.com/Z-Bolt/OctoScreen/obj-arm-linux-gnueabihf/src/github.com/Z-Bolt/OctoScreen/ui/ui.go:245 +0x548
Dec 31 09:54:48 octoprint xinit[2385]: github.com/Z-Bolt/OctoScreen/ui.(*UI).update(0x1299e5a0)
Dec 31 09:54:48 octoprint xinit[2385]:         /go/src/github.com/Z-Bolt/OctoScreen/obj-arm-linux-gnueabihf/src/github.com/Z-Bolt/OctoScreen/ui/ui.go:372 +0x140
Dec 31 09:54:48 octoprint xinit[2385]: github.com/Z-Bolt/OctoScreen/ui.(*UI).(github.com/Z-Bolt/OctoScreen/ui.update)-fm()
[... cut ...]
So there's a regression, too: when `OCTOPRINT_APIKEY` is not defined, but `OCTOPRINT_CONFIG_FILE` is a valid file that contains a valid API key, the error is shown. This was _not_ the case on the 2.5 release.

> do you also have the the OctoScreen plug-in

I didn't know  that there was a plugin. I've added it to my octoprint install but will keep octoscreen disabled until i hear back.

Happy new years and thank you for giving us an alternative to octodash! I'm not a golang dev, but I'm more than happy to get more details on the stack trace sjown above!

JeffB42, 2020-12-31:
Thanks for the info @kquinsland. At this point, with the 2.6.0 release, there are the following known issues (that I know of):
1. The API key (`OCTOPRINT_APIKEY`) needs to be defined in `/etc/octoscreen/config`.  I think you are right - I think 2.5.1 got the API key from OctoPrint's config.yaml (`OCTOPRINT_CONFIG_FILE`).  I think at some point I was cleaning up the code, and didn't like how there were two configuration files, and I missed this point.  I'm going to look into this, and if 2.5 got the API key from OctoPrint's YAML file, I'm going to switch over to that convention.
2. The OctoScreen plugin is required.  If it's missing, OctoScreen references a null pointer and crashes.  This is something I missed.  I run with the plug-in installed and neglected to test when the plug-in wasn't present.
3. OctoScreen crashes if not connected to a printer.  If OctoPrint and OctoScreen start, but the RPi isn't connected to a printer that's turned on, OctoScreen will eventually crash.  I haven't reproduced the crash, but was able to reproduce getting the system to restart over and over until the printer is turned on.  This is a known issue and existed in 2.5.  This is something I will be looking into in a future release.

With issues `#1` and `#2`, this will be fixed in the upcoming 2.6.1 release, which I plan to publish in a day or two.
With issue `#3`, I suggest running with the printer turned on.  This will be fixed in a future update, but there isn't an ETA for this yet.

@kquinsland if you...
1. define the API key in `/etc/octoscreen/config`
2. and also install the OctoScreen plug-in
3. and run with the printer turned on
...do you still get the issue?

If you still experience the issue, could you do the following:
1. In `/etc/octoscreen/config`, set `OCTOSCREEN_LOG_FILE_PATH`.  Set it to something like `/home/pi/logs/logfile.txt`.
2. In the config file, set `OCTOSCREEN_LOG_LEVEL=debug`
2. Create the log file if it doesn't exist (just do `touch /home/pi/logs/logfile.txt`)
3. Reboot your RPi
4. Post the log file here.  NOTE: in the log file, it will most likely display your full API key.  You might want to remove it in your post.  The upcoming 2.6.1 release has a fix for this - the API key is still logged but only a few of the characters, not the full key.
5. After posting the log, you'll probably want to either reduce the logging level (set the level to `error`), or turn off logging (do this by removing the path of OCTOSCREEN_LOG_FILE_PATH)

Thanks,

Jeff

JeffB42, 2021-01-02:
Thanks for the help on this @boydhako and @kquinsland .  I just published v2.6.1.  (https://github.com/Z-Bolt/OctoScreen/releases/tag/2.6.1)  This release should resolve the issue you were each seeing.

Jeff

JeffB42, 2021-01-02:
@boydhako please update to 2.6.1.  If you see any issues, let me know, otherwise, please close this issue and I'll know that the issues have been resolved.

Scraptor86, 2021-01-13:
I have the same problem. I updated my octoscreen to newest version (2.6.1.) but the problem is still there. Any suggestions how to fix it?
Thx in advance.

JeffB42, 2021-01-31:
@Scraptor86 just what's already been mentioned in the issue.

JeffB42, 2021-03-27:
If this is the issue pertaining to OctoScreen rebooting if the printer is off, this has been fixed in 2.7.0.

closed 2021-03-27 by JeffB42