Z-Bolt/OctoScreen #193: Multiple extruders with shared nozzle

Mulle00, 2020-12-30:
I have a Prusa MK3S with MMU2S (for printing with 5 colors).
On the main page, I don't see extruders and heated bed.
Octoscreen recognizes this as 5 separate ones instead of one. 
Is it possible to fix this? This problem occurs in all versions.

[image]

JeffB42, 2020-12-30:
The issue is that OctoScreen only supports up to 4 hotends, and when it tries to created "hotend # 5" there's an error.  One temporary work around for this is to change the extruder count in OctoPrint to 4.

I'm going to publish an update either this week or over the weekend (2.6.1), and it will have a fix for it.  First, I'll limit the number in OctoScreen to 4, just in case there are 5+.  Also, I'll look into whether OctoPrint's API returns back if it's a shared nozzle or not.  If I can get this information, and it's a shared nozzle, I'll set the extruder count within OctoScreen to 1.

JeffB42, 2020-12-30:
I found the issue.  The fix will be published in the next release, 2.6.1 (ETA is a few days from now)

closed 2020-12-30 by JeffB42

Mulle00, 2020-12-30:
Thank you in advance for the further development of OctoScreen. It is not a big problem, but it would be nicer to see the correct information.
I think I found the correct information in the API for the shared nozzle.

[image]

JeffB42, 2020-12-30:
yea, I located the issue.  In https://github.com/Z-Bolt/OctoScreen/blob/e1c28348a41d65eb387c84c0af45742b650ff27d/utils/tools.go#L31

...the code is:

[image]

...and the order is swapped (the check for SharedNozzle should come after).  Anyway, I'll publish a fix for this an some other issues in the next few days.