Z-Bolt/OctoScreen #199: OctoScreen segfaults on startup

ctwise, 2021-01-03:
After installation following these steps: 

```
sudo apt-get install libgtk-3-0 xserver-xorg xinit x11-xserver-utils
wget https://github.com/Z-Bolt/OctoScreen/releases/download/2.6.1/octoscreen_2.6.1_armhf.deb
sudo dpkg -i octoscreen_2.6.1_armhf.deb
```

OctoScreen is installed. Rebooting shows the normal text startup then the screen goes black then returns to the text prompt. It then continues to cycle there. Service status shows very little:

```
$ sudo systemctl status octoscreen
● octoscreen.service - OctoScreen
   Loaded: loaded (/lib/systemd/system/octoscreen.service; enabled; vendor preset: enabled)
   Active: activating (start) since Sun 2021-01-03 20:07:25 GMT; 1s ago
 Main PID: 5350 (xinit)
   Memory: 5.2M
   CGroup: /system.slice/octoscreen.service
           ├─5350 /usr/bin/xinit /usr/bin/OctoScreen -- :0 -nolisten tcp -nocursor
           └─5351 /usr/lib/xorg/Xorg :0 -nolisten tcp -nocursor

Jan 03 20:07:25 octoscreen xinit[5350]: Build Date: 01 December 2020  05:59:57PM
Jan 03 20:07:25 octoscreen xinit[5350]: xorg-server 2:1.20.4-1+rpt2+deb10u2 (https://www.debian.org/support)
Jan 03 20:07:25 octoscreen xinit[5350]: Current version of pixman: 0.36.0
Jan 03 20:07:25 octoscreen xinit[5350]:         Before reporting problems, check http://wiki.x.org
Jan 03 20:07:25 octoscreen xinit[5350]:         to make sure that you have the latest version.
Jan 03 20:07:25 octoscreen xinit[5350]: Markers: (--) probed, (**) from config file, (==) default setting,
Jan 03 20:07:25 octoscreen xinit[5350]:         (++) from command line, (!!) notice, (II) informational,
Jan 03 20:07:25 octoscreen xinit[5350]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Jan 03 20:07:25 octoscreen xinit[5350]: (==) Log file: "/var/log/Xorg.0.log", Time: Sun Jan  3 20:07:25 2021
Jan 03 20:07:25 octoscreen xinit[5350]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
```

There is nothing in the logs about any errors. But if I do a strace -f on the xinit with OctoScreen I see it segfaults w/no additional information as to why.

This is with the Pimoroni HyperPixel screen. I tested the screen w/lightdm just to validate it and it worked perfectly.

I've looked through other issues but everyone else seems to have gotten further and is having issues with API keys, etc. I don't even get OctoScreen to startup.

Suggestions?

JeffB42, 2021-01-03:
I have a HyperPixel as well, but haven't run into this issue.  Did you watch Chris Riley's video on setting up a HyperPixel and OctoScreen?  It's at https://www.youtube.com/watch?v=OJ59hXSyBoI.

The only other suggestions I can think of are:
1. If you haven't done so already, try flashing a SD card with the latest image of OctoPi, and then run through the steps Chris mentions in his video
2. Turn logging on - edit nano /etc/octoscreen/config and set OCTOSCREEN_LOG_FILE_PATH to the path of a log file (like /home/pi/logs/logfile.txt and be sure to create the file as well)
3. Set the log level to OCTOSCREEN_LOG_LEVEL=debug

Jeff

JeffB42, 2021-02-04:
@ctwise any updates on this?

Serferuz, 2021-02-09:
I have this issue with RPi 3B and 3.5" 480x320 TFT screen.

closed 2021-03-27 by JeffB42